gnuton/asuswrt-merlin.ng #605: AX58U - cpu2 (core 3) always idle

nativ69, 2024-06-13:
**Router Model Affected**
Models: AX58U

**Firmware Version Affected**
3004.388.7

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
CPU2 always idle.
After some digging I found that's kernel flag isolcpus is set to 2, which means that the kernel processes scheduler will not interact with CPU2.
It's also looks like irqaffinity is set to 0, which means that by default only CPU0 will handle IRQs.
I don't know if this is the intended behavior or it's a bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Login SSH
2. cat /proc/cmdline
3. output:
`isolcpus=2 root=/dev/ubiblock0_0 ubi.mtd=0 ubi.block=0,0 rootfstype=squashfs console=ttyAMA0 earlyprintk debug irqaffinity=0 pci=pcie_bus_safe`

**Expected behavior**
if that's a bug, I would expect that isolcpus flag will be removed from boot options.
if that's not a bug, it would be nice to have an option (maybe there is already) to edit boot parameters.

robbbaxley, 2024-06-13:
I have seen (and reported) this for CPU3 (Core 4) on the XT8 as well. Gnuton has reported this is normal behavior.

gnuton, 2024-06-13:
As @robbbaxley this is by design like this and it comes from ASUS.
Merlin is not changing this behaviour since we have not evidence of improved performance.
You are free to locally change it to have more aggressive settings.
Let us know if you see any benefit

nativ69, 2024-06-13:
@gnuton how can I change bootloader settings?
currently i use taskset to manually override cpu assigned to each process.

gnuton, 2024-06-14:
here is the file release/src-rt-5.04axhnd.675x/targets/96750GW/96750GW.RT-AX82U_V2

swwwong, 2024-08-05:
@gnuton I have test with my old RT-AX58U installed Merlin (mabe other dev), and find that it have default assigned task to all 3 cpu core,compare with DSL-AX82U ,I can run  FlexQoS with out serioyus  performances impect,e.g ssh lost connection,internet connection lost when doing speed test or downloading large file in full speed (500Mbps).