gnuton/asuswrt-merlin.ng #606: WiFi 6E not working on ASUS ET8

naitkris, 2024-06-13:
**Router Model Affected**
ASUS ET8 H/W Ver. 1.0

**Firmware Version Affected**
3004.388.7_1-gnuton1

**Is this bug present in upstream Merlin releases too?**
No (ET8 not supported except in Gnuton)

**Describe the bug**
Facing an issue with WiFi 6E on the ASUS ET8 H/W Ver. 1.0 (MFG Year 2024). With the latest ASUS official firmware version 3.0.0.4.388_23759 (dated 8 March 2023) WiFi 6E works fine both as a separate 6GHz SSID and when using Tri-Band Smart Connect - I am running just one ET8 unit now as a standalone router. However with the latest GNUton's Asus Merlin 3004.388.7_1-gnuton1 (dated 10 June 2024) the 6GHz SSID shows up in the ASUS web based interface but it is not actually broadcasting and none of my WiFi 6E clients see the network. I played around with changing a few different settings but nothing I tried worked.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash with RT-AXE95Q_3004_388.7_1-gnuton1_puresqubi.w and do a settings reset
2. Configure the ET8 as a router and enable separate 6GHz network (or Tri-Band Smart Connect) - ensure SSID of the 6GHz SSID is broadcasting
3. Have a WiFi 6E client to scan for 6E networks (I am using Samsung S23)
4. No WiFi 6E network is seen (however with latest official ASUS firmware for the ET8 it is seen and clients connect)

**Expected behavior**
WiFi 6E network broadcasts and WiFi 6E clients can connect.

**Screenshots**
No screenshots but can take screenshots or logs of specific things if requested.

CPngN, 2024-06-13:
Seems like this keeps happening.  Honestly I can't say I've ever had a stable Wifi6 on my ET8's.  Lately I just disable it - why bother powering up something I can't use?

naitkris, 2024-06-23:
OK, issue is now solved! See below.

After doing lots of tests including comparing the output of "nvram show" of the latest Gnuton firmware for the ET8 (RT-AXE95Q_3004_388.7_1-gnuton1_puresqubi.w) for which I faced the issue with WiFi 6E "not working" against the latest official stock firmware for the ET8 (ZENWIFI_ET8_3.0.0.4_388_23759-gfd1c030_puresqubi.w) for which WiFi 6E is working fine I noticed the below differences:

Gnuton firmware:
`wl2_sel_bw=160
wl2_sel_channel=**197**`

Official stock firmware:
`wl2_sel_bw=160
wl2_sel_channel=**69**`

So for the 6 GHz they both default the 6 GHz to 160 MHz channel bandwidth which is fine but the Gnuton firmware sets the default channel (which it is set to when "Auto" is selected - and "Auto" is selected by default) to channel 197 while the official firmware sets it to channel 69. Most countries and regions only support up to channel 93 (so up to 6425 MHz frequency) and it appears as per https://eu.community.samsung.com/t5/galaxy-s24-series/samsung-s-decision-to-restrict-wifi-5ghz-and-6ghz-channels-based/td-p/9918035 and elsewhere that Samsung (and perhaps some other vendors) have region locked their WiFi 6E clients to only connect to the 6 GHz channels for WiFi 6E as per where the mobile handset was purchased. In my case my Samsung Galaxy S23+ was purchased in the UAE while my ASUS ET8 is from the US.

Based on this, and also based on a lot of testing and reading that I have done, it would be good if the future version of the Gnuton firmware for the ET8 could have the default 6 GHz channel instead be 53 (which I found to work very well after a lot of testing), 69 (same as official stock firmware), or 37 - these three channels are supported by practically all countries and regions as per https://en.wikipedia.org/wiki/List_of_WLAN_channels#6_GHz_(802.11ax_and_802.11be) and all three are in the list of PSCs (Preferred Scanning Channels) as well - thus allowing maximum clients from all over the world to connect to the 6 GHz WiFi 6E.

CPngN, 2024-06-25:
Nice find, not immediately obvious especially to people newer to routers.
👍🏻

gnuton, 2024-07-01:
Thanks @naitkris! I will check were this issue comes from!
absolutely nice catch!

gnuton, 2024-07-11:
shared/sysdeps/qca/qca.c sets this values..
investigation on-going

CPngN, 2024-07-12:
Cool, thank you!

gnuton, 2024-07-27:
I checked how the firmaware handle this and it basically check a "terrory_code" nvram setting.
According to my tet some process set the terrory_code.

closed 2024-07-28 by gnuton