gnuton/asuswrt-merlin.ng #607: Web history and statistic

dufizke, 2024-06-13:
Router Model Affected
Models: TUF-AX3000 V2

Firmware Version Affected
3004.388.7_1-gnuton1

Is this bug present in upstream Merlin releases too?
No, when I test Merlin fw 386.13 on my old RT-AC68U 

Describe the bug
When logged in to TUF-AX3000V console and switched in General tab to Adaptive QoS then to Web History, then Web History turned ON, it does not show any data even after couple of days, so I can not simply filter per client. The same happens when you switch to Traffic Analyzer in General tab, then to Statistic, when Trafic Analyzer - Statistic is switched ON,  it does not show any data even after couple of days, so I can not simply filter per client.

To Reproduce
Steps to reproduce the behavior:
1. Go to ' General tab'
2. Click on 'Adaptive QoS'
3. Click on 'Web history'
4. Turn On ´Web History´
5. See error

1. Go to ' General tab'
2. Click on 'Traffic Analyzer'
3. Click on 'Statistic'
4. Turn On ´Trafic Analyzer - Statistic´
5. See error

Expected behavior
It should come with details such as address, time, etc.... which web sites were connected devices visiting.

gnuton, 2024-06-13:
Hi,
have you reset the router after flashing Merlin?

dufizke, 2024-06-14:
> Hi,
> have you reset to facyory the router after flashing Merlin?

Yes I did, but without any luck.

dacootmeister, 2024-06-14:
Experience the same with my RT-AX95Q

gnuton, 2024-07-16:
Hi,
I tested this on 3004.388.8_0-gnuton0_alpha1_tuf  and it works 
[image]
This is very likely to happen if you update from stock and you do flash or you restore settings not belonging to merlin.

closed 2024-07-16 by gnuton