gnuton/asuswrt-merlin.ng #609: Request for --ttl-set in iptables. RT-AX82U_V2

IajentI, 2024-06-14:
For example:
Would like to use the command `iptables -t mangle -A POSTROUTING -o usb0 -j TTL --ttl-set 65`
When I currently try it gives me an error of `iptables: No chain/target/match by that name`
Thanks

gnuton, 2024-06-14:
That requires xt ttl module. I am not sure if this a feature we wanna get yet.
Is it supported by other merlin builds?

to enable it you have to rebuild the kernel
with the following modules

CONFIG_NETFILTER
CONFIG_NETFILTER_XTABLES
CONFIG_IP_NF_IPTABLES
CONFIG_IP_NF_MANGLE
CONFIG_NETFILTER_XT_TARGET_TTL
CONFIG_NETFILTER_XT_MATCH_TTL

This is the files where this have to be changed https://github.com/gnuton/asuswrt-merlin.ng/commit/99ed888521d704d7b9a8f99ca2bc7cec9c6f7826#diff-4c8da1983064f7fd1cfb4b4a1a94147ad57fe0e4142e286928c02d14f5646671

IajentI, 2024-06-14:
Thank you for the quick reply.
I'm not sure if any current builds still have it. I last used it many years ago on an old T-mobile branded router that was actually an Asus router. Back then I flashed either merlin or dd-wrt and was able to change the TTL. My only experience with kernels is on android, I've only used scripts to modify them automatically, never manually. I may play around with the info you've given me to see if I can get it working, or brick it, lol.
Thanks again

gnuton, 2024-06-18:
it won't brick. worst scenario oyu can flash back a stock firmwre from when in recovery mode..
it's very safe to restore and recover ASUS routers as long as you do not put your hands in the bootloader,