gnuton/asuswrt-merlin.ng #610: RT-AX82Uv2 - unable flash  3004.388.7_1-gnuton1

bdkacz, 2024-06-17:
**Router Model Affected**
Models: RT-AX82Uv2

**Firmware Version Affected**
3004.388.7_1-gnuton0_beta1 , 3004.388.7_1-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
I have RT-AX82U v2 with firmware 3004.388.7_1-gnuton0_beta1  and I try flash it to newest stable version  with file RT-AX82U_V2_3004_388.7_1-gnuton1_nand_squashfs.pkgtb but after reboot  firmware version is still  3004.388.7_1-gnuton0_beta1 

**To Reproduce**
Steps to reproduce the behavior:
1. Download latest stable release 3004.388.7_1-gnuton1 file  RT-AX82U_V2_3004_388.7_1-gnuton1_nand_squashfs.pkgtb
2. Go to  Administration-Upgrade firmware
3. Click browse and select file
4. Wait till reboot
5. Login to router panel

bdkacz, 2024-06-18:
Logs just after reboot after flashing in attachment
[syslog.txt](https://github.com/user-attachments/files/15887277/syslog.txt)

gnuton, 2024-06-24:
To force the flash you can

1. Enable SSH . https://github.com/gnuton/asuswrt-merlin.ng/issues/168
2. Transfer the firmware.w file to router via WinSCP
3. Login to router via Putty
4. Use command: hnd-write firmware.w
5. Reboot

bdkacz, 2024-06-24:
Flashing via winscp  successfully upgraded the router to 3004.388.7_1-gnuton1, tkank You :)

commit cf87691a (2020-11-11): nano: updated to 5.2 (closes #610)
closed 2024-06-24 by bdkacz