gnuton/asuswrt-merlin.ng #612: Read-only /etc/iproute2/rt_table prevents custom routing

Beta-Blaze, 2024-06-18:
**Router Model Affected**
Models: AX3000_V2

**Firmware Version Affected**
[3004.388.7_1-gnuton1](https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3004.388.7_1-gnuton1)

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
The file `/etc/iproute2/rt_table` is mounted as read-only, preventing users from adding custom routing rules. This issue hinders advanced networking configurations that rely on manipulating routing tables, such as policy-based routing or source-based routing.

**Expected behavior**
The file /etc/iproute2/rt_table should be modifiable, allowing me to tweak its content to suit my networking requirements.

**Additional context**
I have tried searching for solutions and found users suggesting the creation of a script /jffs/scripts/init-start (as per this forum thread: http://www.snbforums.com/threads/port-forward-while-using-vpn-client.28014/#post-215936).
However, that does not seem to resolve the issue. I also referred to the official documentation on Custom config files (https://github.com/RMerl/asuswrt-merlin.ng/wiki/Custom-config-files) but found no suitable solution.

I apologise in advance if it was not possible to ask something like this in Issue.
I wanted to ask in Discord, but the chat link doesn't work(
Is there some easy way to do this that I didn't think of?

RMerl, 2024-07-05:
Not a bug, simply a limitation because the file is stored in flash, it's not generated at boot time.  You need to mount a writable file on top of it to replace it.

```
admin@stargate:/tmp/home/root# cp /etc/iproute2/rt_tables /jffs/rt_tables
admin@stargate:/tmp/home/root# mount -o bind /jffs/rt_tables /etc/iproute2/rt_tables 
admin@stargate:/tmp/home/root# nano /jffs/rt_tables 
```

That's where using a user script is required to remount it at boot time.

Beta-Blaze, 2024-07-05:
Thank you so much for your reply!

closed 2024-07-05 by Beta-Blaze