gnuton/asuswrt-merlin.ng #613: Any way to increase 4tb limit?

ajalberd, 2024-06-20:
**Is your feature request related to a problem? Please describe.**
I love the ability to make a SMB share on the router itself, however my14TB drive cannot mount, seems like a partition limit unfortunately

**Describe the solution you'd like**
Able to export the 14TB drive in router menu

**Describe alternatives you've considered**
n/a :(

**Additional context**
n/a

ajalberd, 2024-06-20:
Got it to work! I formatted using the router's `mke2fs -t ext4 -O ^has_journal /dev/sda1`

It now has these features:

`Filesystem features:      ext_attr resize_inode dir_index filetype extent flex_bg sparse_super large_file huge_file uninit_bg dir_nlink extra_isize`

gnuton, 2024-06-23:
can this be closed?

ajalberd, 2024-06-23:
> can this be closed?

Sure! I was curious if you had any insight on this one.

gnuton, 2024-06-24:
You tried to mount an unsupported filesystem. What's that NTFS?

closed 2024-06-24 by gnuton