gnuton/asuswrt-merlin.ng #614: RT-AX82U v1 bad performance when using wireguard

feiercrack, 2024-06-20:
i have bad performance when using wireguard - download speeds ~180mbps
when using ovpn ~800mbps
when not using any vpn ~ 1 000mbps
i am on win11x64pro24h2
fritz box 6690 cable isp: vodafone down: 1 GBit/s up: 50mbps
connected to my Asus RT-AX82U v1 with current 3004.388.7_1-gnuton1 firmware and surfshark wireguard client protocol
PC is connected to  Asus Router via an CAT 6A cable.
VPN Provider is Surfshark
When using the Surfshark App for Windows, i get ~700mbps
upload speeds are always the same ~50mbps
i played around with the MTU (windows network and in the routers wireguard client tab)
its 1387
i can remember, that the original asus fw had ~400mbps.
ipv6 is completely off, from the fritz box over to the asus router to the windows network connection.
QoS in Router is disabled (when enabled, download performance even worser)
what can be done here?
thanxx ^^

gnuton, 2024-06-24:
Hi,
What's the status of the LAN hw acceleration? http://www.asusrouter.com/Tools_Sysinfo.asp
You can change settings from http://www.asusrouter.com/Advanced_SwitchCtrl_Content.asp

feiercrack, 2024-06-26:
hey,

here are my stats:
[image]
[image]
[image]

no matte what i change there, it even gets worser..

feiercrack, 2024-06-29:
are you still alive @gnuton ???

so what to do?
is the CPU from my RT-AX82U to weak?
may an upgrade to an GT-AX11000 Pro will help in performance?

swwwong, 2024-06-30:
@feiercrack 

I have a DSL-AX82U,when I use a devices connect to router's wireguard client and do a speed test .
It only get about 100Mbps,compare to the PC  in the same VPN connection profile which get 462Mbps whith in my 500M FTTP.

more over,when I use router wireguard client,my doing the speed test ,the router's cpu0,cpu1@100%,cpu2@60%,router's SSH,WIFI,and other devices which no in the wireguard client connetion will drop internet connection .