gnuton/asuswrt-merlin.ng #615: RT-AX58U Cake and FlexQoS not running due to Hardware acceleration runner being enabled.

Nullvoid3771, 2024-06-24:
**Router Model Affected**
Models: (eg: DSL-AX82U/RT-AX82U/...)

**Firmware Version Affected**
388.7_1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
Cake and FlexQoS doesn't work because Hardware acceleration runner won't turn off. 

**To Reproduce**
Steps to reproduce the behavior:

Apparently on gnuton 388.7 flowcache archer(runner) doesn’t disable itself automatically which is affecting both flexqos and cake. You need to manually disable it for either of these to work however despite ssh confirming the value of the runner being disabled manually, even after a reboot the UI still shows archer:enable under tools->sysinfo. 

Runner is the successor to Flow Accelerator. It's a second level of hardware acceleration that is incompatible with a lot of router features. ~Rmerlin

If flow cache is enabled and needs to be disabled:

nvram set fc_disable_force=1
nvram commit
fc disable
fc flush

To check status:

nvram get fc_disable

To undo:

nvram unset fc_disable_force
nvram commit

Disable Archer (Runner/Flow Accelerator)

nvram set runner_disable_force=1
nvram commit

To check status:

 nvram get runner_disable

To undo:

nvram unset runner_disable_force
nvram commit

Reboot to take effect.

**Expected behavior**
Hardware acceleration is not compatible with Cake or FlexQoS. This may also simply be a issue with CakeQoS webui, but since the systools shows archer is enabled as well this appears to be a bug in Gnuton. 
**Screenshots**
Attached. 
[image]
[image]
[image]
[image]
[image]

Nullvoid3771, 2024-06-24:
Discussion on SNB fourms: https://www.snbforums.com/threads/flexqos-settings-problem.90668/page-3

gnuton, 2024-06-24:
Hi, 
if you open the javascript console of your browser when changing  options, do you get any errors?

Nullvoid3771, 2024-06-24:
[image]
[image]
[image]
[image]

Attempted enabling qos, changing settings, disabling hw runner in ssh and rebooting. No errors jump out at me.

Nullvoid3771, 2024-06-24:
[image]

novusB, 2024-06-24:
Not having that issue on my router. AX3000-V2

FW: 3004.388.7_1-gnuton1

[image]
[image]

Nullvoid3771, 2024-06-24:
It actually doesn't show anything referencing hwaccel in Tools_Sysinfo.asp with qos enabled with cake. 

Opened the browser console instead of web inspector:

SearchSuggestionController found an unexpected string value: HTTP request timeout SearchSuggestionController.sys.mjs:645:17
LoginManagerPrompter: Couldn't process supplied URIString: https://192.168.50.1:8443 20 LoginManagerPrompter.sys.mjs:999:16
Unable to find target with innerWindowId:8589934639 watcher.js:188:13
TypeError: currentNode is null
    canMutationImpactCurrentStyles resource://devtools/client/inspector/shared/style-change-tracker.js:66
    onMutations resource://devtools/client/inspector/shared/style-change-tracker.js:84
    _emit resource://devtools/shared/event-emitter.js:242
    emit resource://devtools/shared/event-emitter.js:186
    emit resource://devtools/shared/event-emitter.js:330
    getMutations resource://devtools/client/fronts/walker.js:314
event-emitter.js:257:19
Listener for event 'frame' did not return a promise. event-emitter.js:248:23
Unable to find target with innerWindowId:8589934647 watcher.js:188:13
Listener for event 'frame' did not return a promise. 2 event-emitter.js:248:23
LoginManagerPrompter: Couldn't process supplied URIString: https://192.168.50.1:8443 10 LoginManagerPrompter.sys.mjs:999:16
Listener for event 'frame' did not return a promise. event-emitter.js:248:23
Unable to find target with innerWindowId:8589934653 watcher.js:188:13
Listener for event 'frame' did not return a promise. event-emitter.js:248:23
Unable to find target with innerWindowId:8589934654 watcher.js:188:13
Listener for event 'frame' did not return a promise. 2 event-emitter.js:248:23
Unable to find target with innerWindowId:8589934657 watcher.js:188:13
Listener for event 'frame' did not return a promise. event-emitter.js:248:23
Unable to find target with innerWindowId:8589934661 watcher.js:188:13
Listener for event 'frame' did not return a promise. event-emitter.js:248:23
Unable to find target with innerWindowId:8589934664 watcher.js:188:13
Listener for event 'frame' did not return a promise. 3 event-emitter.js:248:23
Unable to find target with innerWindowId:8589934667 watcher.js:188:13
Listener for event 'frame' did not return a promise. 2 event-emitter.js:248:23
Security state 0 has no known STATE_IS_* flags. 2 DevToolsInfaillibleUtils.sys.mjs:22:13
Listener for event 'frame' did not return a promise. event-emitter.js:248:23
Unable to find target with innerWindowId:8589934676 watcher.js:188:13
Listener for event 'frame' did not return a promise. 2 event-emitter.js:248:23
LoginManagerPrompter: Couldn't process supplied URIString: https://192.168.50.1:8443 10 LoginManagerPrompter.sys.mjs:999:16
Listener for event 'frame' did not return a promise. event-emitter.js:248:23
Unable to find target with innerWindowId:8589934682 watcher.js:188:13
Listener for event 'frame' did not return a promise. 3 event-emitter.js:248:23
TypeError: currentNode is null
    canMutationImpactCurrentStyles resource://devtools/client/inspector/shared/style-change-tracker.js:66
    onMutations resource://devtools/client/inspector/shared/style-change-tracker.js:84
    _emit resource://devtools/shared/event-emitter.js:242
    emit resource://devtools/shared/event-emitter.js:186
    emit resource://devtools/shared/event-emitter.js:330
    getMutations resource://devtools/client/fronts/walker.js:314
event-emitter.js:257:19
Unable to find target with innerWindowId:8589934690 watcher.js:188:13
Unable to find target with innerWindowId:8589934683 watcher.js:188:13
Unable to find target with innerWindowId:8589934690 watcher.js:188:13
Listener for event 'frame' did not return a promise. event-emitter.js:248:23
Unable to find target with innerWindowId:8589934691 watcher.js:188:13
Listener for event 'frame' did not return a promise. 3 event-emitter.js:248:23
Unable to find target with innerWindowId:8589934694 watcher.js:188:13
Listener for event 'frame' did not return a promise. 3 event-emitter.js:248:23
Unable to find target with innerWindowId:8589934699 watcher.js:188:13
Listener for event 'frame' did not return a promise. 2 event-emitter.js:248:23
Unable to find target with innerWindowId:8589934704 watcher.js:188:13
Listener for event 'frame' did not return a promise. event-emitter.js:248:23
Unable to find target with innerWindowId:8589934708 watcher.js:188:13
Listener for event 'frame' did not return a promise. event-emitter.js:248:23
Security state 0 has no known STATE_IS_* flags. 8 DevToolsInfaillibleUtils.sys.mjs:22:13
HTTPS-Only Mode: Not upgrading insecure request “http://o.pki.goog/wr2” because it is exempt.
Security state 0 has no known STATE_IS_* flags. 4 DevToolsInfaillibleUtils.sys.mjs:22:13
HTTPS-Only Mode: Not upgrading insecure request “http://r3.o.lencr.org/” because it is exempt. 2
Security state 0 has no known STATE_IS_* flags. 17 DevToolsInfaillibleUtils.sys.mjs:22:13
NotFoundError: No such JSProcessActor 'DevToolsProcess' 5
HTTPS-Only Mode: Not upgrading insecure request “http://detectportal.firefox.com/success.txt?ipv4” because it is exempt.
HTTPS-Only Mode: Not upgrading insecure request “http://detectportal.firefox.com/success.txt?ipv6” because it is exempt.
HTTPS-Only Mode: Not upgrading insecure request “http://detectportal.firefox.com/canonical.html” because it is exempt.
HTTPS-Only Mode: Not upgrading insecure request “http://detectportal.firefox.com/success.txt?ipv4” because it is exempt.
[... cut ...]
NS_ERROR_NOT_IMPLEMENTED: Component returned failure code: 0x80004001 (NS_ERROR_NOT_IMPLEMENTED) [nsIAutoCompleteInput.popupElement] 4 LoginManagerPrompter.sys.mjs:80
Listener for event 'frame' did not return a promise. event-emitter.js:248:23
Unable to find target with innerWindowId:8589934735 watcher.js:188:13
Listener for event 'frame' did not return a promise. 2 event-emitter.js:248:23
NS_ERROR_NOT_IMPLEMENTED: Component returned failure code: 0x80004001 (NS_ERROR_NOT_IMPLEMENTED) [nsIAutoCompleteInput.popupElement] 4 LoginManagerPrompter.sys.mjs:80
NotFoundError: No such JSProcessActor 'DevToolsProcess' 5

Nullvoid3771, 2024-06-24:
console in multi process

SearchSuggestionController found an unexpected string value: HTTP request timeout SearchSuggestionController.sys.mjs:645:17
    #dedupeAndReturnResults resource://gre/modules/SearchSuggestionController.sys.mjs:645
    fetch resource://gre/modules/SearchSuggestionController.sys.mjs:342
LoginManagerPrompter: Couldn't process supplied URIString: https://192.168.50.1:8443 20 LoginManagerPrompter.sys.mjs:999:16
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Security Error: Content at https://192.168.50.1:8443/user1.asp may not load or link to resource://gre/modules/ExtensionContent.sys.mjs.
: Failed to open input source 'resource://gre/modules/ExtensionContent.sys.mjs'
    mainThreadFetch resource://devtools/shared/DevToolsUtils.js:719
    mainThreadFetch resource://devtools/shared/DevToolsUtils.js:566
    _fetchURLContents resource://devtools/server/actors/utils/sources-manager.js:467
    urlContents resource://devtools/server/actors/utils/sources-manager.js:431
    _resurrectSource resource://devtools/server/actors/thread.js:2211
    addAllSources resource://devtools/server/actors/thread.js:1531
    watch resource://devtools/server/actors/resources/sources.js:83
    watchResources resource://devtools/server/actors/resources/index.js:340
    addOrSetSessionDataEntry resource://devtools/server/actors/targets/session-data-processors/resources.js:19
    addOrSetSessionDataEntry resource://devtools/server/actors/targets/base-target-actor.js:54
    #addOrSetSessionDataEntry resource://devtools/server/connectors/js-process-actor/DevToolsProcessChild.sys.mjs:380
    receiveMessage resource://devtools/server/connectors/js-process-actor/DevToolsProcessChild.sys.mjs:305
ThreadSafeDevToolsUtils.js:82:13
Failed to open input source 'resource://gre/modules/ExtensionContent.sys.mjs' DevToolsUtils.js:719
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Security Error: Content at https://192.168.50.1:8443/user1.asp may not load or link to resource://gre/modules/ExtensionContent.sys.mjs.
: Failed to open input source 'resource://gre/modules/ExtensionContent.sys.mjs'
    mainThreadFetch resource://devtools/shared/DevToolsUtils.js:719
    mainThreadFetch resource://devtools/shared/DevToolsUtils.js:566
    _fetchURLContents resource://devtools/server/actors/utils/sources-manager.js:467
    urlContents resource://devtools/server/actors/utils/sources-manager.js:431
    _resurrectSource resource://devtools/server/actors/thread.js:2211
    addAllSources resource://devtools/server/actors/thread.js:1531
    watch resource://devtools/server/actors/resources/sources.js:83
    watchResources resource://devtools/server/actors/resources/index.js:340
    addOrSetSessionDataEntry resource://devtools/server/actors/targets/session-data-processors/resources.js:19
    addOrSetSessionDataEntry resource://devtools/server/actors/targets/base-target-actor.js:54
    #addOrSetSessionDataEntry resource://devtools/server/connectors/js-process-actor/DevToolsProcessChild.sys.mjs:380
    receiveMessage resource://devtools/server/connectors/js-process-actor/DevToolsProcessChild.sys.mjs:305
ThreadSafeDevToolsUtils.js:82:13
Failed to open input source 'resource://gre/modules/ExtensionContent.sys.mjs' DevToolsUtils.js:719
Referrer Policy: Ignoring the less restricted referrer policy “no-referrer-when-downgrade” for the cross-site request: https://github.githubassets.com/favicons/favicon.svg FaviconLoader.sys.mjs:129:7
Referrer Policy: Ignoring the less restricted referrer policy “no-referrer-when-downgrade” for the cross-site request: https://raw.githubusercontent.com/gnuton/asuswrt-merlin.ng/master/www/donate.png donate.png
Referrer Policy: Ignoring the less restricted referrer policy “no-referrer-when-downgrade” for the cross-site request: https://github.githubassets.com/favicons/favicon.svg FaviconLoader.sys.mjs:129:7
Referrer Policy: Ignoring the less restricted referrer policy “no-referrer-when-downgrade” for the cross-site request: https://github.githubassets.com/favicons/favicon-dark.svg FaviconLoader.sys.mjs:129:7
Referrer Policy: Ignoring the less restricted referrer policy “no-referrer-when-downgrade” for the cross-site request: https://github.githubassets.com/favicons/favicon-dark.svg FaviconLoader.sys.mjs:129:7
Unable to find target with innerWindowId:8589934639 watcher.js:188:13
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
Use of nsIFile in content process is deprecated. NetUtil.sys.mjs:249:50
[... cut ...] aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI0VDA3MzgyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE3MjJhNDFiNjM4ODY5M2YzNjM3YzJmNmMyZDc3OWYzODgyOGNjYzE5OGViODNhMzdiN2YyNmZkZTMxYTBlMzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Lw3MSP7pkGu7CEdDTOpHmMN-iYpQUfwngCy4hB0lFMQ chunk-app_components_branch_create-branch-element_ts-0bc26eebee18.js:1:12594

Adi2071, 2024-06-24:
https://github.com/Last-times/CAKE-QoS-Script-OpenWrt/blob/main/cake.sh

Maybe it's enough to remove the Asus cake qos and use another script. Adapt it so that you can set values ​​in the GUI.  Asus cake and adaptive qos is terrible. It doesn't work properly or doesn't work at all. The only improvement I see when using qos is that the dots are close together from the ping to bufferbloat.

Nullvoid3771, 2024-06-24:
[image]

Adi2071, 2024-06-24:
In asus cake qos merlin sucks just like flexqos, adaptive qos. This qos is broken. Qos must work identically to SQM in OpenWRT. There qos works perfectly.

dave14305, 2024-06-25:
![Screenshot 2024-06-24 002935](https://private-user-images.githubusercontent.com/88042880/342201815-f9440f32-d91b-40a0-84ee-fc48c681c6ec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkyNzczMzAsIm5iZiI6MTcxOTI3NzAzMCwicGF0aCI6Ii84ODA0Mjg4MC8zNDIyMDE4MTUtZjk0NDBmMzItZDkxYi00MGEwLTg0ZWUtZmM0OGM2ODFjNmVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI1VDAwNTcxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNlN2Y4ZTc1YWVhODk5MGU1OGRhNTdjNzBlMzhlOTYyNTMzYTU5ZTkwYjFlZmYxMzIzYjJlYTQ3NmFiODMzOTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Nr7likyu0hzcJaeOJbnNjv5p2iJWEflaRfH7wAijmHw)
What is the output of `tc qdisc` when you see this message?

Nullvoid3771, 2024-06-25:
`@RT-AX58U-B260:/tmp/home/root# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2
qdisc noqueue 0: dev archer root refcnt 2
qdisc htb 1: dev eth0 root refcnt 2 r2q 10 default 0 direct_packets_stat 47664 d                                                                                                                                           irect_qlen 1000
qdisc sfq 1005: dev eth0 parent 10:5 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc sfq 1003: dev eth0 parent 10:3 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc htb 15: dev eth0 parent 1:15 r2q 10 default 0x256 direct_packets_stat 0 di                                                                                                                                           rect_qlen 1000
qdisc sfq 2: dev eth0 parent 1:2 limit 127p quantum 1514b depth 127 divisor 1024                                                                                                                                           
qdisc sfq 1007: dev eth0 parent 10:7 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc sfq 8005: dev eth0 parent 17:5 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc sfq 8003: dev eth0 parent 17:3 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc sfq 7002: dev eth0 parent 16:2 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc htb 17: dev eth0 parent 1:17 r2q 10 default 0x256 direct_packets_stat 0 di                                                                                                                                           rect_qlen 1000
qdisc sfq 1256: dev eth0 parent 10:256 limit 127p quantum 1514b depth 127 diviso                                                                                                                                           r 1024
qdisc sfq 8007: dev eth0 parent 17:7 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc sfq 7004: dev eth0 parent 16:4 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc sfq 8256: dev eth0 parent 17:256 limit 127p quantum 1514b depth 127 diviso                                                                                                                                           r 1024
qdisc sfq 6005: dev eth0 parent 15:5 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc sfq 6003: dev eth0 parent 15:3 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc sfq 5002: dev eth0 parent 14:2 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc sfq 6007: dev eth0 parent 15:7 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc sfq 5004: dev eth0 parent 14:4 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc sfq 6256: dev eth0 parent 15:256 limit 127p quantum 1514b depth 127 diviso                                                                                                                                           r 1024
qdisc sfq 4005: dev eth0 parent 13:5 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc sfq 4003: dev eth0 parent 13:3 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
qdisc sfq 3002: dev eth0 parent 12:2 limit 127p quantum 1514b depth 127 divisor                                                                                                                                            1024
[... cut ...]
qdisc sfq 2002: dev br0 parent 11:2 limit 127p quantum 1514b depth 127 divisor 1024
qdisc htb 11: dev br0 parent 1:11 r2q 10 default 0x256 direct_packets_stat 0 direct_qlen 1000
qdisc sfq 3007: dev br0 parent 12:7 limit 127p quantum 1514b depth 127 divisor 1024
qdisc sfq 2004: dev br0 parent 11:4 limit 127p quantum 1514b depth 127 divisor 1024
qdisc htb 13: dev br0 parent 1:13 r2q 10 default 0x256 direct_packets_stat 0 direct_qlen 1000
qdisc sfq 3256: dev br0 parent 12:256 limit 127p quantum 1514b depth 127 divisor 1024
qdisc noqueue 0: dev wgs1 root refcnt 2
`

Nullvoid3771, 2024-06-25:
@RT-AX58U-B260:/tmp/home/root# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2
qdisc noqueue 0: dev archer root refcnt 2
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc ingress ffff: dev eth0 parent ffff:fff1 ----------------
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_us_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_ds_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc noqueue 0: dev br0 root refcnt 2
qdisc noqueue 0: dev wgs1 root refcnt 2
qdisc pfifo_fast 0: dev ifb4eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1

Nullvoid3771, 2024-06-25:
@RT-AX58U-B260:/tmp/home/root# tc -s -d class show dev br0
@RT-AX58U-B260:/tmp/home/root# tc -s -d class show dev eth0
@RT-AX58U-B260:/tmp/home/root# tc -s -d class show dev lo
@RT-AX58U-B260:/tmp/home/root# tc -s -d class show dev eth1
@RT-AX58U-B260:/tmp/home/root#

Maybe this command is not compatible. hm.

Nullvoid3771, 2024-06-25:
@RT-AX58U-B260:/tmp/home/root# tc qdisc       This is with qos off. 
qdisc noqueue 0: dev lo root refcnt 2
qdisc noqueue 0: dev archer root refcnt 2
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_us_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_ds_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc noqueue 0: dev br0 root refcnt 2
qdisc noqueue 0: dev wgs1 root refcnt 2

Yakumounr, 2024-06-25:
DJones on the snbforum asked me to post here also.

DSL-AX82U (Not v2).
3004.388.7_1-gnuton1 (clean install).

Bufferbloat tests performed via https://www.waveform.com/tools/bufferbloat

When CakeQoS-Merlin v2.1.2 is running the Tools page reports Archer Disabled,  Flow Cache Disabled, 
Bufferbloat tests 
DL 58.0 Mbps +4ms
UL15.7 Mbps +1ms

When FlexQoS v1.4.3 is running the Tools page reports Archer Enabled, Flow Cache Enabled
Bufferbloat test 
DL 34.3 Mbps +60ms  
UL 16Mbps +30ms
- significant DL bandwidth loss and high UL and DL latency.

All previous versions of the firmware disabled Flow Cache as soon as Adaptive QoS was enabled.

The solution when FlexQoS is enabled is to disabling flow cache via SSH-
```
fc disable
fc flush
```
Bufferbloat test results then show FlexQoS working as expected :  
DL 62.3Mbps +9ms  
UL 15.6Mbps +1ms

dave14305, 2024-06-25:
> @RT-AX58U-B260:/tmp/home/root# tc qdisc This is with qos cake enabled. 

That’s very broken. How about:
```
/tmp/qos stop
/tmp/qos start
cat /etc/cake-qos.conf
```

Nullvoid3771, 2024-06-25:
Enabled QoS cake; router reboot. Disabled runner from ssh; router reboot. 

ASUSWRT-Merlin RT-AX58U_V2 3004.388.7_1-gnuton1 Mon Jun 10 09:50:03 UTC 2024
@RT-AX58U-B260:/tmp/home/root# nvram set runner_disable_force=1
@RT-AX58U-B260:/tmp/home/root# nvram commit
@RT-AX58U-B260:/tmp/home/root#       ##Router reboots 
login as: 
@192.168.50.1's password:

ASUSWRT-Merlin RT-AX58U_V2 3004.388.7_1-gnuton1 Mon Jun 10 09:50:03 UTC 2024
@RT-AX58U-B260:/tmp/home/root# /tmp/qos
@RT-AX58U-B260:/tmp/home/root# /tmp/qos start
@RT-AX58U-B260:/tmp/home/root# cat /etc/cake-qos.conf
#!/bin/sh

ULIF='eth0'
DLIF='eth0'
MIF='ifb4eth0'
ULBW='bandwidth 2355kbit'
DLBW='bandwidth 16384kbit'
OVERHEAD='overhead 18 mpu 64'
FRAMING='atm'
ULPRIOQUEUE='diffserv3'
DLPRIOQUEUE='besteffort'
ULOPTIONS='nat dual-srchost'
DLOPTIONS='nat wash dual-dsthost ingress'
ULPRIOQUEUE="diffserv8"
DLPRIOQUEUE="diffserv8"
ULOPTIONS="triple-isolate nat wash ack-filter "
DLOPTIONS="ingress triple-isolate nat wash ack-filter "

@RT-AX58U-B260:/tmp/home/root# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2
qdisc noqueue 0: dev archer root refcnt 2
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc ingress ffff: dev eth0 parent ffff:fff1 ----------------
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_us_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_ds_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc noqueue 0: dev br0 root refcnt 2
qdisc noqueue 0: dev wgs1 root refcnt 2
qdisc pfifo_fast 0: dev ifb4eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1

Nullvoid3771, 2024-06-25:
Anyways I don't want to mess with their internet too much during the day so I'll follow up with anything you'd like to try later.

Nullvoid3771, 2024-06-25:
Anyways I don't want to mess with their internet too much during the day so I'll follow up with anything you'd like to try later. 

Edit: noticed I didn't type qos stop the first time. Tried again to be accurate. 

@RT-AX58U-B260:/tmp/home/root# /tmp/qos stop
@RT-AX58U-B260:/tmp/home/root# /tmp/qos start
@RT-AX58U-B260:/tmp/home/root# cat /etc/cake-qos.conf
#!/bin/sh

ULIF='eth0'
DLIF='eth0'
MIF='ifb4eth0'
ULBW='bandwidth 2355kbit'
DLBW='bandwidth 16384kbit'
OVERHEAD='overhead 18 mpu 64'
FRAMING='atm'
ULPRIOQUEUE='diffserv3'
DLPRIOQUEUE='besteffort'
ULOPTIONS='nat dual-srchost'
DLOPTIONS='nat wash dual-dsthost ingress'
ULPRIOQUEUE="diffserv8"
DLPRIOQUEUE="diffserv8"
ULOPTIONS="triple-isolate nat wash ack-filter "
DLOPTIONS="ingress triple-isolate nat wash ack-filter "
Joan@RT-AX58U-B260:/tmp/home/root# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2
qdisc noqueue 0: dev archer root refcnt 2
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1                                                                                                                                                                              1 1 1 1 1
qdisc ingress ffff: dev eth0 parent ffff:fff1 ----------------
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1                                                                                                                                                                              1 1 1 1 1
qdisc pfifo_fast 0: dev spu_us_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0                                                                                                                                                                              0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_ds_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0                                                                                                                                                                              0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1                                                                                                                                                                              1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1                                                                                                                                                                              1 1 1 1 1
qdisc noqueue 0: dev br0 root refcnt 2
qdisc noqueue 0: dev wgs1 root refcnt 2
qdisc pfifo_fast 0: dev ifb4eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1                                                                                                                                                                              1 1 1 1 1 1 1

Nullvoid3771, 2024-06-25:
Anyways I don't want to mess with their internet too much during the day so I'll follow up with anything you'd like to try later.

Edit: noticed I didn't type qos stop the first time. Tried again to be accurate.  and redacted the username. 

@RT-AX58U-B260:/tmp/home/root# /tmp/qos stop
@RT-AX58U-B260:/tmp/home/root# /tmp/qos start
@RT-AX58U-B260:/tmp/home/root# cat /etc/cake-qos.conf
#!/bin/sh

ULIF='eth0'
DLIF='eth0'
MIF='ifb4eth0'
ULBW='bandwidth 2355kbit'
DLBW='bandwidth 16384kbit'
OVERHEAD='overhead 18 mpu 64'
FRAMING='atm'
ULPRIOQUEUE='diffserv3'
DLPRIOQUEUE='besteffort'
ULOPTIONS='nat dual-srchost'
DLOPTIONS='nat wash dual-dsthost ingress'
ULPRIOQUEUE="diffserv8"
DLPRIOQUEUE="diffserv8"
ULOPTIONS="triple-isolate nat wash ack-filter "
DLOPTIONS="ingress triple-isolate nat wash ack-filter "
@RT-AX58U-B260:/tmp/home/root# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2
qdisc noqueue 0: dev archer root refcnt 2
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc ingress ffff: dev eth0 parent ffff:fff1 ----------------
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_us_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_ds_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc noqueue 0: dev br0 root refcnt 2
qdisc noqueue 0: dev wgs1 root refcnt 2
qdisc pfifo_fast 0: dev ifb4eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1

dave14305, 2024-06-25:
Next time you want to test:
> @RT-AX58U-B260:/tmp/home/root# /tmp/qos stop
@RT-AX58U-B260:/tmp/home/root# /tmp/qos start 
```
cat /tmp/qos
sh -x /tmp/qos stop
sh -x /tmp/qos start
```
I would stop using the Cake add-on for testing. For Gnuton’s purposes, just need to prove if the default Cake implementation works or not.

Nullvoid3771, 2024-06-26:
Uninstalled CakeQoS webgui as recommended. 

@RT-AX58U-B260:/tmp/home/root# nvram set runner_disable_force=1
@RT-AX58U-B260:/tmp/home/root# nvram commit
#Enabled QoSCake
##Router reboots

@RT-AX58U-B260:/tmp/home/root# cat /tmp/qos
#!/bin/sh
source /etc/cake-qos.conf

case "$1" in
start)
# Upload
        tc qdisc add dev $ULIF root cake $ULPRIOQUEUE $ULBW $OVERHEAD $FRAMING $                                                                                                                                                                             ULOPTIONS 2>/dev/null

# Download
        ip link add name $MIF type ifb 2>/dev/null
        tc qdisc add dev $DLIF handle ffff: ingress 2>/dev/null
        tc qdisc add dev $MIF root cake $DLPRIOQUEUE $DLBW $OVERHEAD $FRAMING $D                                                                                                                                                                             LOPTIONS 2>/dev/null
        ip link set $MIF up 2>/dev/null
        tc filter add dev $DLIF parent ffff: prio 10 matchall action mirred egre                                                                                                                                                                             ss redirect dev $MIF 2>/dev/null

        ;;
stop)
        tc qdisc del dev $ULIF root 2>/dev/null
        tc qdisc del dev $DLIF ingress 2>/dev/null
        tc qdisc del dev $MIF root 2>/dev/null
        ip link set $MIF down 2>/dev/null
        ip link del dev $MIF 2>/dev/null
        ;;
*)
esac

@RT-AX58U-B260:/tmp/home/root# sh -x /tmp/qos stop
+ source /etc/cake-qos.conf
+ ULIF=eth0
+ DLIF=eth0
+ MIF=ifb4eth0
+ ULBW=bandwidth 2355kbit
+ DLBW=bandwidth 16384kbit
+ OVERHEAD=overhead 18 mpu 64
+ FRAMING=atm
+ ULPRIOQUEUE=diffserv3
+ DLPRIOQUEUE=besteffort
+ ULOPTIONS=nat dual-srchost
+ DLOPTIONS=nat wash dual-dsthost ingress
+ tc qdisc del dev eth0 root
+ tc qdisc del dev eth0 ingress
+ tc qdisc del dev ifb4eth0 root
+ ip link set ifb4eth0 down
+ ip link del dev ifb4eth0

@RT-AX58U-B260:/tmp/home/root# sh -x /tmp/qos start
+ source /etc/cake-qos.conf
+ ULIF=eth0
+ DLIF=eth0
+ MIF=ifb4eth0
+ ULBW=bandwidth 2355kbit
+ DLBW=bandwidth 16384kbit
+ OVERHEAD=overhead 18 mpu 64
+ FRAMING=atm
+ ULPRIOQUEUE=diffserv3
+ DLPRIOQUEUE=besteffort
+ ULOPTIONS=nat dual-srchost
+ DLOPTIONS=nat wash dual-dsthost ingress
+ tc qdisc add dev eth0 root cake diffserv3 bandwidth 2355kbit overhead 18 mpu 64 atm nat dual-srchost
+ ip link add name ifb4eth0 type ifb
+ tc qdisc add dev eth0 handle ffff: ingress
+ tc qdisc add dev ifb4eth0 root cake besteffort bandwidth 16384kbit overhead 18 mpu 64 atm nat wash dual-dsthost ingress
+ ip link set ifb4eth0 up
+ tc filter add dev eth0 parent ffff: prio 10 matchall action mirred egress redirect dev ifb4eth0

@RT-AX58U-B260:/tmp/home/root# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2
qdisc noqueue 0: dev archer root refcnt 2
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc ingress ffff: dev eth0 parent ffff:fff1 ----------------
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_us_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_ds_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc noqueue 0: dev br0 root refcnt 2
qdisc noqueue 0: dev wgs1 root refcnt 2
qdisc pfifo_fast 0: dev ifb4eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1

It's 3am will pick this up later.

dave14305, 2024-06-26:
If there are no errors from `tc qdisc add dev eth0 root cake diffserv3 bandwidth 2355kbit overhead 18 mpu 64 atm nat dual-srchost`, yet you still end up with pfifo on eth0, then something is cursed in this firmware.

Nullvoid3771, 2024-06-27:
```
@RT-AX58U-B260:/tmp/home/root# nvram set runner_disable_force=1
@RT-AX58U-B260:/tmp/home/root# nvram commit
@RT-AX58U-B260:/tmp/home/root# reboot

@RT-AX58U-B260:/tmp/home/root# tc qdisc add dev eth0 root cake diffserv3 ban
dwidth 2355kbit overhead 18 mpu 64 atm nat dual-srchost
RTNETLINK answers: No such file or directory
@RT-AX58U-B260:/tmp/home/root# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2
qdisc noqueue 0: dev archer root refcnt 2
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1                                                                                         1 1 1 1 1
qdisc ingress ffff: dev eth0 parent ffff:fff1 ----------------
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1                                                                                         1 1 1 1 1
qdisc pfifo_fast 0: dev spu_us_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0                                                                                         0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_ds_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0                                                                                         0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1                                                                                         1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1                                                                                         1 1 1 1 1
qdisc noqueue 0: dev br0 root refcnt 2
qdisc noqueue 0: dev wgs1 root refcnt 2
qdisc pfifo_fast 0: dev ifb4eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1                                                                                         1 1 1 1 1 1 1

@RT-AX58U-B260:/tmp/home/root# cat /tmp/qos
#!/bin/sh
source /etc/cake-qos.conf

case "$1" in
start)
# Upload
        tc qdisc add dev $ULIF root cake $ULPRIOQUEUE $ULBW $OVERHEAD $FRAMING $ULOPTIONS 2>/dev/null

# Download
        ip link add name $MIF type ifb 2>/dev/null
        tc qdisc add dev $DLIF handle ffff: ingress 2>/dev/null
        tc qdisc add dev $MIF root cake $DLPRIOQUEUE $DLBW $OVERHEAD $FRAMING $DLOPTIONS 2>/dev/null
        ip link set $MIF up 2>/dev/null
        tc filter add dev $DLIF parent ffff: prio 10 matchall action mirred egress redirect dev $MIF 2>/dev/null

        ;;
stop)
        tc qdisc del dev $ULIF root 2>/dev/null
        tc qdisc del dev $DLIF ingress 2>/dev/null
        tc qdisc del dev $MIF root 2>/dev/null
        ip link set $MIF down 2>/dev/null
        ip link del dev $MIF 2>/dev/null
        ;;
*)
esac

@RT-AX58U-B260:/tmp/home/root# sh -x /tmp/qos stop
+ source /etc/cake-qos.conf
+ ULIF=eth0
+ DLIF=eth0
+ MIF=ifb4eth0
+ ULBW=bandwidth 2355kbit
+ DLBW=bandwidth 16384kbit
+ OVERHEAD=overhead 18 mpu 64
+ FRAMING=atm
+ ULPRIOQUEUE=diffserv3
+ DLPRIOQUEUE=besteffort
+ ULOPTIONS=nat dual-srchost
+ DLOPTIONS=nat wash dual-dsthost ingress
+ tc qdisc del dev eth0 root
+ tc qdisc del dev eth0 ingress
+ tc qdisc del dev ifb4eth0 root
+ ip link set ifb4eth0 down
+ ip link del dev ifb4eth0

@RT-AX58U-B260:/tmp/home/root# sh -x /tmp/qos start
+ source /etc/cake-qos.conf
+ ULIF=eth0
+ DLIF=eth0
+ MIF=ifb4eth0
+ ULBW=bandwidth 2355kbit
+ DLBW=bandwidth 16384kbit
+ OVERHEAD=overhead 18 mpu 64
+ FRAMING=atm
+ ULPRIOQUEUE=diffserv3
+ DLPRIOQUEUE=besteffort
+ ULOPTIONS=nat dual-srchost
+ DLOPTIONS=nat wash dual-dsthost ingress
+ tc qdisc add dev eth0 root cake diffserv3 bandwidth 2355kbit overhead 18 mpu 64 atm nat dual-srchost
+ ip link add name ifb4eth0 type ifb
+ tc qdisc add dev eth0 handle ffff: ingress
+ tc qdisc add dev ifb4eth0 root cake besteffort bandwidth 16384kbit overhead 18 mpu 64 atm nat wash dual-dsthost ingress
+ ip link set ifb4eth0 up
+ tc filter add dev eth0 parent ffff: prio 10 matchall action mirred egress redirect dev ifb4eth0

@RT-AX58U-B260:/tmp/home/root# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2
qdisc noqueue 0: dev archer root refcnt 2
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc ingress ffff: dev eth0 parent ffff:fff1 ----------------
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_us_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_ds_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc noqueue 0: dev br0 root refcnt 2
qdisc noqueue 0: dev wgs1 root refcnt 2
qdisc pfifo_fast 0: dev ifb4eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1

@RT-AX58U-B260:/tmp/home/root# tc qdisc add dev eth0 root cake diffserv3 bandwidth 2355kbit overhead 18 mpu 64 atm nat dual-srchost
RTNETLINK answers: No such file or directory
```
`HW acceleration | Archer: Enabled  -  Flow Cache: Disabled `

```
@RT-AX58U-B260:/tmp/home/root# /bin/fc status | grep "HW Acceleration"
        HW Acceleration <Enabled>

```

dave14305, 2024-06-27:
Ok, I missed the /dev/null silencers reading on my iPad. So gnuton must not build CAKE on this router.

Nullvoid3771, 2024-06-27:
> Ok, I missed the /dev/null silencers reading on my iPad. So gnuton must not build CAKE on this router.

That would explain a lot. Thanks dave14305 I appreciate your help identifying the problem.

Nullvoid3771, 2024-07-14:
I went stumbling through the code so see if I could find any markers to determine how or if cake is compiled. So far I’ve realized cake isn’t baked into the kernel because if it’s age so it has to be manually added.

[image]
[image]
[image]

as shown here it seems to be set to compile.

If it’s a supported device seems to be indicated here dependent on the profile?

[image]

Unfortunately I don’t fully understand so that’s as far as I dived into this. I haven’t tried 388.8 beta so if it was simply a build issue it might be present in 388.8, but I’m unsure. Am I on the right path in my assessment?

Nullvoid3771, 2024-07-15:
under release/src/router/rc/qos.c I’m seeing the same null pointers I was getting so maybe it’s not a case of cake not being installed.

[CODE]

#ifdef HND_ROUTER

int start_cake(void)
{
    unsigned int ibw, obw;
    FILE *f;
    char overheadstr[32];
    char ibwstr[32];
    char obwstr[32];
    char *mode;
    char nvnat[16];
    int nat;

    if((f = fopen("/etc/cake-qos.conf", "w")) == NULL) return -2;

    switch (nvram_get_int("qos_atm")) {
        case 0:
            mode = "";
            break;
        case 1:
            mode = "atm";
            break;
        case 2:
            mode = "ptm";
            break;
        default:
            mode = "";
    }

    ibw = strtoul(nvram_safe_get("qos_ibw"), NULL, 10);
    obw = strtoul(nvram_safe_get("qos_obw"), NULL, 10);

    if (ibw == 0)
        *ibwstr = '\0';
    else
        snprintf(ibwstr, sizeof(ibwstr), "bandwidth %dkbit", ibw);

    if (obw == 0)
        *obwstr = '\0';
    else
        snprintf(obwstr, sizeof(obwstr), "bandwidth %dkbit", obw);

    snprintf(overheadstr, sizeof(overheadstr), "overhead %d mpu %d", nvram_get_int("qos_overhead"), nvram_get_int("qos_mpu"));

    const char *wan_ifname = get_wan_ifname(wan_primary_ifunit());

    snprintf(nvnat, sizeof (nvnat), "wan%d_nat_x", wan_primary_ifunit());
    nat = nvram_get_int(nvnat);

    /* Config parameters */
    fprintf(f,
        "#!/bin/sh\n\n"
        "ULIF='%s'\n"
        "DLIF='%s'\n"
        "MIF='ifb4%s'\n"
        "ULBW='%s'\n"
        "DLBW='%s'\n"
        "OVERHEAD='%s'\n"
        "FRAMING='%s'\n"
        "ULPRIOQUEUE='diffserv3'\n"
        "DLPRIOQUEUE='besteffort'\n"
        "ULOPTIONS='%s dual-srchost'\n"
        "DLOPTIONS='%s wash dual-dsthost ingress'\n",

        wan_ifname,
        wan_ifname,
        wan_ifname,
        obwstr,
        ibwstr,
        overheadstr,
        mode,
        (nat ? "nat" : ""),
        (nat ? "nat" : "")
    );

    append_custom_config("cake-qos.conf",f);
    fclose(f);

    if((f = fopen(qosfn, "w")) == NULL) return -2;

    /* Stop/start rules */
    fprintf(f,
        "#!/bin/sh\n"
        "source /etc/cake-qos.conf\n\n"

        "case \"$1\" in\n"
        "start)\n"
        "# Upload\n"
        "\ttc qdisc add dev $ULIF root cake $ULPRIOQUEUE $ULBW $OVERHEAD $FRAMING $ULOPTIONS 2>/dev/null\n\n"

        "# Download\n"
        "\tip link add name $MIF type ifb 2>/dev/null\n"
        "\ttc qdisc add dev $DLIF handle ffff: ingress 2>/dev/null\n"
        "\ttc qdisc add dev $MIF root cake $DLPRIOQUEUE $DLBW $OVERHEAD $FRAMING $DLOPTIONS 2>/dev/null\n"
        "\tip link set $MIF up 2>/dev/null\n"
        "\ttc filter add dev $DLIF parent ffff: prio 10 matchall action mirred egress redirect dev $MIF 2>/dev/null\n\n"

        "\t;;\n"
        "stop)\n"
        "\ttc qdisc del dev $ULIF root 2>/dev/null\n"
        "\ttc qdisc del dev $DLIF ingress 2>/dev/null\n"
        "\ttc qdisc del dev $MIF root 2>/dev/null\n"
        "\tip link set $MIF down 2>/dev/null\n"
        "\tip link del dev $MIF 2>/dev/null\n"
        "\t;;\n"
        "*)\n"
        "esac\n");

    fclose(f);
    chmod("/etc/cake-qos.conf", 0755);
    chmod(qosfn, 0755);
    run_custom_script("qos-start", 120, "init", NULL);
    eval((char *)qosfn, "start");

    return 0;
}
#endif[/CODE]
[CODE]1

@RT-AX58U-B260:/tmp/home/root# cat /tmp/qos
#!/bin/sh
source /etc/cake-qos.conf

case "$1" in
start)
# Upload
        tc qdisc add dev $ULIF root cake $ULPRIOQUEUE $ULBW $OVERHEAD $FRAMING $ULOPTIONS 2>/dev/null

# Download
        ip link add name $MIF type ifb 2>/dev/null
        tc qdisc add dev $DLIF handle ffff: ingress 2>/dev/null
        tc qdisc add dev $MIF root cake $DLPRIOQUEUE $DLBW $OVERHEAD $FRAMING $DLOPTIONS 2>/dev/null
        ip link set $MIF up 2>/dev/null
        tc filter add dev $DLIF parent ffff: prio 10 matchall action mirred egress redirect dev $MIF 2>/dev/null

        ;;
stop)
        tc qdisc del dev $ULIF root 2>/dev/null
        tc qdisc del dev $DLIF ingress 2>/dev/null
        tc qdisc del dev $MIF root 2>/dev/null
        ip link set $MIF down 2>/dev/null
        ip link del dev $MIF 2>/dev/null
        ;;
*)
esac
[/CODE]

Nullvoid3771, 2024-07-15:
under release/src/router/rc/qos.c I’m seeing the same null pointers I was getting so maybe it’s not a case of cake not being installed.

`

#ifdef HND_ROUTER

int start_cake(void)
{
    unsigned int ibw, obw;
    FILE *f;
    char overheadstr[32];
    char ibwstr[32];
    char obwstr[32];
    char *mode;
    char nvnat[16];
    int nat;

    if((f = fopen("/etc/cake-qos.conf", "w")) == NULL) return -2;

    switch (nvram_get_int("qos_atm")) {
        case 0:
            mode = "";
            break;
        case 1:
            mode = "atm";
            break;
        case 2:
            mode = "ptm";
            break;
        default:
            mode = "";
    }

    ibw = strtoul(nvram_safe_get("qos_ibw"), NULL, 10);
    obw = strtoul(nvram_safe_get("qos_obw"), NULL, 10);

    if (ibw == 0)
        *ibwstr = '\0';
    else
        snprintf(ibwstr, sizeof(ibwstr), "bandwidth %dkbit", ibw);

    if (obw == 0)
        *obwstr = '\0';
    else
        snprintf(obwstr, sizeof(obwstr), "bandwidth %dkbit", obw);

    snprintf(overheadstr, sizeof(overheadstr), "overhead %d mpu %d", nvram_get_int("qos_overhead"), nvram_get_int("qos_mpu"));

    const char *wan_ifname = get_wan_ifname(wan_primary_ifunit());

    snprintf(nvnat, sizeof (nvnat), "wan%d_nat_x", wan_primary_ifunit());
    nat = nvram_get_int(nvnat);

    /* Config parameters */
    fprintf(f,
        "#!/bin/sh\n\n"
        "ULIF='%s'\n"
        "DLIF='%s'\n"
        "MIF='ifb4%s'\n"
        "ULBW='%s'\n"
        "DLBW='%s'\n"
        "OVERHEAD='%s'\n"
        "FRAMING='%s'\n"
        "ULPRIOQUEUE='diffserv3'\n"
        "DLPRIOQUEUE='besteffort'\n"
        "ULOPTIONS='%s dual-srchost'\n"
        "DLOPTIONS='%s wash dual-dsthost ingress'\n",

        wan_ifname,
        wan_ifname,
        wan_ifname,
        obwstr,
        ibwstr,
        overheadstr,
        mode,
        (nat ? "nat" : ""),
        (nat ? "nat" : "")
    );

    append_custom_config("cake-qos.conf",f);
    fclose(f);

    if((f = fopen(qosfn, "w")) == NULL) return -2;

    /* Stop/start rules */
    fprintf(f,
        "#!/bin/sh\n"
        "source /etc/cake-qos.conf\n\n"

        "case \"$1\" in\n"
        "start)\n"
        "# Upload\n"
        "\ttc qdisc add dev $ULIF root cake $ULPRIOQUEUE $ULBW $OVERHEAD $FRAMING $ULOPTIONS 2>/dev/null\n\n"

        "# Download\n"
        "\tip link add name $MIF type ifb 2>/dev/null\n"
        "\ttc qdisc add dev $DLIF handle ffff: ingress 2>/dev/null\n"
        "\ttc qdisc add dev $MIF root cake $DLPRIOQUEUE $DLBW $OVERHEAD $FRAMING $DLOPTIONS 2>/dev/null\n"
        "\tip link set $MIF up 2>/dev/null\n"
        "\ttc filter add dev $DLIF parent ffff: prio 10 matchall action mirred egress redirect dev $MIF 2>/dev/null\n\n"

        "\t;;\n"
        "stop)\n"
        "\ttc qdisc del dev $ULIF root 2>/dev/null\n"
        "\ttc qdisc del dev $DLIF ingress 2>/dev/null\n"
        "\ttc qdisc del dev $MIF root 2>/dev/null\n"
        "\tip link set $MIF down 2>/dev/null\n"
        "\tip link del dev $MIF 2>/dev/null\n"
        "\t;;\n"
        "*)\n"
        "esac\n");

    fclose(f);
    chmod("/etc/cake-qos.conf", 0755);
    chmod(qosfn, 0755);
    run_custom_script("qos-start", 120, "init", NULL);
    eval((char *)qosfn, "start");

    return 0;
}
#endif[/CODE]
[CODE]1

@RT-AX58U-B260:/tmp/home/root# cat /tmp/qos
#!/bin/sh
source /etc/cake-qos.conf

case "$1" in
start)
# Upload
        tc qdisc add dev $ULIF root cake $ULPRIOQUEUE $ULBW $OVERHEAD $FRAMING $ULOPTIONS 2>/dev/null

# Download
        ip link add name $MIF type ifb 2>/dev/null
        tc qdisc add dev $DLIF handle ffff: ingress 2>/dev/null
        tc qdisc add dev $MIF root cake $DLPRIOQUEUE $DLBW $OVERHEAD $FRAMING $DLOPTIONS 2>/dev/null
        ip link set $MIF up 2>/dev/null
        tc filter add dev $DLIF parent ffff: prio 10 matchall action mirred egress redirect dev $MIF 2>/dev/null

        ;;
stop)
        tc qdisc del dev $ULIF root 2>/dev/null
        tc qdisc del dev $DLIF ingress 2>/dev/null
        tc qdisc del dev $MIF root 2>/dev/null
        ip link set $MIF down 2>/dev/null
        ip link del dev $MIF 2>/dev/null
        ;;
*)
esac
`

Nullvoid3771, 2024-07-15:
Something seems wrong with tc qdisc and I’m going to chalk it up to my initial concerns that hw acceleration runner would not turn off in 388.7.. will test 388.8 on gnuton tomorrow.

[CODE]@RT-AX58U-B260:/tmp/home/root# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2
qdisc noqueue 0: dev archer root refcnt 2
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc ingress ffff: dev eth0 parent ffff:fff1 ----------------
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_us_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_ds_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc noqueue 0: dev br0 root refcnt 2
qdisc noqueue 0: dev wgs1 root refcnt 2
qdisc pfifo_fast 0: dev ifb4eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1

@RT-AX58U-B260:/tmp/home/root# tc qdisc add dev eth0 root cake diffserv3 bandwidth 2355kbit overhead 18 mpu 64 atm nat dual-srchost
RTNETLINK answers: No such file or directory[/CODE]

Nullvoid3771, 2024-07-15:
Here’s the diff of both config_base.6a.67XX files comparing V1 to V2 of rt-ax58u what jumps out particularly is codel is not set on v2.

# CONFIG_NET_SCH_CODEL is not set

https://www.diffchecker.com/eyFX3IJn/

Nullvoid3771, 2024-07-15:
config.gz file from the router: 

#
# Automatically generated file; DO NOT EDIT.
# Linux/arm 4.19.183 Kernel Configuration
#

#
# Compiler: arm-buildroot-linux-gnueabi-gcc.br_real (Buildroot 2019.11.1) 9.2.0
#
CONFIG_CC_IS_GCC=y
CONFIG_GCC_VERSION=90200
CONFIG_CLANG_VERSION=0
CONFIG_CC_HAS_ASM_GOTO=y
CONFIG_IRQ_WORK=y
CONFIG_BUILDTIME_EXTABLE_SORT=y

#
# General setup
#
CONFIG_INIT_ENV_ARG_LIMIT=32
# CONFIG_COMPILE_TEST is not set
CONFIG_LOCALVERSION=""
# CONFIG_LOCALVERSION_AUTO is not set
CONFIG_BUILD_SALT=""
CONFIG_HAVE_KERNEL_GZIP=y
CONFIG_HAVE_KERNEL_LZMA=y
CONFIG_HAVE_KERNEL_XZ=y
CONFIG_HAVE_KERNEL_LZO=y
CONFIG_HAVE_KERNEL_LZ4=y
CONFIG_KERNEL_GZIP=y
# CONFIG_KERNEL_LZMA is not set
# CONFIG_KERNEL_XZ is not set
# CONFIG_KERNEL_LZO is not set
# CONFIG_KERNEL_LZ4 is not set
CONFIG_DEFAULT_HOSTNAME="(none)"
CONFIG_SWAP=y
CONFIG_SYSVIPC=y
CONFIG_SYSVIPC_SYSCTL=y
# CONFIG_POSIX_MQUEUE is not set
# CONFIG_CROSS_MEMORY_ATTACH is not set
# CONFIG_USELIB is not set
# CONFIG_AUDIT is not set

#
# IRQ subsystem
#
CONFIG_GENERIC_IRQ_PROBE=y
CONFIG_GENERIC_IRQ_SHOW=y
CONFIG_GENERIC_IRQ_SHOW_LEVEL=y
CONFIG_GENERIC_IRQ_EFFECTIVE_AFF_MASK=y
CONFIG_GENERIC_IRQ_MIGRATION=y
CONFIG_HARDIRQS_SW_RESEND=y
CONFIG_IRQ_DOMAIN=y
CONFIG_IRQ_DOMAIN_HIERARCHY=y
CONFIG_GENERIC_MSI_IRQ=y
CONFIG_GENERIC_MSI_IRQ_DOMAIN=y
CONFIG_HANDLE_DOMAIN_IRQ=y
CONFIG_IRQ_FORCED_THREADING=y
CONFIG_SPARSE_IRQ=y
# CONFIG_GENERIC_IRQ_DEBUGFS is not set
CONFIG_GENERIC_IRQ_MULTI_HANDLER=y
CONFIG_ARCH_CLOCKSOURCE_DATA=y
CONFIG_GENERIC_CLOCKEVENTS=y
CONFIG_ARCH_HAS_TICK_BROADCAST=y
CONFIG_GENERIC_CLOCKEVENTS_BROADCAST=y

#
# Timers subsystem
#
[... 3866 lines cut ...]
#
# SOUND
#
# CONFIG_BCM_SOUND is not set
# CONFIG_SND_SOC_MAPLELEAF is not set
# CONFIG_SND_SOC_MAPLETREE is not set
# CONFIG_SND_SOC_CS4345 is not set
# CONFIG_SND_SOC_PCM5100 is not set

#
# SOUND
#
# CONFIG_SND_BCM63xx_I2S_WHISTLER is not set
# CONFIG_SND_BCM63xx_I2S_MERRITT is not set
# CONFIG_SND_BCM63xx_PCM_SQUAMISH is not set
# CONFIG_SND_BCM63xx_SOC_I2S_ENG is not set
# CONFIG_SND_BCM63XX_SOC_I2S_MACH is not set
# CONFIG_SND_BCM63XX_SOC_TDM_MACH is not set
# CONFIG_SND_BCM6856_SOC_I2S is not set
# CONFIG_SND_BCM63148_SOC_I2S is not set
CONFIG_BCM_IN_KERNEL=y

#
# ASUSWRT
#

Nullvoid3771, 2024-07-15:
Also not set in beta 388.8 

"# CONFIG_NET_SCH_CAKE is not set"

gnuton, 2024-07-15:
Fixed 875b94d50c457d28662d526875828cb9c56566e0

Nullvoid3771, 2024-07-15:
Wonderful once the build is released and I've got a chance to test it I'll close or inform you of any other issues. Thanks appreciate it.

gnuton, 2024-07-16:
@Nullvoid3771 if you like you can test a snapshot with changes here https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/9945740113

Nullvoid3771, 2024-07-16:
snapshot of release-rt-ax58u_v2-default (388.8 beta1)

Enabled cake and noticed right off the bat Archer is still stuck on so I think that's the crux of the problem. Need to figure out why archer can't be forced off or automatically switched off when cake is enabled.  

[image]
[image]
![Scre
[image]
enshot 2024-07-16 052643](https://github.com/user-attachments/assets/3f5dca9a-90a3-4fad-8590-431015d9a891)
[image]

config.gz file
[image]

I do notice with cake on tc qdisc shows PFIFO_FAST instead of sfq, htb, or fq_codel maybe a result of cake not working.

Nullvoid3771, 2024-07-16:
[image]

Additionally I notice in the beta1 Mhz for cpu model is showing ???

Nullvoid3771, 2024-07-16:
system logs show this: 

[image]

Nullvoid3771, 2024-07-16:
config of 388.8 beta1

https://pastebin.com/uNbHmy8R

/rom/etc/buildprofile (96756GW)

https://pastebin.com/vHRYUT6R

Nullvoid3771, 2024-07-17:
Here is the relevant code for qos cake I believe. 

release/src/router/rc/qos.c

> #ifdef HND_ROUTER          ##### This might need unhashed?
	else if (IS_CAKE_QOS()) {
		status = start_cake();
	}
#endif

> int start_cake(void)
{
	unsigned int ibw, obw;
	FILE *f;
	char overheadstr[32];
	char ibwstr[32];
	char obwstr[32];
	char *mode;
	char nvnat[16];
	int nat;

	if((f = fopen("/etc/cake-qos.conf", "w")) == NULL) return -2;

	switch (nvram_get_int("qos_atm")) {
		case 0:
			mode = "";
			break;
		case 1:
			mode = "atm";
			break;
		case 2:
			mode = "ptm";
			break;
		default:
			mode = "";
	}

	ibw = strtoul(nvram_safe_get("qos_ibw"), NULL, 10);
	obw = strtoul(nvram_safe_get("qos_obw"), NULL, 10);

	if (ibw == 0)
		*ibwstr = '\0';
	else
		snprintf(ibwstr, sizeof(ibwstr), "bandwidth %dkbit", ibw);

	if (obw == 0)
		*obwstr = '\0';
	else
		snprintf(obwstr, sizeof(obwstr), "bandwidth %dkbit", obw);

	snprintf(overheadstr, sizeof(overheadstr), "overhead %d mpu %d", nvram_get_int("qos_overhead"), nvram_get_int("qos_mpu"));

	const char *wan_ifname = get_wan_ifname(wan_primary_ifunit());

	snprintf(nvnat, sizeof (nvnat), "wan%d_nat_x", wan_primary_ifunit());
	nat = nvram_get_int(nvnat);

	/* Config parameters */
	fprintf(f,
		"#!/bin/sh\n\n"
		"ULIF='%s'\n"
		"DLIF='%s'\n"
		"MIF='ifb4%s'\n"
		"ULBW='%s'\n"
		"DLBW='%s'\n"
		"OVERHEAD='%s'\n"
		"FRAMING='%s'\n"
		"ULPRIOQUEUE='diffserv3'\n"
		"DLPRIOQUEUE='besteffort'\n"
		"ULOPTIONS='%s dual-srchost'\n"
		"DLOPTIONS='%s wash dual-dsthost ingress'\n",

		wan_ifname,
		wan_ifname,
		wan_ifname,
		obwstr,
		ibwstr,
		overheadstr,
		mode,
		(nat ? "nat" : ""),
		(nat ? "nat" : "")
	);

	append_custom_config("cake-qos.conf",f);
	fclose(f);

	if((f = fopen(qosfn, "w")) == NULL) return -2;

	/* Stop/start rules */
	fprintf(f,
		"#!/bin/sh\n"
		"source /etc/cake-qos.conf\n\n"

		"case \"$1\" in\n"
		"start)\n"
		"# Upload\n"
		"\ttc qdisc add dev $ULIF root cake $ULPRIOQUEUE $ULBW $OVERHEAD $FRAMING $ULOPTIONS 2>/dev/null\n\n"

		"# Download\n"
		"\tip link add name $MIF type ifb 2>/dev/null\n"
		"\ttc qdisc add dev $DLIF handle ffff: ingress 2>/dev/null\n"
		"\ttc qdisc add dev $MIF root cake $DLPRIOQUEUE $DLBW $OVERHEAD $FRAMING $DLOPTIONS 2>/dev/null\n"
		"\tip link set $MIF up 2>/dev/null\n"
		"\ttc filter add dev $DLIF parent ffff: prio 10 matchall action mirred egress redirect dev $MIF 2>/dev/null\n\n"

		"\t;;\n"
		"stop)\n"
		"\ttc qdisc del dev $ULIF root 2>/dev/null\n"
		"\ttc qdisc del dev $DLIF ingress 2>/dev/null\n"
		"\ttc qdisc del dev $MIF root 2>/dev/null\n"
		"\tip link set $MIF down 2>/dev/null\n"
		"\tip link del dev $MIF 2>/dev/null\n"
		"\t;;\n"
		"*)\n"
		"esac\n");

	fclose(f);
	chmod("/etc/cake-qos.conf", 0755);
	chmod(qosfn, 0755);
	run_custom_script("qos-start", 120, "init", NULL);
	eval((char *)qosfn, "start");

	return 0;
}
#endif

release/src/router/www/QoS_EZQoS.asp

> }
	else if(value == 9){		//Cake QoS
		document.getElementById('int_type').checked = false;
		document.getElementById('trad_type').checked = false;
		document.getElementById('bw_limit_type').checked = false;
		document.getElementById('cake_type').checked = true;
		if(geforceNow_support)
			document.getElementById('GeForce_type').checked = false;
		document.getElementById('bandwidth_setting_tr').style.display = "";
		show_up_down(1);
		document.getElementById('list_table').style.display = "none";
		if (cake_support) {
			document.getElementById('qos_overhead_tr').style.display = "";
			document.getElementById('qos_mpu').style.display = "";
			document.getElementById('qos_mpu_label').style.display = "";
		}
		if(document.form.qos_type_orig.value == 0 && document.form.qos_enable_orig.value != 0){
			document.form.action_script.value = "restart_qos;restart_firewall";
		}
		else{
			document.form.action_script.value = "reboot";
			document.form.next_page.value = "QoS_EZQoS.asp";
			document.form.action_wait.value = "<% get_default_reboot_time(); %>";
		}
		if(document.getElementById("auto").checked){
			show_up_down(0);
		}
		else{
			show_up_down(1);
		}
		show_settings("NonAdaptive");
	}
	else {
		alert("no such qos");
	}

Nullvoid3771, 2024-07-18:
> @gnuton might need to add `RTCONFIG_HND_ROUTER_AX_6756` to these `if`s.
> 
> https://github.com/gnuton/asuswrt-merlin.ng/blob/875b94d50c457d28662d526875828cb9c56566e0/release/src/router/rc/sysdeps/init-broadcom.c#L12156-L12166

I think that file might be complied as a .o object file. Under release/src/router/rc/prebuild/RT-AX58U_V2

The prebuild for rt-ax58u_v2 doesn’t exist under Merlin directly and as far as I can tell init-broadcom.c is a template file. Since .o files are not human readable I can’t say for certain.

Nullvoid3771, 2024-07-18:
@dave14305 Guess not? Sorry I'm spit balling here.

Nullvoid3771, 2024-07-19:
[image]

https://github.com/RMerl/asuswrt-merlin.ng/blob/f3b306fd8d9e05e3ad12c2be4cbaf2f307bcc499/release/src-rt-5.02axhnd.675x/kernel/linux-4.1/config_base.6a.6750#L3003

Least I’ve got plenty of good eyes on this.

gnuton, 2024-07-26:
@Nullvoid3771 can you please specify the version of your router?

Nullvoid3771, 2024-07-26:
> @Nullvoid3771 can you please specify the version of your router? @dave14305 thanks for the suggestion I checekd the target.mak and indeed RT-AX58U V2 seems to required RTCONFIG_HND_ROUTER_AX_6756. Fix [47eb98a](https://github.com/gnuton/asuswrt-merlin.ng/commit/47eb98ab31f8e080cb6fe6da7608e7ddcd16c31a)
> 
> Test images https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/10106108189

I own a rt-ax58u v2 I use remotely at my fathers house, and I own a gt-ax11000 and a rt-ax58u v1. So I can test between rt-ax58u v1&v2 

In the context of this topic I’m only working on the rt-ax58u v2 because the v1 works fine.

codokie, 2024-07-28:
Hey I know this issue is about AX58U V2 but with the **AX82U V2** I also can't get CAKE to work (see #496, still occurs with 3004.388.7_1-gnuton1).
The config.gz shows
```
# CONFIG_NET_SCH_CAKE is not set
```
so I guess CAKE is not actually built into the firmware?
Is there any other log/command output I should provide to help identify the cause?

Nullvoid3771, 2024-07-30:
[image]
[image]

@RT-AX58U-B260:/tmp/home/root# cat /tmp/qos
#!/bin/sh
source /etc/cake-qos.conf

case "$1" in
start)
# Upload
        tc qdisc add dev $ULIF root cake $ULPRIOQUEUE $ULBW $OVERHEAD $FRAMING $ULOPTIONS 2>/dev/null

# Download
        ip link add name $MIF type ifb 2>/dev/null
        tc qdisc add dev $DLIF handle ffff: ingress 2>/dev/null
        tc qdisc add dev $MIF root cake $DLPRIOQUEUE $DLBW $OVERHEAD $FRAMING $DLOPTIONS 2>/dev/null
        ip link set $MIF up 2>/dev/null
        tc filter add dev $DLIF parent ffff: prio 10 matchall action mirred egress redirect dev $MIF 2>/dev/null

        ;;
stop)
        tc qdisc del dev $ULIF root 2>/dev/null
        tc qdisc del dev $DLIF ingress 2>/dev/null
        tc qdisc del dev $MIF root 2>/dev/null
        ip link set $MIF down 2>/dev/null
        ip link del dev $MIF 2>/dev/null
        ;;
*)
esac
@RT-AX58U-B260:/tmp/home/root# sh -x /tmp/qos stop
+ source /etc/cake-qos.conf
+ ULIF=eth0
+ DLIF=eth0
+ MIF=ifb4eth0
+ ULBW=bandwidth 2355kbit
+ DLBW=bandwidth 16384kbit
+ OVERHEAD=overhead 18 mpu 64
+ FRAMING=atm
+ ULPRIOQUEUE=diffserv3
+ DLPRIOQUEUE=besteffort
+ ULOPTIONS=nat dual-srchost
+ DLOPTIONS=nat wash dual-dsthost ingress
+ tc qdisc del dev eth0 root
+ tc qdisc del dev eth0 ingress
+ tc qdisc del dev ifb4eth0 root
+ ip link set ifb4eth0 down
+ ip link del dev ifb4eth0
@RT-AX58U-B260:/tmp/home/root# sh -x /tmp/qos start
+ source /etc/cake-qos.conf
+ ULIF=eth0
+ DLIF=eth0
+ MIF=ifb4eth0
+ ULBW=bandwidth 2355kbit
+ DLBW=bandwidth 16384kbit
+ OVERHEAD=overhead 18 mpu 64
+ FRAMING=atm
+ ULPRIOQUEUE=diffserv3
+ DLPRIOQUEUE=besteffort
+ ULOPTIONS=nat dual-srchost
+ DLOPTIONS=nat wash dual-dsthost ingress
+ tc qdisc add dev eth0 root cake diffserv3 bandwidth 2355kbit overhead 18 mpu 64 atm nat dual-srchost
+ ip link add name ifb4eth0 type ifb
+ tc qdisc add dev eth0 handle ffff: ingress
+ tc qdisc add dev ifb4eth0 root cake besteffort bandwidth 16384kbit overhead 18 mpu 64 atm nat wash dual-dsthost ingress
+ ip link set ifb4eth0 up
+ tc filter add dev eth0 parent ffff: prio 10 matchall action mirred egress redirect dev ifb4eth0
@RT-AX58U-B260:/tmp/home/root# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2
qdisc noqueue 0: dev archer root refcnt 2
qdisc cake 8005: dev eth0 root refcnt 2 bandwidth 2355Kbit diffserv3 dual-srchost nat nowash no-ack-filter split-gso rtt 100ms atm overhead 18 mpu 64
qdisc ingress ffff: dev eth0 parent ffff:fff1 ----------------
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_us_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev spu_ds_dummy root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc noqueue 0: dev br0 root refcnt 2
qdisc noqueue 0: dev wgs1 root refcnt 2
qdisc cake 8006: dev ifb4eth0 root refcnt 2 bandwidth 16384Kbit besteffort dual-dsthost nat wash ingress no-ack-filter split-gso rtt 100ms atm overhead 18 mpu 64
@RT-AX58U-B260:/tmp/home/root#

Summery: 

-No errors in syslogs
-Still showing pfifo_fast which I believe is the wrong qdisc
-Hardware Acceleration Archer is now off. 👍 
[image]
-Cpu speed still shows ???

Cake should be enabled and running without error, and Hardware Acceleration is off. However from seeing the tc qdisc I don't believe it's actually working.

Nullvoid3771, 2024-08-02:
[image]

Seems to be working although I'm not super impressed with how well cake is handling it. 

Their isp plan is around 18 down 3 up on cable internet (just a slow plan not rural) anyways it doesn't seem to improve the bufferbloat even lowering it down to 5 down 2 up it still shows the speedtest hitting 17 down so I'm not sure if it's working. Maybe someone else running a test can confirm it.

codokie, 2024-08-02:
This is the kind of result I get when with cake on an AX82U v2. Obviously it is not working.
My ISP speed is about 90 down 4 up and VoIP cuts off when a download is ongoing.
The only way I managed to get grade A+ and no complaints in the house is by using a traditional QOS that favors web surfing and throttles downloads by about 50% (!)
When you set a speed limit it should be reflected in speed tests, that is how it works with traditional QOS at least.

Nullvoid3771, 2024-08-02:
> This is the kind of result I get when with cake on an AX82U v2. Obviously it is not working.
> My ISP speed is about 90 down 4 up and VoIP cuts off when a download is ongoing.
> The only way I managed [...]

I haven’t done any testing on the ax82u so is possible it might be having the same issues as the rt-ax58u v2. The current snapshot of 388.8 beta 1 has significant changes to get it functioning. Although in the 388.8 beta I’m not sure if the ax82u has the changes yet. Specifically looking to see if people are experiencing the same kind of bufferbloat issues unable to get it adequately setup to a A or A+ score. If it can get working on one device then more devices should be easier to fix.

Nullvoid3771, 2024-08-02:
I might bring my rt-ax58u v1 to the location to see if it handles Qos cake better and if their is a noticeable difference.

Yakumounr, 2024-08-05:
> Although in the 388.8 beta I’m not sure if the ax82u has the changes yet

Cake has worked on the DSL-AX82U **v1** with Gnuton's stable builds since 386.03_2-gnuton1 which was the first release including fixes Gnuton had to add specifically to enable it's support, A+ bufferbloat tests only with it enabled. It has worked through all stable builds since for the v1 up to and including 3004.388.7_1-gnuton1, I have yet to try 388.8 yet as I'm waiting for at least a beta version before considering flashing.

The v2 is a different matter, I'm just affirming that there is no issue with the v1.

codokie, 2024-08-07:
Upon upgrading to 3004.388.8_2-gnuton1, cake still doesn't work on AX**82**U v2 so I guess it's not a "5.04axhnd 6750 or 6756" device.
I suspect the V2 of AX82U suffers from the same lack of cake support that has/had been present in the V2 of the AX58U model.
I apologize for jumping on this issue with a different model, it's just that #496 has not progressed and I really need a better QOS

Nullvoid3771, 2024-08-07:
Closing issue. Thank you everyone and gnuton that has helped in bringing cake support to the ax58u. I understand there might be other models that have similar issues with cake. Start a new bug report per model if that is the case as it’s likely going to involve a lot of investigation.

That concludes this bug report if you have further cake related issues on the rt-ax58u make a new report. Thanks again.

closed 2024-08-07 by Nullvoid3771

codokie, 2024-08-07:
Did the latest build really fix cake for you? because it did not seem like it was working when you were doing the bufferbloat test last week. I think you should be able to control cake with the [cakeqos-merlin](https://github.com/ttgapers/cakeqos-merlin) addon (if cake is indeed running).
I've already posted a bug report about 9 months ago, not sure why this one got a lot more attention.
But I'm happy they fixed it for you, hopefully it is worth the hype when compared to the traditional QOS.

Nullvoid3771, 2024-08-07:
> Did the latest build really fix cake for you? because it did not seem like it was working when you were doing the bufferbloat test last week. I think you should be able to control cake with the [...]

It took some time playing with the settings it’s “good enough” I needed it to prioritize a port for voip and it does that now. Before cake wasn’t even running. The bufferbloat results I did get better to a B, but well I do think there might be an issue there I’ll let others draw attention to it.