gnuton/asuswrt-merlin.ng #616: IPv6 firewall test has mostly RFSD results when it should be stealth

Yakumounr, 2024-06-24:
**Router Model Affected**
Models: DSL-AX82U

**Firmware Version Affected**
3004.388.7_1-gnuton1

**Is this bug present in upstream Merlin releases too?**
Unknown

**Describe the bug**
dsl-ax82u does superbly on the grc.com Shields up ipv4 firewall test with all common and service ports reported as stealthed.

The ipv6 firewall test https://ipv6.chappell-family.com/ipv6tcptest/ does not do so well, nothing is open but most are RFSD, with only seven ports from the standard scan coming up stealthed.

I am currently having to use ipv6 passthrough mode, but an earlier forum post shows an  AX86U in ipv6 passthrough on Merlin 386.7_2 returning every port as stealth https://www.snbforums.com/threads/ipv6-question.81520/post-800988 so something does seems to be amiss.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable ipv6 (passthrough, others should be tested if possible)
2. run the port scan tests on https://ipv6.chappell-family.com/ipv6tcptest/
3. If results are not all or mostly stealth there is probably an issue.

**Expected behavior**
scan result tests should all be stealth.

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

RMerl, 2024-07-05:
IPv6 is routed, not NAT'ed.  Double check what IP address you are testing, there's a good chance you are testing your client's IPv6, not your router's.  Which means the firewall for you to check is the one running on your client.

Yakumounr, 2024-07-06:
> IPv6 is routed, not NAT'ed. Double check what IP address you are testing, there's a good chance you are testing your client's IPv6, not your router's. Which means the firewall for you to check is the one running on your client.

Sorry it was so long before I saw the reply. I had checked the IP didn't match my machines IP at the time, but I had not known the windows settings -> network status showed Ipv6, Local IPv6 and basically everything *except* Temp IPv6,  the test was going to the Temp IPv6 and checking ipconfig now revealed the match.