gnuton/asuswrt-merlin.ng #617: DoT breaks DNS on GT-BE98

Adamm00, 2024-06-25:
**Router Model Affected**
Models: GT-BE98

**Firmware Version Affected**
3006.102.1_beta1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
Enabling DoT on the GT-BE98 on 3006.102.1_beta1 causes DNS to stop resolving on clients until DoT is disabled and router is rebooted. I wasn't sure whether to post an issue as the firmware is not officially released but I figure bug reports are still helpful.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable DoT
2. Try resolve DNS on any clients

**Expected behavior**
Stubby to work as normal.

**Screenshots**

Nslookup on a windows client after enabling DoT

> C:\Users\Adamm>nslookup adam.com
> Server:  GT-BE98-38A4
> Address:  192.168.1.1
> 
> DNS request timed out.
>     timeout was 2 seconds.
> DNS request timed out.
>     timeout was 2 seconds.
> DNS request timed out.
>     timeout was 2 seconds.
> DNS request timed out.
>     timeout was 2 seconds.
> *** Request to GT-BE98-38A4 timed-out

The router can still directly resolve DNS though;

> skynet@GT-BE98-38A4:/tmp/home/root# nslookup adam.com
> Server:    1.1.1.1
> Address 1: 1.1.1.1 one.one.one.one
> 
> Name:      adam.com
> Address 1: 20.110.222.158

Syslog output after enabling DoT;

> Jun 25 17:44:02 pppd[5024]: Hangup (SIGHUP)
> Jun 25 17:44:02 pppd[5024]: Terminating on signal 15
> Jun 25 17:44:02 pppd[5024]: Connect time 1333.0 minutes.
> Jun 25 17:44:02 pppd[5024]: Sent 3385129152 bytes, received 3069596647 bytes.
> Jun 25 17:44:02 lldpd[4275]: removal request for address of 220.233.1.54%30, but no knowledge of it
> Jun 25 17:44:02 pppd[5024]: Connection terminated.
> Jun 25 17:44:02 kernel: Freeing devid[2078] for ppp0
> Jun 25 17:44:02 pppd[5024]: Sent PADT
> Jun 25 17:44:02 pppd[5024]: Exit.
> Jun 25 17:44:02 lldpd[4275]: removal request for address of 169.254.10.11%19, but no knowledge of it
> Jun 25 17:44:04 pppd[16853]: Plugin rp-pppoe.so loaded.
> Jun 25 17:44:04 pppd[16853]: RP-PPPoE plugin version 3.11 compiled against pppd 2.4.7
> Jun 25 17:44:04 pppd[16855]: pppd 2.4.7 started by skynet, uid 0
> Jun 25 17:44:04 custom_script: Running /jffs/scripts/service-event (args: restart stubby)
> Jun 25 17:44:04 pppd[16855]: PPP session is 44224 (0xacc0)
> Jun 25 17:44:04 pppd[16855]: Connected to 88:1d:fc:68:3a:00 via interface eth0
> Jun 25 17:44:04 kernel: Assigning idx[35] for ppp0
> Jun 25 17:44:04 pppd[16855]: Using interface ppp0
> Jun 25 17:44:04 pppd[16855]: Connect: ppp0 <--> eth0
> Jun 25 17:44:15 services: apply rules error(21849)
> Jun 25 17:44:16 custom_script: Running /jffs/scripts/firewall-start (args: )
[... 134 lines cut ...]
> Jun 25 17:44:56 ovpn-server1[17737]: xxx.xxx.xxx.xxx:39205 peer info: IV_COMP_STUBv2=1
> Jun 25 17:44:56 ovpn-server1[17737]: xxx.xxx.xxx.xxx:39205 PLUGIN_CALL: POST /usr/lib/openvpn-plugin-auth-pam.so/PLUGIN_AUTH_USER_PASS_VERIFY status=0
> Jun 25 17:44:56 ovpn-server1[17737]: xxx.xxx.xxx.xxx:39205 TLS: Username/Password authentication succeeded for username 'skynet' 
> Jun 25 17:44:56 ovpn-server1[17737]: xxx.xxx.xxx.xxx:39205 TLS: move_session: dest=TM_ACTIVE src=TM_INITIAL reinit_src=1
> Jun 25 17:44:56 ovpn-server1[17737]: xxx.xxx.xxx.xxx:39205 TLS: tls_multi_process: initial untrusted session promoted to trusted
> Jun 25 17:44:56 ovpn-server1[17737]: xxx.xxx.xxx.xxx:39205 Control Channel: TLSv1.3, cipher TLSv1.3 TLS_CHACHA20_POLY1305_SHA256, peer certificate: 2048 bits RSA, signature: RSA-SHA256, peer temporary key: 253 bits X25519
> Jun 25 17:44:56 ovpn-server1[17737]: xxx.xxx.xxx.xxx:39205 [client] Peer Connection Initiated with [AF_INET]xxx.xxx.xxx.xxx:39205 (via [AF_INET]xx.xx.xx.xx%ppp0)
> Jun 25 17:44:56 ovpn-server1[17737]: client/xxx.xxx.xxx.xxx:39205 MULTI_sva: pool returned IPv4=10.8.0.2, IPv6=(Not enabled)
> Jun 25 17:44:56 ovpn-server1[17737]: client/xxx.xxx.xxx.xxx:39205 MULTI: Learn: 10.8.0.2 -> client/xxx.xxx.xxx.xxx:39205
> Jun 25 17:44:56 ovpn-server1[17737]: client/xxx.xxx.xxx.xxx:39205 MULTI: primary virtual IP for client/xxx.xxx.xxx.xxx:39205: 10.8.0.2
> Jun 25 17:44:56 ovpn-server1[17737]: client/xxx.xxx.xxx.xxx:39205 SENT CONTROL [client]: 'PUSH_REPLY,route 192.168.1.0 255.255.255.0 vpn_gateway 500,dhcp-option DNS 192.168.1.1,redirect-gateway def1,route-gateway 10.8.0.1,topology subnet,ping 15,ping-restart 60,ifconfig 10.8.0.2 255.255.255.0,peer-id 0,cipher AES-256-GCM,protocol-flags cc-exit tls-ekm dyn-tls-crypt,tun-mtu 1500' (status=1)
> Jun 25 17:44:57 ovpn-server1[17737]: client/xxx.xxx.xxx.xxx:39205 Data Channel: cipher 'AES-256-GCM', peer-id: 0
> Jun 25 17:44:57 ovpn-server1[17737]: client/xxx.xxx.xxx.xxx:39205 Timers: ping 15, ping-restart 120
> Jun 25 17:44:57 ovpn-server1[17737]: client/xxx.xxx.xxx.xxx:39205 Protocol options: protocol-flags cc-exit tls-ekm dyn-tls-crypt
> Jun 25 17:45:20 dropbear[20020]: Child connection from 192.168.1.150:54053
> Jun 25 17:45:20 dropbear[20020]: Pubkey auth succeeded for 'skynet' with ssh-rsa key SHA256:xxxxxxxxxxxxxxxxxxxxxxxx from 192.168.1.150:54053
> Jun 25 17:45:21 hostapd: wl2: STA c4:c1:7d:a2:cf:cd IEEE 802.11: disassociated
> Jun 25 17:45:21 wlceventd: wlceventd_proc_event(645): wl2: Deauth_ind C4:C1:7D:A2:CF:CD, status: 0, reason: Station requesting (re)association is not authenticated with responding station (9), rssi:-51
> Jun 25 17:45:21 wlceventd: wlceventd_proc_event(662): wl2: Disassoc C4:C1:7D:A2:CF:CD, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:-51
> Jun 25 17:45:21 hostapd: wl2: STA c4:c1:7d:a2:cf:cd IEEE 802.11: disassociated
> Jun 25 17:45:22 wlceventd: wlceventd_proc_event(685): wl2: Auth C4:C1:7D:A2:CF:CD, status: Successful (0), rssi:0
> Jun 25 17:45:22 hostapd: wl2: STA c4:c1:7d:a2:cf:cd IEEE 802.11: associated (aid 1)
> Jun 25 17:45:22 wlceventd: wlceventd_proc_event(722): wl2: Assoc C4:C1:7D:A2:CF:CD, status: Successful (0), rssi:-52
> Jun 25 17:45:22 hostapd: wl2: STA c4:c1:7d:a2:cf:cd RADIUS: starting accounting session 12D79CE761D47CE0
> Jun 25 17:45:22 hostapd: wl2: STA c4:c1:7d:a2:cf:cd WPA: pairwise key handshake completed (RSN)

Adamm00, 2024-06-25:
> I would like to test things but have no firmware. Bummer.

https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/9636521268

decoderman, 2024-06-26:
Didn't realize I was logged out in the browser I checked it with.
Go time!

gnuton, 2024-07-01:
has anybody debugged what's happening with stubby here?

kanuki1984, 2024-07-02:
> has anybody debugged what's happening with stubby here?

Nope.. May I know that asuswrt-merlin 3006.102.1_beta1 based
on which asuswrt version?

eg; 3.0.0.6.102_xxxxx

gnuton, 2024-07-03:
@kanuki1984 that's based on 3.0.0.6.102.34369

kanuki1984, 2024-07-03:
> @kanuki1984 that's based on 3.0.0.6.102.34369

ASUS GT-BE98 Firmware version 3.0.0.6.102_34372
- Addressed DNS errors affecting internet access for Guest Network Pro and VPN connections.

@gnuton
Do you think this fix being included in version 3.0.0.6.102.34369?

I had the weird problem where the router is disconnected from the internet
but received WAN IP after reboot.

Turning the WAN off and on from the GUI will resolve the issue until the next reboot.

I've already reverted back to the stock firmware for now.

Adamm00, 2024-07-09:
> has anybody debugged what's happening with stubby here?

If you want the output of anything in particular let me know.

mattheweli, 2024-07-18:
I can't recall where I saw this, but it seems this issue could be fixed by enabling the default WiFi on Guest Network Pro

gnuton, 2024-07-27:
@kanuki1984 no I do not think so.

Adamm00, 2024-07-28:
@decoderman have you had any luck reproducing this?

Adamm00, 2024-08-18:
@gnuton Anything you would like me to try? Would be great to see this one resolved as it’s the only real feature missing from the GT-98

RMerl, 2024-10-01:
@Adamm00 Try renabling the IoT Guest Network (or any other Guest Network that uses VLAN 0).

Adamm00, 2024-10-01:
> @Adamm00 Try re-enabling the IoT Guest Network (or any other Guest Network that uses VLAN 0 and that is disabled).

Thanks that did the trick, somehow overlooked @mattheweli 's comment a few months ago. Hopefully this gets resolved in future releases.

RMerl, 2024-10-01:
It's a bug in the SDN code.  When you ahve two SDN sharing the same VLAN and one is disabled, the disabled one will cause the running stubby to be killed, and not restarted.

I already devised a fix, but I'm waiting to see if Asus' engineers have a better fix, or if mine would do the trick.

RMerl, 2024-10-06:
I haven't heard back yet from Asus, so I'm going ahead with my own fix, and will revert it if needed if they provide a different fix later.

https://github.com/RMerl/asuswrt-merlin.ng/commit/c49cabfbacdf3ff5575318b2bfeb39477eae13f4

RMerl, 2024-10-20:
The initial fix took care of a scenario where all SDNs were getting restarted.

This new fix will also take care of a scenario when only the IoT SDN itself gets restarted (for instance when you disable IoT):

https://github.com/RMerl/asuswrt-merlin.ng/commit/696e35bb4ece5de6531d07d24fef739330c86baf

Once again caused by the fact that the Main LAN and the IoT SDN both share idx=0, making it impossible for the router to distinguish what to stop/disable when stopping/disabling just the IoT SDN itself.

RMerl, 2024-10-22:
I've been exchanging with an Asus engineer these past few days over this issue.  He sent me a new patch that addresses both cases at a lower level than my two patches/workarounds.  I'll test it later this week and will keep you informed.