gnuton/asuswrt-merlin.ng #618: Dsl AX82U için CakeQos

SametKoyak, 2024-07-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Nullvoid3771, 2024-07-01:
Is this a request for cake on DSL-AX82U?

gnuton, 2024-07-01:
please write in english

closed 2024-07-01 by gnuton

SametKoyak, 2024-07-01:
So sorry I forgot to write in English. I need help with Cake Qos. I have AsusMerlin in DSL AX82U modem/router.

Nullvoid3771, 2024-07-02:
I would suggest you start a discussion at https://www.snbforums.com/ if it’s technical support you need. As it stands we don’t know what you require help with or if you’re experiencing a bug or a lack of a feature such as cake and are requesting cake be added. Apologies as I only speak English and your translation is in broken English.