gnuton/asuswrt-merlin.ng #623: 3004.388.8_0-gnuton0_alpha1 no good on AX-3000

Ichimikichiki, 2024-07-07:
**Router Model Affected**
TUF-AX3000

**Firmware Version Affected**
3004.388.8_0-gnuton0_alpha1

So it allows me to ping from within the router. But won't allow me to connect to internet via any connected devices.

Unsure of relevance, but I stumbled upon this file going through the logs
```
cat /jffs/ahs/ahs_jffs.log 
May  5 05:06:33 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 05:06:33 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton0_beta1]
.
.
.
May  5 15:05:15 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:15 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton0_beta1]
May  5 15:05:25 2018 [process_json:(1784)]ntp_ready false
May  5 15:05:15 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:15 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton0_beta1]
May  5 15:05:25 2018 [process_json:(1784)]ntp_ready false
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton0_beta1]
May  5 15:05:15 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:15 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton0_beta1]
May  5 15:05:55 2018 [process_json:(1784)]ntp_ready false
May  5 15:05:15 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:15 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton0_beta1]
May  5 15:05:25 2018 [process_json:(1784)]ntp_ready false
May  5 15:05:15 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:15 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton0_beta1]
.
.
May  5 15:05:15 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:15 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton0_beta1]
May  5 15:05:15 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:15 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton0_beta1]
May  5 15:05:55 2018 [process_json:(1784)]ntp_ready false
May  5 15:06:25 2018 [process_json:(1784)]ntp_ready false
.
.
.
May  5 15:52:56 2018 [process_json:(1784)]ntp_ready false
May  5 15:53:26 2018 [process_json:(1784)]ntp_ready false
May  5 15:53:56 2018 [process_json:(1784)]ntp_ready false
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton0_beta1]
May  5 15:05:56 2018 [process_json:(1784)]ntp_ready false
May  5 15:06:26 2018 [process_json:(1784)]ntp_ready false
May  5 15:06:56 2018 [process_json:(1784)]ntp_ready false
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton0_beta1]
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:15 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:15 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:26 2018 [process_json:(1784)]ntp_ready false
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
.
.
.
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:56 2018 [process_json:(1784)]ntp_ready false
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:26 2018 [process_json:(1784)]ntp_ready false
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:26 2018 [process_json:(1784)]ntp_ready false
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
.
.
.
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.7_1-gnuton1]
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.8_0-gnuton0_alpha1]
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.8_0-gnuton0_alpha1]
May  5 15:05:16 2018 [compare_firmware_version:(539)]Failed to parse target!!
May  5 15:05:16 2018 [read_json:(1437)]Not match error::target_fwver=[3.0.0.4_388.8_0-gnuton0_alpha1]

```
Had to roll back to get it working.

Nullvoid3771, 2024-07-07:
That error can be ignored. It seems like it’s detecting their is a released firmware for gnuton, but since what your running is a non released branch of firmware it doesn’t understand. 

As to why you can only ping. Maybe check your dns.

Ichimikichiki, 2024-07-07:
DNS is set to the router. Worked immediately when I rolled back. No settings changed.

gnuton, 2024-07-11:
Thanks for checking the Alpha.
I tried it and it works for me. You can see me pinging google.it while connected to the TUF unit.
Do you have anyscript running?

[image]

closed 2024-07-13 by gnuton

Ichimikichiki, 2024-07-15:
> Thanks for checking the Alpha. I tried it and it works for me. You can see me pinging google.it while connected to the TUF unit. I do not even get the error in the logs you get. Do you have anyscript running?

Sure you can. But can you ping google.it from your PC? That's the issue.