gnuton/asuswrt-merlin.ng #625: The VPN -- Others GUI is bugged

Jacky10X, 2024-07-08:
**Router Model Affected**
Models: TUF-AX3000

**Firmware Version Affected**
3004.388.8_0-gnuton0_alpha1, 3004.388.7_1-gnuton1

**Is this bug present in upstream Merlin releases too?**
Unknown

**Describe the bug**
The VPN -- Others GUI is bugged. 
TUF theme and stock theme both have this issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to VPN
2. Click on Others
3. GUI is bugged.

**Expected behavior**
The GUI should be normal

**Screenshots**

[image]

[image]

gnuton, 2024-07-10:
this problem is due to  your browser cache.

Jacky10X, 2024-07-10:
@gnuton That is weird. I have cleaned the browser cache (edge) and the problem is persistent. I also install Firefox and the GUI is still bugged.

gnuton, 2024-07-11:
I will give it a look. Is this happening only on that page?

Jacky10X, 2024-07-11:
> I will give it a look. Is this happening only on that page?

Yes, only on that page.

Jacky10X, 2024-07-12:
@gnuton 
Sorry to bother you 
I just found the VPN Client --> PP2P/L2TP also has a similar issue. (Stock Theme using TUF UI).

[image]

gnuton, 2024-07-12:
Tested and it works for me. 
I cannot reproduce it.

closed 2024-07-12 by gnuton

Jacky10X, 2024-07-12:
I guess it's my browser problem while I am using the TUF theme firmware. But the Stock Theme has some UI bugs. Such as using the TUF UI button. Maybe it is related to Stock firmware? By the way, thanks for checking it.

gnuton, 2024-07-13:
I have closed this, but if you need more help feel free to keep wriiting here.
I can reopen it if we manage to understand if this can be reproduced. But as far as I see this is not a bug for now.
3004.388.7_1-gnuton1 is a stable release and no body has complained abou it either.. unlikely this is a problem of the firmware

If you browse tuf and not tuf UI with same browser the browser caches the CSS and this generates artifact as you have noticed (bad renders, wrong ui elements...)

TUF UI theme has some small bugs, but nothing major.