gnuton/asuswrt-merlin.ng #627: 3006.102.1_1-gnuton0_beta1 Firewall - URL Filter broken - Hitting Apply doesn't save entries

CaveWaverider, 2024-07-10:
**Router Model Affected**
Models: GT-BE98

**Firmware Version Affected**
3006.102.1_1-gnuton0_beta1

**Is this bug present in upstream Merlin releases too?**
Unknown (not available for this Router)

**Describe the bug**
When adding entries to the URL filter and hitting Apply, they aren't saved and the URL filter doesn't work

**To Reproduce**
Steps to reproduce the behavior:
1. Login to the Router.
2. Click on "Firwall".
3. Click on the "URL Filter" Tab.
4. In the text field under "URL Filter List: (Max Limit : 64)" -> URL Filter List:" enter an URL to filter out.
5. Click the "+" button in the "Add/Delete" column to add the entry to the list.
6. Click the "Apply" button.
7. Notice that the entry isn't saved.
8. Double check by going to the blocked URL and see it isn't blocked for real.

**Expected behavior**
URL Entries being saved and blocked as expected.

gnuton, 2024-07-10:
do you get any error in the JS console?

CaveWaverider, 2024-07-11:
> do you get any error in the JS console?
Where can I find the console?

That said, after rebooting the router a couple of times (I had to do it multiple times as connection to WAN/LAN ports would randomly stop being recognized a few minutes after reboot, which I don't know if it's a Gnuton/Merlin or ASUS issue, although I haven't experienced this with the latest ASUS Firmware before I switched to this one - I'll investigate further and write a separate bug report on this if it turns out to be a firmware bug) and now I can't reproduce this anymore as the entries are added properly now. It's still odd that it didn't work before, though.

CaveWaverider, 2024-07-12:
Alright, the bug is back. Not sure where I can find the JS console, though. Do you mean the General Log page?

Drakknar, 2024-07-16:
@CaveWaverider I believe the JS console @gnuton refers to is the browser console. On Chromium-based browsers and on Firefox you can access it by right-clicking on the page, then click "Inspect". A new panel will open, then click on the "Console" tab and browse through the output to see if you can find any errors - they usually have a light red background.

RMerl, 2024-10-01:
WFM on the GT-BE98_PRO.