gnuton/asuswrt-merlin.ng #628: Version: 3004.388.8-gnuton_alpha1 and Current Stable Build - Guest Wi-Fi Doesn't Provide a Connection to Any Devices on Guest on AX82-U v1

Blue49974, 2024-07-11:
**Router Model Affected**
Models: (RT-AX82-U v1)

**Firmware Version Affected**
Stable:3004.388.7_1-gnuton1 
and
Alpha: 3004.388.8-gnuton_alpha1

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
When creating a guest Wi-Fi network it will successfully create them. Just that any device doesn't have internet with VLAN on or off.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to guest network on an AX82-U v1 and create a guest network
2. Apply settings and reboot router
3. Then try setting up the guest Wi-Fi on anything you have that
4. It will say no internet connection with VLAN on or even off

**Expected behavior**
I should be able to connect devices with the guest network on VLAN or off of a VLAN. I need the ability to separate wireless Wi-Fi so nobody is on my main LAN.

gnuton, 2024-07-14:
can you please explain better the way you reproduce this bug? how do you set the vlan?

Blue49974, 2024-08-08:
On OFW the VLAN for guest Wi-Fi is enabled by setting intranet to off.