gnuton/asuswrt-merlin.ng #629: is there a way to use DSL AC68U as AImesh node or repater with wired backhaul connection?

junkkari3, 2024-07-11:
I hope this is the right way to open the ticket. I request a change to firmware which allows DSL AC68U to work as AImesh node wired backhaul connection? Thank you for the great firmware development you do!

### Discussed in https://github.com/gnuton/asuswrt-merlin.ng/discussions/626

<div type='discussions-op-text'>

<sup>Originally posted by **junkkari3** July  9, 2024</sup>
I have two DSL AC68U which I would like to add as wired AI mesh nodes or repeaters. Is it possible? When they are in router mode, I can configure so that instead of the DSL port, it uses one of the four "ethernet out" (lan or what ever) ports as "ethernet in" port (wan or what ever). However, I do not want to use them in router mode, I want to add them to my existing wifi set up (router and mesh node) as mesh nodes, or as repeaters - with wired backhaul connection. Is this somehow possible with the latest gnuton merlin firmware, or with the (old) experimental repeater firmware? I.e. configuring the device in AI node / repeater mode to use wired backhaul (because of the DSL port this is not possible in the ASUS standard software).  Thank you anyway, for a long while I used these two devices with your built through the wireless connection. The build was great (but the devices started acting up so I bought new ones, and now want to add these as wired nodes)</div>

gnuton, 2024-07-11:
Investigation:

Dual wan enabled:
```gnuton@DSL-AC68U-CC88:/tmp/home/root# nvram get wans_dualwan
lan dsl
gnuton@DSL-AC68U-CC88:/tmp/home/root# nvram get wan_ifnames
vlan4 vlan3880
gnuton@DSL-AC68U-CC88:/tmp/home/root# nvram get wandevs
vlan3880 vlan4
gnuton@DSL-AC68U-CC88:/tmp/home/root# nvram get lan_ifnames
vlan1 eth1 eth2 wl0.1```

LAN1 as WAN
gnuton@DSL-AC68U-CC88:/tmp/home/root# nvram get wans_dualwan
lan none
gnuton@DSL-AC68U-CC88:/tmp/home/root# nvram get wan_ifnames
vlan4
gnuton@DSL-AC68U-CC88:/tmp/home/root# nvram get wandevs
et0
gnuton@DSL-AC68U-CC88:/tmp/home/root# nvram get lan_ifnames
vlan1 eth1 eth2 wl0.1
```

gnuton, 2025-06-12:
@Kai-Denzel-Jane what are you trying to achieve?

Kai-Denzel-Jane, 2025-06-12:
> [@Kai-Denzel-Jane](https://github.com/Kai-Denzel-Jane) what are you trying to achieve?

Use Ethernet Backhaul with the DSL-AC68U

gnuton, 2025-06-12:
@Kai-Denzel-Jane that'\s not supported, but I guess it may be possible also with the lastest image (not repeater)

Kai-Denzel-Jane, 2025-06-12:
> [@Kai-Denzel-Jane](https://github.com/Kai-Denzel-Jane) that'\s not supported, but I guess it may be possible also with the lastest image (not repeater)

Yeah I was thinking if you set one of the LAN Ports to WAN surely it might work.

gnuton, 2025-07-24:
@Harrenkyym i have never investigated this properly, but i feel like it should be possible if there is nothing hardcoded in the prebuilt binaries. You can explore this by changing the nvram and restarting few daemons/interfaces or the router itself.