gnuton/asuswrt-merlin.ng #630: Some bugs for GT-BE98 and 3006.102.1_1-gnuton0_beta1 FW

mattheweli, 2024-07-12:
**Router Model Affected**
GT-BE98

**Firmware Version Affected**
3006.102.1_1-gnuton0_beta1

**Is this bug present in upstream Merlin releases too?**
Don't know

**Describe the bug**

1. In Tools - System Information the temperature menu does not work
2. Aura RGB won't stay off if I set it on the WebUI. If you try to change something on WLAN for example, it resets to on. It does not survive a reboot; it will revert to on after a reboot
3. If I turn off the 5 GHz WiFi n.2, in the Wireless Log page 5 GHz WiFi n.2 is indicated as off, but it's on. No 5GHz WiFi n.2 status at all. I think the page shall be edited for the difference between GT-BE98 and GT-BE98Pro

mattheweli, 2024-07-12:
**SCREESHOTS:**

Point 1:

[image]

Point 3:

[image]

sucka666, 2024-07-12:
And missing temperature values i see, i have those missing as well.

sucka666, 2024-07-13:
If it is ok i can add 2 more problems i have found yesterday.

1. One is that on network map page in client list sometimes devices are wrongfully displayed as connected on other interfaces, for example sometimes i see wifi devices appearing as connected to the wired or 5ghz devices connected to 2.4ghz.

2. Second problem i've found myself in, is that last night i was tweaking settings here and there and checking everything is set properly, and on wan page i have changed the default "Enable VPN + DHCP Connection" from yes to no, then suddenly my pppoe connection wouldn't connect. I have tried putting the settings back to yes and nothing changed afterwards, tried setting my connection to automatic and back to pppoe, tried changing every setting on wan connection, restarting the router, nothing changed until i tried resetting to factory, even resetting the router was not working, i mean, i pressed to reset to factory in administration page, it took allot of time to even display the percent it is done, and after that nothing changed, it was not reset, i had to reset the router by keeping the wps button pressed while turning it on.
After resetting and importing settings backup, everything works as it did prior.
I've just received the router yesterday morning when i have clean flashed the firmware as well, so everything was fresh.
The only things i could see in log related to this are these lines:

Jul 12 22:04:29 WAN(0)_Connection: Fail to connect with some issues.
Jul 12 22:04:32 miniupnpd[6485]: no AF_INET address found for ppp0
Jul 12 22:05:02 miniupnpd[6485]: no AF_INET address found for ppp0

Hope it is easy to understand what i have experienced, and that it is helpful, i have a jffs backup from the time my wan was not working if something might be helpful. 
I've been waiting for this firmware to come so i can upgrade from my axe16000 to be98 so if there is anyway i can help the dev process i will try my best.
Thank you.

mattheweli, 2024-07-13:
> If it is ok i can add 2 more problems i have found yesterday.
>
> 1. One is that on network map page in client list sometimes devices are wrongfully displayed as connected on other interfaces, for [...]

For point 1, I can confirm the same issue when in Network Map I group the clients by type of connection. For point 2 I don't want to risk to re-configure all from scratch, but I trust you!

kanuki1984, 2024-07-14:
> If it is ok i can add 2 more problems i have found yesterday.
>
>     1. One is that on network map page in client list sometimes devices are wrongfully displayed as connected on other interfaces, for [...]

Do you have any USB device connected to your router?

mattheweli, 2024-07-14:
> > If it is ok i can add 2 more problems i have found yesterday.
> > ```
> > 1. One is that on network map page in client list sometimes devices are wrongfully displayed as connected on other interfaces, for example sometimes i see wifi devices appearing as connected to the wired or 5ghz devices connected to 2.4ghz.
> > 
> > 2. Second problem i've found myself in, is that last night i was tweaking settings here and there and checking everything is set properly, and on wan page i have changed the default "Enable VPN + DHCP Connection" from yes to no, then suddenly my pppoe connection wouldn't connect. I have tried putting the settings back to yes and nothing changed afterwards, tried setting my connection to automatic and back to pppoe, tried changing every setting on wan connection, restarting the router, nothing changed until i tried resetting to factory, even resetting the router was not working, i mean, i pressed to reset to factory in administration page, it took allot of time to even display the percent it is done, and after that nothing changed, it was not reset, i had to reset the router by keeping the wps button pressed while turning it on.
> >    After resetting and importing settings backup, everything works as it did prior.
> >    I've just received the router yesterday morning when i have clean flashed the firmware as well, so everything was fresh.
> >    The only things i could see in log related to this are these lines:
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Jul 12 22:04:29 WAN(0)_Connection: Fail to connect with some issues. Jul 12 22:04:32 miniupnpd[6485]: no AF_INET address found for ppp0 Jul 12 22:05:02 miniupnpd[6485]: no AF_INET address found for ppp0
> > Hope it is easy to understand what i have experienced, and that it is helpful, i have a jffs backup from the time my wan was not working if something might be helpful. I've been waiting for this firmware to come so i can upgrade from my axe16000 to be98 so if there is anyway i can help the dev process i will try my best. Thank you.
> 
> @sucka666 Do you have any USB device connected to your router?

It's not a question for me, but related to point 1, I've a USB connected

sucka666, 2024-07-14:
> 
> 
> 
> @sucka666 
> 
> Do you have any USB device connected to your router?

@kanuki1984 

Yes, i have a usb 3 samsung t7 ssd connected.

kanuki1984, 2024-07-14:
> > @sucka666
> > Do you have any USB device connected to your router?
> 
> @kanuki1984
> 
> Yes, i have a usb 3 samsung t7 ssd connected.

@sucka666 

Check in "WAN > Dual WAN" whether "4G / 5G Auto Mobile Tethering"
is enabled or not.

Disable it if it's enabled.

sucka666, 2024-07-14:
> > > @sucka666
> 
> > > Do you have any USB device connected to your router?
> 
> > 
> 
> > @kanuki1984
> 
> > 
> 
> > Yes, i have a usb 3 samsung t7 ssd connected.
> 
> 
> 
> @sucka666 
> 
> 
> 
> Check in "WAN > Dual WAN" whether "4G / 5G Auto Mobile Tethering"
> 
> is enabled or not.
> 
> 
> 
> Disable it if it's enabled.

@kanuki1984 

Yes it is disabled, it always was, after the reset everything works anyway, but i haven't touched the setting that produced the initial problem.

Thanks

kanuki1984, 2024-07-14:
> @kanuki1984
>
> Yes it is disabled, it always was, after the reset everything works anyway, but i haven't touched the setting that produced the initial problem.
>
> Thanks

@sucka666 

This setting caused my WAN to stay disconnected after reboot while showing WAN IP
and produced the same error message "WAN(0)_Connection: Fail to connect with some issues." if there is some USB device connected to the router.

I'm suspecting that this setting was not enabled by default in the initial firmware released "3.0.0.6.102_32745" but then it become enabled by default in later firmware version.

That's why I did not have this problem when upgrading the firmware to the latest version
but did not restore the factory default setting and the problem start to appear after
you restore the default setting and there is some USB device connected to the router.

sucka666, 2024-07-14:
> > > > > @sucka666
> > > 
> > > 
> > > > > Do you have any USB device connected to your router?
> > > 
> > > 
> > > > 
> > > 
> > > 
> > > > @kanuki1984
> > > 
> > > 
> > > > 
> > > 
> > > 
> > > > Yes, i have a usb 3 samsung t7 ssd connected.
> > > 
> > > 
> > > @sucka666
> > > Check in "WAN > Dual WAN" whether "4G / 5G Auto Mobile Tethering"
> > > is enabled or not.
> > > Disable it if it's enabled.
> > 
> > 
> > @kanuki1984
> > Yes it is disabled, it always was, after the reset everything works anyway, but i haven't touched the setting that produced the initial problem.
> > Thanks
> 
> @sucka666
> 
> This setting caused my WAN to stay disconnected after reboot while showing WAN IP and produced the same error message "WAN(0)_Connection: Fail to connect with some issues." if there is some USB device connected to the router.
> 
> I'm suspecting that this setting was not enabled by default in the initial firmware released "3.0.0.6.102_32745" but then it become enabled by default in later firmware version.
> 
> That's why I did not have this problem when upgrading the firmware to the latest version but did not restore the factory default setting and the problem start to appear after I restored the default setting and there is some USB device connected to the router.

@kanuki1984 

Wow, that is nasty. Definitely something to keep an eye on, thank you, i will try to test more scenarios when i'l have enough time this days. I have everything disabled there luckily, thank you.
[image]

I had something weirder that made every save stay more time till i see displayed percent on a save event, and then doing nothing that i have request, along the wan problem, i couldn't even restart or reset from the gui, as the problem started when i have changed the "Enable VPN + DHCP Connection" setting, anyway had to be something wrong that showed its effects in the gui as well.

gnuton, 2024-07-16:
Salve gente!!
So many italians here! :D

Thanks a lot for the feedback I was not expecting to get so much reviews. I will try to tackle these issues ASAP.
I would also suggest you to separately file bugs to otherwise something may be lost.. 
Thanks again and feel free to ping me if I miss anyhing.

gnuton, 2024-07-16:
Temperatures -> Fix b292fef115

mattheweli, 2024-07-17:
> Temperatures -> Fix [b292fef](https://github.com/gnuton/asuswrt-merlin.ng/commit/b292fef11537bcbbee20683f3c108c839ba4f434) AURA -> cannot reproduce the bug. Works on ROG UI. Not tested standard.

Ciao Antonio e grazie.
For Aura, I'm using the standard UI. To reproduce, set off Aura LED and apply any new setting everywhere: the Aura LED will revert to on

gnuton, 2024-07-17:
Please check both. As for now I do have plan to support both themes(blue and rog). The blue UI I guess should be considered as the main one.

mattheweli, 2024-07-17:
> Please check both. As for now I do have plan to support both themes(blue and rog). The blue UI I guess should be considered as the main one.

Ok, I'll try ROG UI in the next beta release

mattheweli, 2024-07-19:
> Temperatures -> Fix [b292fef](https://github.com/gnuton/asuswrt-merlin.ng/commit/b292fef11537bcbbee20683f3c108c839ba4f434) AURA -> cannot reproduce the bug. Works on ROG UI. Not tested standard.

Hi @gnuton, I tried to test the b292fef fix on standard UI, by using the following code:

{
    mount | grep -q "/www/Tools_Sysinfo.asp" && \
    umount /www/Tools_Sysinfo.asp 2>/dev/null
    [ -x /jffs/scripts/Tools_Sysinfo_NEW.asp ] && \
    mount -o bind /jffs/scripts/Tools_Sysinfo_NEW.asp /www/Tools_Sysinfo.asp
    mount | grep -q "/www/Tools_Sysinfo.asp" && echo "MOUNTED OK" || echo "NOT MOUNTED"
}

Once I've downloaded the updated Tools_Sysinfo.asp and renamed it to Tools_Sysinfo_NEW.asp.

The problem is temperatures are not shown either:

[image]

mattheweli, 2024-07-19:
Added a pull request for this

mattheweli, 2024-07-19:
I fixed also Wireless log page; I'm pushing the fix in #630 branch

gnuton, 2024-07-27:
@sucka666 I have just tested the latest build and in network map it was all fine (I tested 2.4/5.1/5.2 WiFi. I skipped 6 since I do not have a 6 dev right now).
As for the WAN issue I will move it to a separate issue. That has to be investigated more.

gnuton, 2024-07-27:
@mattheweli 
I will close this issue because it's a mix of different issues.. and it's hard to track for you and me (and full of noise)  feel free to open a new issues if required.

closed 2024-07-27 by gnuton

sucka666, 2024-07-27:
> @sucka666 I have just tested the latest build and in network map it was all fine (I tested 2.4/5.1/5.2 WiFi. I skipped 6 since I do not have a 6 dev right now). As for the WAN issue I will move it to a separate issue. That has to be investigated more.

Thanks, i have already opened a new issue since the last time you have requested separate issues.  
#634