gnuton/asuswrt-merlin.ng #631: GT-BE98LAN/WAN 1 not working if VLAN ID is specified in IPTV for WAN

mattheweli, 2024-07-16:
**Router Model Affected**
GT-BE98

**Firmware Version Affected**
3006.102.1_1-gnuton0_beta1

**Is this bug present in upstream Merlin releases too?**
Probably

**Describe the bug**
I'm in Italy with a 2,5 Gbit fiber connection and my ISP needs VLAN ID 835 to be working (PPPoE connection).

If I specify the VLAN ID on IPTV tab setting in WAN section a message pop up, specifying I can only use 10 Gbit WAN port instead of 2,5 Gbit WAN/LAN 1 port. I can say this is not an issue for me, thanks to the 4x 2,5 Gbit port of the GT-BE98, but for example, with GT-AXE16000, I could specify VLAN ID on IPTV and using either 2,5 Gbit or 10 Gbit port without any issue.

That said, 10 Gbit port as WAN works flawlessly. But LAN/WAN 1 port does not work as LAN port. If I plug the cable, I can see the link is up at 2,5 Gbit, but no IP addresses nor any data transfer (after static IPv4 settings on the client port) happen. I think this could be a bug affection both GT-BE98 and GT-BE98pro with the 3006.102 FW release.

Specifically, if you don't use the VLAN ID for WAN connection, you can use the LAN/WAN 1 port, but it syncs at 1Gbit instead of 2,5 Gbit (reference here, in italian: https://forum.fibra.click/d/55839-router-asus-gt-be98/38)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to IPTV Tab on LAN section
2. Select Manual selection
3. Set any VLAN ID for 'Internet'

**Expected behavior**
If I can't use LAN/WAN 1 as WAN port if VLAN ID tag is set, it should be working as LAN port

CaveWaverider, 2024-07-28:
I don't know if the problem is related, but for some reason none of my GT-98's WAN ports are working after Reboot via UI of the router (the ports show connected but it says No Internet). Unplugging and replugging the cables or plugging into the other WAN port doesn't help. I have to turn the router off and on again and it will work again until the next reboot.

mattheweli, 2024-08-01:
There are reports official FW allow WAN/LAN port 1 2,5Gbit working with latest official FW as LAN port:

https://forum.fibra.click/d/55839-router-asus-gt-be98/93

I strongly suspect this is aconfiguration mismatch from GT-BE98 Pro GPL used with this release and GT-BE98 one used in the latest  official FW for GT-BE98.

@gnuton or @RMerl do you know a way to test why WAN/LAN port 1 2,5 Gbit not working correctly as LAN? Rather the link is up, but there is no IP assigned by DHCP and even with a static IP, there is no traffic

mattheweli, 2024-08-07:
In ASUS ROG forum, another italian user, with official FW version 33001, states the port LAN/WAN 1 is not working as WAN with a specified VLAN ID, but it works as LAN:

https://rog-forum.asus.com/t5/gaming-network-products/rog-rapture-gt-be98-wan-lan1-not-working-as-wan/td-p/986919

I strongly suspect it's not related to GPL at this point (since the verison is using there is lower than the one on 3006.102), but maybe there is some switch configuration to be done on GT-BE98 different than GT-BE98Pro.

Eventually, which is the part of the source involved in this? Is it closed source? Maybe there is some partial merge to be done with GT-BE98 GPL for that part?

gnuton, 2024-08-13:
Hey @mattheweli 
I briefly checked this wan.c in router/rc seems to use get_phy_port_mapping that is defined in  ./release/src/router/shared/sysdeps/broadcom/broadcom.c
The merge with the GPL is correct and the configuration is as follow

```
#elif defined(GTAXE16000)
		.count = 9,
		.port[0] = { .phy_port_id = 5, .ext_port_id = -1, .label_name = "W0", .cap = PHY_PORT_CAP_WAN | PHY_PORT_CAP_WANLAN, .max_rate = 2500, .ifname = "eth0", .flag = (0 | PHY_PORT_FLAG_BYPASS_CABLE_DIAG), .seq_no = -1, .ui_display = NULL },
		.port[1] = { .phy_port_id = 1, .ext_port_id = -1, .label_name = "L1", .cap = PHY_PORT_CAP_LAN, .max_rate = 1000, .ifname = "eth1", .flag = 0, .seq_no = -1, .ui_display = NULL },
		.port[2] = { .phy_port_id = 2, .ext_port_id = -1, .label_name = "L2", .cap = PHY_PORT_CAP_LAN, .max_rate = 1000, .ifname = "eth2", .flag = 0, .seq_no = -1, .ui_display = NULL },
		.port[3] = { .phy_port_id = 3, .ext_port_id = -1, .label_name = "L3", .cap = PHY_PORT_CAP_LAN, .max_rate = 1000, .ifname = "eth3", .flag = 0, .seq_no = -1, .ui_display = NULL },
		.port[4] = { .phy_port_id = 4, .ext_port_id = -1, .label_name = "L4", .cap = PHY_PORT_CAP_LAN, .max_rate = 1000, .ifname = "eth4", .flag = 0, .seq_no = -1, .ui_display = NULL },
		.port[5] = { .phy_port_id = 6, .ext_port_id = -1, .label_name = "L5", .cap = PHY_PORT_CAP_LAN | PHY_PORT_CAP_WANLAN, .max_rate = 10000, .ifname = "eth5", .flag = (0 | PHY_PORT_FLAG_BYPASS_CABLE_DIAG), .seq_no = -1, .ui_display = "10GE 1" },
		.port[6] = { .phy_port_id = 7, .ext_port_id = -1, .label_name = "L6", .cap = PHY_PORT_CAP_LAN | PHY_PORT_CAP_WANLAN, .max_rate = 10000, .ifname = "eth6", .flag = (0 | PHY_PORT_FLAG_BYPASS_CABLE_DIAG), .seq_no = -1, .ui_display = "10GE 2" },
		.port[7] = { .phy_port_id = -1, .ext_port_id = -1, .label_name = "U1", .cap = PHY_PORT_CAP_USB, .max_rate = 5000, .ifname = NULL, .flag = 0, .seq_no = -1, .ui_display = NULL },
		.port[8] = { .phy_port_id = -1, .ext_port_id = -1, .label_name = "U2", .cap = PHY_PORT_CAP_USB, .max_rate = 480, .ifname = NULL, .flag = 0, .seq_no = -1, .ui_display = NULL }
#elif defined(GTBE98) || defined(GTBE98_PRO) || defined(GTBE96) || defined(GTBE19000)
		.count = 9,
		.port[0] = { .phy_port_id = 0, .ext_port_id = -1, .label_name = "W0", .cap = PHY_PORT_CAP_WAN | PHY_PORT_CAP_WANLAN | PHY_PORT_CAP_WANAUTO, .max_rate = 10000, .ifname = "eth0", .flag = (0 | PHY_PORT_FLAG_BYPASS_CABLE_DIAG), .seq_no = -1, .ui_display = "10G WAN/LAN-1" },
		.port[1] = { .phy_port_id = 1, .ext_port_id = 1, .label_name = "L1", .cap = PHY_PORT_CAP_LAN | PHY_PORT_CAP_WANLAN | PHY_PORT_CAP_WANAUTO, .max_rate = 2500, .ifname = "vlan4094", .flag = (0 | PHY_PORT_FLAG_BYPASS_CABLE_DIAG), .seq_no = -1, .ui_display = "2.5G WAN/LAN-1" },
		.port[2] = { .phy_port_id = 2, .ext_port_id = 2, .label_name = "L2", .cap = PHY_PORT_CAP_LAN, .max_rate = 2500, .ifname = "eth1", .flag = (0 | PHY_PORT_FLAG_BYPASS_CABLE_DIAG), .seq_no = -1, .ui_display = "2.5G LAN-2" },
		.port[3] = { .phy_port_id = 3, .ext_port_id = 3, .label_name = "L3", .cap = PHY_PORT_CAP_LAN, .max_rate = 2500, .ifname = "eth1", .flag = (0 | PHY_PORT_FLAG_BYPASS_CABLE_DIAG), .seq_no = -1, .ui_display = "2.5G LAN-3" },
		.port[4] = { .phy_port_id = 4, .ext_port_id = 4, .label_name = "L4", .cap = PHY_PORT_CAP_LAN, .max_rate = 2500, .ifname = "eth1", .flag = (0 | PHY_PORT_FLAG_BYPASS_CABLE_DIAG), .seq_no = -1, .ui_display = "2.5G LAN-4" },
		.port[5] = { .phy_port_id = 5, .ext_port_id = -1, .label_name = "L5", .cap = PHY_PORT_CAP_LAN, .max_rate = 1000, .ifname = "eth2", .flag = 0, .seq_no = -1, .ui_display = "1G LAN-5" },
		.port[6] = { .phy_port_id = 6, .ext_port_id = -1, .label_name = "L6", .cap = PHY_PORT_CAP_LAN, .max_rate = 10000, .ifname = "eth3", .flag = (0 | PHY_PORT_FLAG_BYPASS_CABLE_DIAG), .seq_no = -1, .ui_display = "10G LAN-6" },
		.port[7] = { .phy_port_id = -1, .ext_port_id = -1, .label_name = "U1", .cap = PHY_PORT_CAP_USB, .max_rate = 5000, .ifname = NULL, .flag = 0, .seq_no = -1, .ui_display = NULL },
		.port[8] = { .phy_port_id = -1, .ext_port_id = -1, .label_name = "U2", .cap = PHY_PORT_CAP_USB, .max_rate = 480, .ifname = NULL, .flag = 0, .seq_no = -1, .ui_display = NULL }
```

mrfr34ky, 2024-08-13:
Not sure if this is a bug, as I have the same behavior on stock firmware however would be really nice to use 2.5G WAN with IPTV on. Loosing a 10g port is not an option :(

mattheweli, 2024-08-13:
> Not sure if this is a bug, as I have the same behavior on stock firmware however would be really nice to use 2.5G WAN with IPTV on. Loosing a 10g port is not an option :(

But on stock you still would be able to use it as LAN port, would you?

mrfr34ky, 2024-08-13:
Yes that was never an issue

mrfr34ky, 2024-08-13:
> > Not sure if this is a bug, as I have the same behavior on stock firmware however would be really nice to use 2.5G WAN with IPTV on. Loosing a 10g port is not an option :(
> 
> But on stock FW you still would be able to use it as LAN port, would you? With Gnuton the WAN/LAN 1 port 2,5Gb would not work any longer, either as WAN or LAN port

Yes that was never an issue

mattheweli, 2024-08-13:
> Yes that was never an issue

Plese, review my previous post, I edited it

mattheweli, 2024-08-13:
> .port[1] = { .phy_port_id = 1, .ext_port_id = 1, .label_name = "L1", .cap = PHY_PORT_CAP_LAN | PHY_PORT_CAP_WANLAN | PHY_PORT_CAP_WANAUTO, .max_rate = 2500, .ifname = "vlan4094", .flag = (0 | PHY_PORT_FLAG_BYPASS_CABLE_DIAG), .seq_no = -1, .ui_display = "2.5G WAN/LAN-1" },

I think this is the main issue:

`.port[1] = { .phy_port_id = 1, .ext_port_id = 1, .label_name = "L1", .cap = PHY_PORT_CAP_LAN | PHY_PORT_CAP_WANLAN | PHY_PORT_CAP_WANAUTO, .max_rate = 2500, .ifname = "vlan4094", .flag = (0 | PHY_PORT_FLAG_BYPASS_CABLE_DIAG), .seq_no = -1, .ui_display = "2.5G WAN/LAN-1" },`

Why its ifname is defined as vlan4094? where is defined vlan4094? Cannot be its name eth0, just like 10Gbit WAN port?

mattheweli, 2024-08-13:
I think ASUS has already fixed this issue in their GPL. If you check the 34491 change log for GT-BE98Pro, there is a particular line my eyes stumble upon:

- Fixed dual WAN and VLAN 4094 IoT issues.

I think they changed the "as is" code for broadcom.c to allow using VLAN 4094 (and any other ID) on BE router series; what do you think @gnuton?

rdegaud, 2025-02-07:
647 / 5.000
If we want to take full advantage of the ports (without creating problems with disabled wan/lan ports) we must insert the 802.1Q configuration in the WAN menu.
I tried the RT-BE88U (hw ver 1) 

[image]

which works very well by inserting the ISP parameter (VLAN ID) in this section instead of IPTV which creates problems with the ports of both the RT-BE88U and the GT-BE98
Currently I have the BE98 (hw ver 2) with gnuton 3_1-gnuton0_beta1 firm which does not solve the iptv problem. If I insert IPTV automatically disables the WAN10G port forcing me to use the WAN2.5G port; the BE88U also behaves identically if I use IPTV instead of 802.1Q VLAN ID