gnuton/asuswrt-merlin.ng #632: Wrong WAN IPv6 prefix lenght on GT-BE98

mattheweli, 2024-07-16:
**Router Model Affected**
GT-BE98

**Firmware Version Affected**
3006.102.1_1-gnuton0_beta1

**Is this bug present in upstream Merlin releases too?**
No, it's not present in 388 FW release (tested on GT-AX6000)

**Describe the bug**
Even if I set the /48 WAN prefix and my ISP is capable of assigning me a /48 IPv6 lenght adresses, the router will assign a /64 prefix

**To Reproduce**
Simply try to assign a /48 prefix on IPv6 page, but your ISP should be capable of a /48 (or less, but "higher" than /64 prefix) 

**Expected behavior**
A /48 IPv6 address lenght

**Screenshots**
[image]
[image]

gnuton, 2024-07-17:
This is the correct behaviour.

Reason:
The prefix option allows you to ask the ISP to delegate a /48 prefix. LAN use /64. So even if the ISP allows you that, the router automatically use the first /64 subnet for the primary LAN and the other subnet must be configured for other purposes eg: guest network.

you can debug this using
```
nvram set ipv6_debug=1
service restart_wan
```
you will get outputs like
```
Jul 17 10:01:17 dhcp6_client: WAN Prefix Size Requested:/56, Received:/56
```

closed 2024-07-17 by gnuton

mattheweli, 2024-07-18:
Thanks for the explanation Antonio. Have a nice day!