gnuton/asuswrt-merlin.ng #635: WiFi Network list and active client are not correctly listed on sysinfo page on GT-BE98

mattheweli, 2024-07-18:
**Router Model Affected**
Models: GT-BE98

**Firmware Version Affected**
3006.102.1_1-gnuton0_beta1

**Is this bug present in upstream Merlin releases too?**
Probably not

**Describe the bug**
In the sysinfo page on Tools section the WiFi Network are listed. I disabled 5GHz-2 WiFi network, but it's marked as active with client connected. In the reality the client is connected to 6GHz network. I suspect this is due to the different WiFi architecture of the GT-BE98Pro

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Tools section
2. Click on sysinfo
3. Scroll down to Network
4. See error

**Expected behavior**
The active WiFi network should be listed correctly

**Screenshots**
[image]

mattheweli, 2024-07-19:
I added a pull request for this:

[image]

gnuton, 2024-07-21:
I merged tthe pull request! thanks @mattheweli !

gnuton, 2024-07-27:
tested te latest build and this was fixed

closed 2024-07-27 by gnuton