gnuton/asuswrt-merlin.ng #636: DNS Director Web page crashes when adding a new entry on GT-BE98

mattheweli, 2024-07-19:
**Router Model Affected**
Models: GT-BE98

**Firmware Version Affected**
3006.102.1_1-gnuton0_beta1

**Is this bug present in upstream Merlin releases too?**
No, it's fixed in 3e36b02 (also for TOR page)

**Describe the bug**
In DNS Director web page on LAN section, when you try to add any new entry, the web page crashes because of the wrong syntax of some vars/methods (pls, see commit 3e36b02 in Merlin upostream for details)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to LAN Section
2. Click on DNS Director
3. Try to add any new entry and click on apply
4. See error

**Expected behavior**
The DNS Director web page should not crash when a new entry is added

gnuton, 2024-07-21:
merged 3e36b02 to master.

gnuton, 2024-07-27:
tested no console errors

commit 32113ba8 (2020-11-29): libovpn: allow multiple routes per client to be defined in client-specific rules (closes #636)
closed 2024-07-27 by gnuton