gnuton/asuswrt-merlin.ng #639: "Firmware upgrade unsuccessful" RT-AX82U v1

Eric-Johnston22, 2024-07-20:
**Router Model Affected**
Models: RT-AX82U v1

**Firmware Version Affected**
388.7

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
Any attempt to upgrade firmware from stock results in a "Firmware upgrade unsuccessful" 

**To Reproduce**
Steps to reproduce the behavior:
Following the instructions from [this thread](https://www.snbforums.com/threads/fork-dsl-gnutons-merlin-builds-for-dsl-router-386-1_2-released.70980/):

1. Download the latest image for your model from [here.](https://github.com/gnuton/asuswrt-merlin.ng/releases/latest) Please download the .trx for DSL-AC68U or the .w for DSL-AX82U.
2. Open http://router.asus.com/Advanced_FirmwareUpgrade_Content.asp in your browser and upload the image

**Expected behavior**
A successful installation of the firmware.

gnuton, 2024-07-20:
Are you sure you have a V1? Can you also try the latest pre-release?

adricrist87, 2024-08-06:
Hello, I am running through this issue above, my router is a ver1 I know because I just got it and the sticker on the box says H/W Ver.: 1.0  see the screenshots below:

[image]
[image]
[image]
[image]

Let me know if there is anything else you need to validate.

gnuton, 2024-08-06:
can I close this, right?

alexriveraz, 2024-08-14:
I see this issue was recently reopened, I'm having this problem when updating my AX82U V2 using the latest firmware 388.8_2-gnuton1, previous updates went smooth but this one: 
_**RT-AX82U_V2_3004_388.8_2-gnuton1_nand_squashfs.pkgtb**_
I'm getting the _"Firmware upgrade unsuccessful"_ message and then router reboots automatically.