gnuton/asuswrt-merlin.ng #647: Wireless setting "Apply" button does not work

ttickell, 2024-08-01:
**Router Model Affected**
ZenWifi XT8 / RT-AX95Q

**Firmware Version Affected**
RT-AX95Q_3004_388.7_1-gnuton1_puresqubi.w

**Is this bug present in upstream Merlin releases too?**
No - fully functional on RC-AC86U running 386.14

**Describe the bug**
Wireless settings cannot be changed in the GUI.  Apply button doesn't work on pages:
/Advanced_Wireless_Content.asp - main or professional tabs
/index.asp - wireless tab of system status

**To Reproduce**
Steps to reproduce the behavior:
Go to above mention pages, try to change any wireless setting, hit Apply - nothing happens.

**Expected behavior**
The apply button should take action save/apply/reload etc.

**Screenshots**
Screenshots won't show this - can provide a screen capture video?

ttickell, 2024-08-01:
Screen recording of behavior in both Safari and Firefox - terrible quality to meet upload size limitation.

gnuton, 2024-08-02:
This happens in access point mode,
it's a known issue

gnuton, 2024-08-02:
This issue will be fixed as part of https://github.com/gnuton/asuswrt-merlin.ng/issues/542

closed 2024-08-02 by gnuton