gnuton/asuswrt-merlin.ng #655: Feature Request MFA/2FA (TOTP) - Time-based one-time password

Nullvoid3771, 2024-08-15:
Gnuton/Merlin feature request: Two factor authentication web login. (TOTP - Time-based one-time password)

I think this would benefit PAM (Pluggable Authentication Modules) authentication the routers use. 

For those that have no idea what I’m talking about it’s a security feature that requires you to get a code from a third party authentication application usually from mobile phones. ie. Google Authencator, Authy, Duo, etc. 

2FA might be annoying, but would add a second layer of protection from password guessing for those people that open their routers up to the internet. Not a perfect solution, but better than what is currently available. 

Implementing it into ssh would be nice as well.

Knowing users prefer the KISS method where the least involvement and knowledge is needed to setup remote management would be preferred as router based.

Proxy’s, cloudflare, and vpn server, even ddns are unknowns to well I’d like to say most user with minimal experience.

As was suggested here there is an alternative however “simple” this is not for most users. 

[image]

https://www.snbforums.com/threads/forward-fqdn-to-lan-ip.91109/post-919411

I’ve made this request upstream to Merlin and Asus, but likelihood of Rmerlin or Asus taking this up is questionable. https://www.snbforums.com/threads/feature-request-two-factor-authentication-web-login-totp.91406/

Seems to me not having 2FA on the web gui login that can be made remotely available to WAN and remote management is encouraged to be enabled via the mobile applications… seem like a very blatant security vulnerability. In this day just using passwords is a security risk, as is enabling remote management. But if your manufacturer is poking you to enable this feature under insights tab then the problem is systemic. 

I think adding 2FA to Merlin/Gnuton would be a huge win for security, and not something I see on routers. 

My only concern is it asking for a TOTP - Time-based one-time password if your router cannot access the internet in which case I would implement a fallback to disable 2FA only if WAN or internet assess is not detected.

Nullvoid3771, 2024-08-15:
[image]

Unfortunately Rmerlins explored this a few months back. Sadly it looks like a reverse proxy would be the better method, but the worst method for normal users. 

Just seems wild to me 2FA TOTP isn’t featured in routers. Still I’d love to see someone come up with a user friendly solution or addon.

gnuton, 2024-08-16:
I agree adding 2FA would be a nice feature. Is there any router firmware or there already it? If you have Dev skills you may lead the effort and ask also for our support.

Nullvoid3771, 2024-08-16:
> I agree adding 2FA would be a nice feature. Is there any router firmware out there already it? If you have Dev skills you may lead the effort and ask also for our support.

As far as I know Openwrt and PFsense has MFA/2FA for OpenSSH/dropbear. 

https://openwrt.org/docs/guide-user/services/ssh/ssh.mfa.auth

https://www.comparitech.com/blog/vpn-privacy/pfsense-two-factor-authentication/

Unfortunately my development skills are limited, but I can certainly look at what other opensouce projects use with Linux. Certainly 2fa isn’t unique so finding something uses totp shouldn’t be too hard to find given most NAS’s, Proxmox, Nextcloud etc all can use totp and Authy / google has some documentation for development. 

As Rmerlin pointed out the Asus webgui login page is closed source, so I’m not sure I’m how to get around that without assistance from Asus.