gnuton/asuswrt-merlin.ng #658: ZenWifi AX add mesh node fails - Disconnected & Red light

formeroosid, 2024-08-21:
**Router Model Affected**
Models: ZenWiFi AX XT8

**Firmware Version Affected**
3004.388.8_2

**Is this bug present in upstream Merlin releases too?**
Unknown

**Describe the bug**
Adding nodes to AIMesh immediatly shows red light and show as Disconnected. 

**To Reproduce**
Steps to reproduce the behavior
1. Resent node to factory default
2. Place node next to primary
3. Open UI and click Add AiMesh Node
4. Run through the process
5. observe the success message
6. Open AiMesh UI and click on the added node. 

**Actual Behavior**
Node shows Disconnected. 

**Expected behavior**
The node is added and shows as connected (Great!, of course)

**Screenshots**
[image]
[image]

Log on primary shows this:
```
Aug 21 06:58:34 kernel: CSIMON:  CSIMON[1.1.0] Initialization
Aug 21 06:58:34 kernel: CSIMON: M2M usr already registered ...
Aug 21 06:58:35 wlceventd: wlceventd_proc_event(685): eth4: Auth 32:10:8A:EC:24:50, status: Successful (0), rssi:0
Aug 21 06:58:35 wlceventd: wlceventd_proc_event(722): eth4: Assoc 32:10:8A:EC:24:50, status: Successful (0), rssi:-55
Aug 21 06:59:32 wlceventd: wlceventd_proc_event(685): eth4: Auth 7C:10:C9:06:16:81, status: Successful (0), rssi:0
Aug 21 06:59:32 wlceventd: wlceventd_proc_event(722): eth4: Assoc 7C:10:C9:06:16:81, status: Successful (0), rssi:-19
Aug 21 06:59:36 wlceventd: wlceventd_proc_event(645): eth4: Deauth_ind 7C:10:C9:06:16:81, status: 0, reason: Previous authentication no longer valid (2), rssi:-19
Aug 21 06:59:36 wlceventd: wlceventd_proc_event(645): eth4: Deauth_ind 7C:10:C9:06:16:81, status: 0, reason: Previous authentication no longer valid (2), rssi:-19
Aug 21 06:59:36 wlceventd: wlceventd_proc_event(685): eth4: Auth 7C:10:C9:06:16:81, status: Successful (0), rssi:0
Aug 21 06:59:36 wlceventd: wlceventd_proc_event(722): eth4: Assoc 7C:10:C9:06:16:81, status: Successful (0), rssi:-19
Aug 21 06:59:37 wlceventd: wlceventd_proc_event(685): eth6: Auth 7C:10:C9:06:16:88, status: Successful (0), rssi:-39
Aug 21 06:59:37 wlceventd: wlceventd_proc_event(722): eth6: Assoc 7C:10:C9:06:16:88, status: Successful (0), rssi:-39
Aug 21 06:59:40 wlceventd: wlceventd_proc_event(645): eth4: Deauth_ind 7C:10:C9:06:16:81, status: 0, reason: Previous authentication no longer valid (2), rssi:-19
Aug 21 06:59:41 wlceventd: wlceventd_proc_event(645): eth6: Deauth_ind 7C:10:C9:06:16:88, status: 0, reason: Previous authentication no longer valid (2), rssi:-38
Aug 21 06:59:41 wlceventd: wlceventd_proc_event(645): eth6: Deauth_ind 7C:10:C9:06:16:88, status: 0, reason: Previous authentication no longer valid (2), rssi:-38
Aug 21 06:59:41 wlceventd: wlceventd_proc_event(645): eth4: Deauth_ind 7C:10:C9:06:16:81, status: 0, reason: Unspecified reason (1), rssi:-15
Aug 21 06:59:41 wlceventd: wlceventd_proc_event(685): eth4: Auth 7C:10:C9:06:16:81, status: Successful (0), rssi:-15
Aug 21 06:59:41 wlceventd: wlceventd_proc_event(722): eth4: Assoc 7C:10:C9:06:16:81, status: Successful (0), rssi:-19
Aug 21 06:59:44 wlceventd: wlceventd_proc_event(685): eth6: Auth 7C:10:C9:06:16:88, status: Successful (0), rssi:-38
Aug 21 06:59:44 wlceventd: wlceventd_proc_event(722): eth6: Assoc 7C:10:C9:06:16:88, status: Successful (0), rssi:-38
Aug 21 06:59:44 kernel: Register interface [wds2.0.1]  MAC: 04:42:1a:c8:13:b8
Aug 21 07:00:03 wlceventd: wlceventd_proc_event(685): eth4: Auth EA:D2:FB:24:27:8A, status: Successful (0), rssi:0
Aug 21 07:00:03 wlceventd: wlceventd_proc_event(695): eth4: ReAssoc EA:D2:FB:24:27:8A, status: Successful (0), rssi:-83
Aug 21 07:00:03 wlceventd: wlceventd_proc_event(685): eth5: Auth 00:A5:54:5A:D0:F2, status: Successful (0), rssi:0
Aug 21 07:00:03 wlceventd: wlceventd_proc_event(695): eth5: ReAssoc 00:A5:54:5A:D0:F2, status: Successful (0), rssi:-70
Aug 21 07:00:06 rc_service: cfg_server 2022:notify_rc start_bsd
Aug 21 07:00:12 kernel: Flushing net_device wds0.0.1.
Aug 21 07:00:12 wlceventd: wlceventd_proc_event(662): wds0.0.1: Disassoc 7C:10:C9:06:16:81, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Aug 21 07:00:12 wlceventd: wlceventd_proc_event(662): eth4: Disassoc 7C:10:C9:06:16:81, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Aug 21 07:00:12 wlceventd: wlceventd_proc_event(662): wds2.0.1: Disassoc 7C:10:C9:06:16:88, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Aug 21 07:00:12 wlceventd: wlceventd_proc_event(662): eth6: Disassoc 7C:10:C9:06:16:88, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Aug 21 07:00:18 wlceventd: wlceventd_proc_event(685): eth4: Auth 7C:10:C9:06:16:81, status: Successful (0), rssi:0
Aug 21 07:00:18 wlceventd: wlceventd_proc_event(722): eth4: Assoc 7C:10:C9:06:16:81, status: Successful (0), rssi:-19
Aug 21 07:00:22 wlceventd: wlceventd_proc_event(645): eth4: Deauth_ind 7C:10:C9:06:16:81, status: 0, reason: Previous authentication no longer valid (2), rssi:-19
Aug 21 07:00:23 wlceventd: wlceventd_proc_event(645): eth4: Deauth_ind 7C:10:C9:06:16:81, status: 0, reason: Previous authentication no longer valid (2), rssi:-19
Aug 21 07:00:23 wlceventd: wlceventd_proc_event(685): eth4: Auth 7C:10:C9:06:16:81, status: Successful (0), rssi:0
Aug 21 07:00:23 wlceventd: wlceventd_proc_event(722): eth4: Assoc 7C:10:C9:06:16:81, status: Successful (0), rssi:-19
[... 6 lines cut ...]
Aug 21 07:00:28 wlceventd: wlceventd_proc_event(685): eth4: Auth 7C:10:C9:06:16:81, status: Successful (0), rssi:0
Aug 21 07:00:28 wlceventd: wlceventd_proc_event(722): eth4: Assoc 7C:10:C9:06:16:81, status: Successful (0), rssi:-15
Aug 21 07:00:28 wlceventd: wlceventd_proc_event(645): eth6: Deauth_ind 7C:10:C9:06:16:88, status: 0, reason: Previous authentication no longer valid (2), rssi:-37
Aug 21 07:00:28 bsd: bsd: STA:ea:d2:fb:24:27:8a no response
Aug 21 07:00:28 bsd: bsd: Sending act Frame to ea:d2:fb:24:27:8a with transition target eth5 ssid 04:42:1a:c8:13:b4
Aug 21 07:00:29 bsd: bsd: STA:ea:d2:fb:24:27:8a no response
Aug 21 07:00:30 wlceventd: wlceventd_proc_event(685): eth6: Auth 7C:10:C9:06:16:88, status: Successful (0), rssi:-39
Aug 21 07:00:30 wlceventd: wlceventd_proc_event(722): eth6: Assoc 7C:10:C9:06:16:88, status: Successful (0), rssi:-39
Aug 21 07:00:30 kernel: Register interface [wds2.0.1]  MAC: 04:42:1a:c8:13:b8
Aug 21 07:00:31 wlceventd: wlceventd_proc_event(662): eth6: Disassoc EA:D2:FB:24:27:8A, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Aug 21 07:00:33 wlceventd: wlceventd_proc_event(662): wds2.0.1: Disassoc 7C:10:C9:06:16:88, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Aug 21 07:00:33 wlceventd: wlceventd_proc_event(662): eth6: Disassoc 7C:10:C9:06:16:88, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Aug 21 07:00:35 kernel: Flushing net_device wds0.0.1.
Aug 21 07:00:35 wlceventd: wlceventd_proc_event(662): wds0.0.1: Disassoc 7C:10:C9:06:16:81, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Aug 21 07:00:35 wlceventd: wlceventd_proc_event(662): eth4: Disassoc 7C:10:C9:06:16:81, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Aug 21 07:00:36 wlceventd: wlceventd_proc_event(685): eth4: Auth 7C:10:C9:06:16:81, status: Successful (0), rssi:0
Aug 21 07:00:36 wlceventd: wlceventd_proc_event(722): eth4: Assoc 7C:10:C9:06:16:81, status: Successful (0), rssi:-19
Aug 21 07:00:38 wlceventd: wlceventd_proc_event(645): eth6: Deauth_ind 00:A5:54:5A:D0:F2, status: 0, reason: Disassociated due to inactivity (4), rssi:-69
Aug 21 07:00:38 wlceventd: wlceventd_proc_event(645): eth6: Deauth_ind 00:A5:54:5A:D0:F2, status: 0, reason: Previous authentication no longer valid (2), rssi:-69
Aug 21 07:00:39 wlceventd: wlceventd_proc_event(662): eth6: Disassoc 00:A5:54:5A:D0:F2, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Aug 21 07:00:39 wlceventd: wlceventd_proc_event(662): wds0.0.1: Disassoc 7C:10:C9:06:16:81, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Aug 21 07:00:39 kernel: Flushing net_device wds0.0.1.
Aug 21 07:00:39 wlceventd: wlceventd_proc_event(662): eth4: Disassoc 7C:10:C9:06:16:81, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0

```

formeroosid, 2024-08-21:
This does not appear to occur when there is a wired connection between the LAN of the primary and the WAN of the remote.

gnuton, 2024-08-27:
Hi @formeroosid 
does the node remain in a disconnected state? 
XT8 AiMesh has some issues but as far as know those happen only when in Access point mode.

formeroosid, 2024-08-27:
It does stay disconnected. I'm using Router Mode.

Jamie

shama84, 2024-09-16:
Hello guys, just to add my 2 cents: 

_Hardware: 3x XT8 with either wired and wireless backhaul, router mode_
About 20 days ago after doing dirty upgrade from Gnuton 388.7-1 to 388.8-2 I got JFFS issue. The mount point was full. Not sure if it has anything to do with 388.8-2 but I decided to stay on 388.8-2 and factory reset all nodes. The procedure I used:
- configure the 1st node (router) xith basic config and Internet connectivity
- power on the wired node. Unable to join but this is "normal", the aimseh setup starts it seems only it the pairing is done with the phone app or if an aimesh node is available during the initial setup.
-restart the router (1st node) with the wired node connected make it right
-power on the wireless node (1m distance) and add it through the web interface

Running for 20 days now, JFFS stable, network stable.
I used the WPS reset method, so holding the WPS button up to the second XT8 reboot.

gnuton, 2024-09-16:
@shama84 Correct. The factory reset was already suggested.
The router mod should be fine.