gnuton/asuswrt-merlin.ng #659: Can't enable nfs exports on usb application section

Lobo2022, 2024-08-22:
**Router Model Affected**
Models: RT-AX58U V2

**Firmware Version Affected**
3004.388.8_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
There is no Merlin version for this model

**Describe the bug**
Turning button to enable NFSD not working

**To Reproduce**
Steps to reproduce the behavior:
1. Go to   usb application section
2. Click on  Enable NFSD
3. NFSD remains off

lexiismadd, 2025-01-05:
Also experiencing it on 3006.102.2_2

derekrobertson, 2025-06-07:
Same problem on 3004.388.8_4-gnuton1 on a RT-AX58U v2

Lobo2022, 2025-06-09:
Hi,

I tested this image on my RT-AX58U v2. Now the switch to enable nfs exports works l and i was able to create an export.
BUT, I couldn’t establish a connection.

_**Router side:**_

 **noname@Router:/#** cat /etc/exports 
 
/mnt/Router/entware 192.168.1.0/24(no_root_squash,rw,sync,no_subtree_check)

**_Client side:_**

**client@Hass:~ $** sudo /bin/mount -v -t nfs 192.168.xxx.:mnt/Router/entware testes -o nolock,rw,tcp

mount.nfs: timeout set for Mon Jun  9 08:28:38 2025
mount.nfs: trying text-based options 'nolock,tcp,vers=4.2,addr=192.168.1.xxx,clientaddr=192.168.1.xxx'
mount.nfs: mount(2): Connection refused
mount.nfs: trying text-based options 'nolock,tcp,addr=192.168.1.xxx'
mount.nfs: prog 100003, trying vers=3, prot=6
mount.nfs: portmap query failed: RPC: Program not registered

I guess this is because the other issue #660 I raised. **_Nfs services are not working on RT-AX58U v2. No kernel  nfs.ko_**

**_Router side:_**

**noname@Router:/#** find / -name '*ko' | grep fs

/lib/modules/4.19.183/Router side:kernel/fs/nls/nls_cp437.ko
/lib/modules/4.19.183/kernel/fs/nls/nls_cp850.ko
/lib/modules/4.19.183/kernel/fs/nls/nls_cp852.ko
/lib/modules/4.19.183/kernel/fs/nls/nls_cp866.ko
/lib/modules/4.19.183/kernel/fs/nls/nls_cp932.ko
/lib/modules/4.19.183/kernel/fs/nls/nls_cp936.ko
/lib/modules/4.19.183/kernel/fs/nls/nls_cp949.ko
/lib/modules/4.19.183/kernel/fs/nls/nls_cp950.ko
/lib/modules/4.19.183/kernel/fs/nls/nls_euc-jp.ko
/lib/modules/4.19.183/kernel/fs/tfat.ko
/lib/modules/4.19.183/kernel/fs/thfsplus.ko
/lib/modules/4.19.183/kernel/fs/tntfs.ko
/lib/modules/4.19.183/kernel/net/sched/sch_hfsc.ko

I hope it helps