gnuton/asuswrt-merlin.ng #660: mountinf nfs share not working

Lobo2022, 2024-08-22:
**Router Model Affected**
Models: RT-AX 58U v2

**Firmware Version Affected**
3004.388.8_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
I have an old RT-AC66U B1 with firmware 386.14. and the described command works perfectly

**Describe the bug**
1- Entware installed through amtm.
2. using BusyBox v1.25.1  mount cmd:
  /bin/mount -t nfs 192.168.1.114:yyyy /tmp/xxxx -o nolock,rw,tcp
3. return:
    mount.nfs: No such device
    mount: mounting 192.168.1.114:scripts on /tmp/scripts failed: No such device
4. using exactly the same (BusyBox v1.25.1)  mount cmd on RT-AC66U B1 works succesfully and remote dir is mounted
5. all addittional software installed with opkg is the same in both routers ( some cmds version are newer on  RT-AX 58U v2)

Lobo2022, 2024-08-23:
This pb happens with stock firmware as well.

gnuton, 2024-08-27:
Hi,
as you correctly checked NFS is not enabled yet.

Naolador, 2026-03-18:
@gnuton I'm facing the same issue. Neither cifs nor nfs module could be found. Those commands returns nothing:
```find /lib/modules -name "*.ko" | grep cif```
```find /lib/modules -name "nfs.ko"```

Is there a way to enable any one of them?

LiroyvH, 2026-10-13:
@Naolador Did you have any luck with finding a solution to that one? Was surprised to find CIFS missing.

Naolador, 2026-10-14:
@LiroyvH Unfortunately no. I've already reverted to the stock FW since this mod is causing other strange issues.