gnuton/asuswrt-merlin.ng #667: Upload firmware does not work

mattdale77, 2024-08-29:
**Router Model Affected**
Models: (eg: DSL-AX82U/RT-AX82U/...)

**Firmware Version Affected**
384.xx

**Is this bug present in upstream Merlin releases too?**
I have an XT8 so I don't know if it also affects them

**Describe the bug**
The router has 386.07_2-gnuton1 installed and it claims the latest release is 388.2_2-gnuton1. I have downloaded the file RT-AX95Q_388.2_2_0-gnuton1_puresqubi.w and clicked on the upload button and tried to upload the file but it merely says "Please wait applying settings" and goes no further

**To Reproduce**
Steps to reproduce the behavior:
1. Firmware update page
2. Upload
3. Try to upload file RT-AX95Q_388.2_2_0-gnuton1_puresqubi.w
4. The upload does not progress

**Expected behavior**
Firmware would be installed. It may be that this upload mechanism does not work with the older firmware installed and I have to flash it but this was done by the previous owner and I'm not yet familiar so I wanted to see if this should be expected to work first or if I'm maybe doing something wrong

mattdale77, 2024-08-31:
> *DuckDuckGo* removed one tracker. More <https://duckduckgo.com/-hGyvyiXmf6u7VQd9sTh6kcEsdlr7FNVN-QhZ2DrCcFNg-DNxUNMLAMwnwvG9ZFp2LRVj0gcInUGADxMG_WzPIoEMQ5LLc3p_d0hq4NtBiQNdCsjKs1MxkVZvhuScAT0pTVXI1BAo6KIGHu1NanECD7lgL_RoAXmEWu_MEloitu_vsCARWJG-x6xO85otKEbXHxA1ijibPIu-FHvyCN7nvcfqlE-wV6INNCEoBamqMXwtG1o8JTr-zC25fBb0Ral-YxJRBPdTWDM3F8tJ0LgYTfwR2zMUjyQCcnfgEPmgLy2-EysiKx5eIQ6L7fICPVfQySkLJ444dGDJ5HUpQXxqKPwPhl2DQNsONZa0v8SqRftia7lPO2P8IL5gu7CdPdBdnXx4us4hk9D4JvvXMhHgv4-oU-FB2evrUE6TU2t_8HPtfR1s4dltQb2KGqZONIGYDX6MTg6T2OHgVCU53tPwwNN8CAeNufcOnf38JktQvzXeU73_C_JTT4r0SV2EMEZPGdQ>
> Report Spam <https://duckduckgo.com/-hGyvyiXmf6u7VQd9sTh6kcEsdlr7FNVN-QhZ2DrCcFNg-DNxUNMLAMwnwvG9ZFp2LRVj0gcInUGADxMG_WzPIoEMQ5LLc3p_d0hq4NtBiQNdCsjKs1MxkVZvhuScAT0pTVXI1BAo6KIGHu1NanECD7lgL_RoAXmEWu_MEloitu_vsCARWJG-x6xO85otKEbXHxA1ijibPIu-FHvyCN7nvcfqlE-wV6INNCEoBamqMXwtG1o8JTr-zC25fBb0Ral-YxJRBPdTWDM3F8tJ0LgYTfwR2zMUjyQCcnfgEPmgLy2-EysiKx5eIQ6L7fICPVfQySkLJ444dGDJ5HUpQXxqKPwPhl2DQNsONZa0v8SqRftia7lPO2P8IL5gu7CdPdBdnXx4us4hk9D4JvvXMhHgv4-oU-FB2evrUE6TU2t_8HPtfR1s4dltQb2KGqZONIGYDX6MTg6T2OHgVCU53tPwwNN8CAeNufcOnf38JktQvzXeU73_C_JTT4r0SV2EMEZPGdQ>
>
>
> I have the same issue, <https://github.com/gnuton/asuswrt-merlin.ng/issues/664>, but with an TUF-AX3000 V2.
>
> Are you automatically forwarded to the login screen after 99% is reached in the updating window?

I don't even get a percentage

troymuller, 2024-09-06:
Reboot each leaf node and the main router and try again