gnuton/asuswrt-merlin.ng #668: guest login MUST be enabled to have access to my ext. NAS

feiercrack, 2024-08-29:
**Router Model Affected**
Models: RT-AX82U v1

**Firmware Version Affected**
004.388.8_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
dunno

**Describe the bug**
when i like to access my external SSD (NAS)  i have to enable the guest login, otherwise windows 11 pro x64 24H2
cant connect with my login/passwd -  error: 0x80070043

In windows i simply cant activate the network discovery and the file / printer sharing option. 
i can turn them on, both, but nothing happens. when i reopen the network & internet tab, all my settings are gone -.-

i had invested the whole day to find a solution, but nothing helps

[image]

Nullvoid3771, 2024-09-24:
This isn’t a bug. This is a function of you using smb2-3 on windows which requires encryption, and signing. By enabling guest without authentication you’re allowing smb2-3 to function to not use encryption and file signing.

samba4 is not available on ASUS routers without installing the entware. Meaning smb3 isn’t supported which is default on windows I believe. Smb1 is disabled by default on windows 10 & 11.

Nullvoid3771, 2024-09-24:
SSH into your router.

Do this:
- have jffs enabled 
- have ssh or telnet access to your router enabled
- log in using either of the two
- create the file /jffs/configs/smb.conf.add
- add the parameters to smb.conf.add
- reboot router

You can look for examples online for the smb.conf configuration note that this file handles all of your shares meaning with may ignore the shares already setup. This file will override anything the gui says. By default ASUS Merlin/gnuton would overwrite the normal smb.conf file each reboot so you need an add this script to replace the smb.conf file the gui uses. This script will be used each reboot. 

By the same respects if you use entware samba4 it will likely ignore the gui and rely solely on the entwares smb.conf file which is likely separate from the built in smb.conf file or the smb.conf.add file. 

The smb.conf configuration file is how you’d setup samba on any Linux machine without a gui. 

You can refer to the man file for additional parameters. Avoid the tutorials that tell you how to tune or optimize samba they are usually incorrect or obsolete functions with modern samba versions. 

https://www.samba.org/samba/docs/current/man-html/smb.conf.5.html

codyrocco, 2024-09-24:
wouldn't it be nice to have this possibility in asus' default firmware? :)
just for server / client_min_protocol 1 ...

feiercrack, 2024-09-24:
> SSH into your router.
> 
> Do this:
> 
> * have jffs enabled
> * have ssh or telnet access to your router enabled
> * log in using either of the two
> * create the file /jffs/configs/smb.conf.add
> * add the parameters to smb.conf.add
> * reboot router
> 
> You can look for examples online for the smb.conf configuration note that this file handles all of your shares meaning with using this it may ignore the shares already setup. This file should override anything the gui says. By default ASUS Merlin/gnuton would overwrite the normal smb.conf file each reboot so you need an add this config file to replace the smb.conf file the gui uses. This configuration file will be used each reboot replacing the content of the tmp smb.conf file the system uses by default.
> 
> I’m not sure off the top of my head the location of the main smb.conf file (check the tmp directory) the built in Samba server uses but you can copy its contents to the smb.conf.add file to make your life easier.
> 
> By the same respects if you use entware samba4 it will likely ignore the gui and rely solely on the entwares smb.conf file which is likely separate from the built in smb.conf file or the smb.conf.add file. Samba4 entware may also need a startup script to tell it to start each boot. (Not needed for the built in samba server)
> 
> Refer to this: https://github.com/RMerl/asuswrt-merlin.ng/wiki/User-scripts
> 
> The smb.conf configuration file is how you’d setup samba on any Linux machine without a gui. But because this router is for the most part read-only you need to use jffs or equivalent to have a writable file.
> 
> You can refer to the man file for additional parameters. Avoid the tutorials that tell you how to tune or optimize samba they are usually incorrect or obsolete functions with modern samba versions.
> 
> https://www.samba.org/samba/docs/current/man-html/smb.conf.5.html

Awesome, that helps alot, ty!

feiercrack, 2024-12-04:
> SSH into your router.
>
> Do this:
>
> * have jffs enabled
> * have ssh or telnet access to your router enabled
> * log in using either of the two
> * create the file /jffs/configs/smb.conf.add
> * add the [...]

ok,
i installed samba4-client, samba4-server, samba4-utils and samba4-admin v4.18.8-2  via opk inside my RT-AX82U.
then i come to the point -->  create the file /jffs/configs/smb.conf.add
well...
but after that, i really dont know what to write in it. i copied the smb.conf in same directory, so that i can rename it later to smb.conf.add or copy-pasta the content if needed...
here's my smb.conf 
[https://pastebin.com/udV69HMG](url)

any help please?
i am not unix guru, just another n00b with tons of interests, but that [https://www.samba.org/samba/docs/current/man-html/smb.conf.5.html](url) is quite overloading my brain -.-

p.s. do i need to install samba4 in my win11 too? cause find any -.-

feiercrack, 2024-12-04:
i think i got it to work:

removed the line "max protocol = SMB2"
added the line "protocol = SMB3"

rebooted the router
changed inside the routers UI -->USB Application-->Servers Center -->Network Place (Samba) Share / Cloud Disk
from
[image]
to 
[image]

and it seems now to work ^^

Nullvoid3771, 2024-12-04:
> i think i got it to work:
> 
> removed the line "max protocol = SMB2" added the line "protocol = SMB3"
> 
> rebooted the router changed inside the routers UI -->USB Application-->Servers Center -->Network Place (Samba) Share / Cloud Disk from ![1](https://private-user-images.githubusercontent.com/47046040/392191769-df4614b8-5ad4-4a43-bbdc-50d1193d467f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMyODcyMDMsIm5iZiI6MTczMzI4NjkwMywicGF0aCI6Ii80NzA0NjA0MC8zOTIxOTE3NjktZGY0NjE0YjgtNWFkNC00YTQzLWJiZGMtNTBkMTE5M2Q0NjdmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA0VDA0MzUwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU5OWE1MjdmNDljYzU4NzIwOTc2MWRhNTA5MGE2NGEyYzI1YTc4ZDkyOGU4MTJhNmQ0MWY4NzJjN2MyYjA2MmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.6JqZb40rZ0kSOTNJUdi5VmXvKCb5VlWApCTol77mAyo) to ![2](https://private-user-images.githubusercontent.com/47046040/392191489-74892a1a-65c3-428f-b299-9486530ac543.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMyODcyMDMsIm5iZiI6MTczMzI4NjkwMywicGF0aCI6Ii80NzA0NjA0MC8zOTIxOTE0ODktNzQ4OTJhMWEtNjVjMy00MjhmLWIyOTktOTQ4NjUzMGFjNTQzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA0VDA0MzUwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcwNzcyZTkxOGIzODRjZGU4MTBkYzYyYmJiOTNjNTQzMGZmOWNmOTM2OTkwNWQyNzhkYjMxMTUwYWZiNTA3ZWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.lRpElzI3_hNJ_3MIgZ42WwWIoOFbYBcvoD0_osmw2hA)
> 
> and it seems now to work ^^

Glad it’s working. No you don’t need samba 4 on windows it’s already capable of that.

Assuming the mounts still point to the correct play you shouldn’t need to change much in the /jffs/configs/smb.conf.add file. I’m unawares if changing the smb setting in the routers gui interface affect Samba4 or the older built it samba after installing samba4. You might be running two instances of samba servers if you enable the samba server from the router gui.. I haven’t tested any of this because I don’t use the router as a NAS myself. 

But if it’s working awesome.