gnuton/asuswrt-merlin.ng #669: The number of devices in the app is unstable

lenovomen, 2024-09-02:
Asus TUF 3000 v2
Firmware: the latest

I installed the latest Merlin firmware on my tuf 3000v2 router. In the application it shows 8 devices connected but only 7. What is the reason for the unstable operation? Not showing well many times.

lenovomen, 2024-09-02:
Does anyone know what is causing this error? It shows 8 devices connected in the bottom bubble, but only 7 are actually connected. Otherwise, the connection time and all the data is correct, but sometimes a device appears in the list that is not connected to the router anyway.

gnuton, 2024-09-02:
no idea can you share some screenshots?

572289049, 2024-09-04:
This is happening to me, and I've noticed that the refresh on my TUF-AX5400 router interface is useless.

Router Model Affected
Models: TUF-AX5400

Firmware Version Affected
3004_388.8_2-Gnuton1

Is this bug present in upstream Merlin releases too?
Yes

[image]