gnuton/asuswrt-merlin.ng #671: Weird wifi AP shows up after resetting router

shnoah641, 2024-09-02:
**Router Model Affected**
Models: RT-AX82U V2

**Firmware Version Affected**
384.xx

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
A weird wifi AP that has random numbers and letters shows up after resetting router and or just restarting my wifi AP altogether after modifying settings or even just restarting the router. Also my router gets cut off from being able to access the web gui even with settings in place to let specific devices into the web gui only.  The weird AP in the screenshot shows just below the Open Sesame AP in the wifi list. 

**To Reproduce**
Steps to reproduce the behavior:
1. Restart router after initial setup
2. Modify wifi settings then apply 
4. See error

**Expected behavior**
I expected it to run smoothly but unfortunately this happens everytime no matter what I do from fully resetting it to changing the wifi settings and starting fresh.

**Screenshots**
[image]

gnuton, 2024-09-02:
if this is happening also on other routers, can you please point me to some web pages to undersstand what's the problem,?

shnoah641, 2024-09-02:
> if this is happening also on other routers, can you please point me to some web pages to undersstand what's the problem,?

It only happens to my RT-AX82U V2 router only that I know of  and every web page that I try to access gives me this error until I reset and upload my old backup of the router with the AP radio for both 2.4 ghz and 5 ghz completely off then it starts working again. But as soon as both radios are on and wifi networks are visible it begins to happen again where I am unable to access both the router web gui and any website that I type in. Also the NTP server is refusing to sync as well and I think that is also why my Wireguard server has been unable to run again. 

Screenshots:
[image]

[image]

shnoah641, 2024-09-02:
Here is a photo of the fake router login page and also one of the one it should be. First photo is the fake one after resetting and applying my backup and second photo is the original and the correct one after I had to restart my laptop connected to the router then it works just fine to login.

[image]

[image]

gnuton, 2024-09-03:
Did you try to run a fresh install, no addons and no backups?
The problem seems to be it's in some of your settings

shnoah641, 2024-09-04:
Yes I tried to run a fresh install and it showed back up again with no add ons or any backups

Ichimikichiki, 2024-09-06:
It's more of this issue, which I keep getting gaslit about.

https://github.com/gnuton/asuswrt-merlin.ng/issues/672

shnoah641, 2024-09-06:
I got the VPN to run again it was the current setup I had with my ports
that I blocked so I started fresh and it worked again for me. And now the
weird AP has finally disappeared as well. It's just weird that it ever
showed up to begin with. But definitely not similar to that problem
though...mine turned out to be me not opening up the correct port necessary
for the vpn to work and when I did the fresh restart of the router it
started working again without any issues. I just made sure this time not to
block port 51820 for Wireguard in the VPN service area.

gnuton, 2024-09-08:
@shnoah641 if you have solved this issue, can we close issue?

closed 2024-09-08 by shnoah641