gnuton/asuswrt-merlin.ng #672: Last two updates - you activate all killswitches even though VPN is not active, preventing any outgoing connection

Ichimikichiki, 2024-09-03:
Router Model Affected
Models: TUF-AX3000

Firmware Version Affected
3004.388.8_2-gnuton1
3004.388.8_0-gnuton0_alpha1

No VPN is active at all. However the killswitch is actively blocking connections, when it should not be due to OpenVPN or any other VPN currently inactive.

```
May  5 15:05:10 openvpn-routing: Initializing killswitch
May  5 15:05:36 wan: finish adding multi routes
May  5 15:05:36 openvpn-routing: Applying all killswitches
```

Disabling kill switch on all (inactive) openvpn settings resolves the issue.

gnuton, 2024-09-06:
An OpenVPN killswitch is a security feature designed to prevent your device from sending any internet traffic if your VPN connection unexpectedly drops. Its purpose is to ensure that your real IP address and data remain private by blocking all internet activity outside of the VPN tunnel.

Here’s how it works:

Monitors the VPN Connection: The killswitch constantly checks if the VPN connection is active.
Cuts Off Internet Access if VPN Fails: If the VPN connection drops or fails, the killswitch automatically disables your device's access to the internet. This ensures that no traffic goes through the unencrypted, regular internet.
Restores Connection When VPN is Back: Once the VPN is re-established, the killswitch allows internet access again.
Without a killswitch, if your VPN disconnects unexpectedly, your device would continue using the unprotected internet connection, potentially exposing your IP address and traffic to your ISP or other third parties.

Ichimikichiki, 2024-09-06:
Thanks @gnuton I'm aware what it is...

You seem confused... assume I'm an expert Network Engineer... The kill switch should not be enabled unless I'm using the VPN or attempting to... it should not be active permanently... whether you attempt to activate the VPN or not....

Ichimikichiki, 2024-09-06:
Stop gaslighting me. And pay attention to what I am saying....

Ichimikichiki, 2024-09-06:
Go turn on your VPN kill switches to all your VPN settings. However don't activate any VPN. And tell me good your internet connection goes @gnuton