gnuton/asuswrt-merlin.ng #675: RT-AX82U Can't find VPN function in Web UI

minimoonleaf, 2024-09-05:
**Router Model Affected**
Models: RT-AX82U

**Firmware Version Affected**
3004.388.8_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
I don't know

**Describe the bug**
Can't find VPN function in Web UI

**To Reproduce**
Steps to reproduce the behavior:
1. Login Web UI
2. Find VPN function.

**Expected behavior**
Have the VPN function can choice.

**Screenshots**
[image]

Rovasteen, 2024-09-05:
You might clear your browser cache.  Sometimes the browser gets stuffed up and clearing the cache will allow the Web UI to display correctly.

gnuton, 2024-09-06:
When your ASUS router is in access point (AP) mode, it generally doesn't handle VPN functions like it would in its default router mode. AP mode is designed to extend your network's wireless coverage and provide more LAN ports, but it offloads most routing features—like NAT (Network Address Translation), firewall, and VPN configuration—to the main router.

In AP mode, the ASUS router with Merlin firmware typically disables the VPN options, as those are routing functions, and the device is no longer acting as the primary router. The VPN setup would need to be done on the main router or another device in the network if the ASUS router is in AP mode.

If you need VPN functionality on the ASUS router, you'd need to switch it back to router mode or use a dedicated VPN client on devices connected to the access point.

closed 2024-09-06 by gnuton