gnuton/asuswrt-merlin.ng #677: Stop gaslighting me and listen

Ichimikichiki, 2024-09-06:
Why is the killswitch blocking my internet right now?
Shouldn't it only be working when my VPN drops out? And not 100% of the time permanently?

[image]

[image]

[image]

Ichimikichiki, 2024-09-06:
Prick...
[image]

gnuton, 2024-09-07:
Please rewrite this ticket in a professional way and I will listen to you.
What's the problem? 
How can I reproduce the bug?
How this should work instead?
If this is a bug follow the template. 

For question or ranting use the chat.
Thanks