gnuton/asuswrt-merlin.ng #679: Unable to obtain IPV6.

pymumu, 2024-09-08:
**Router Model Affected**
Models: RT-AX82U

**Firmware Version Affected**
388.8_2

**Is this bug present in upstream Merlin releases too?**
NA

**Describe the bug**
After setting the IPV6 rules, restarting the router cannot obtain the IPV6 address assigned by the operator.
After deleting the IPV6 rules or turning off the IPV6 firewall, IPV6 can be successfully obtained.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'IPV6'
2. Select 'Connection type' 'Native', 'Interface' PPP
3. No IPV6 address obtained

**Expected behavior**
There is no problem with the official firmware.
Set IP rules and restart the router to quickly obtain the IPV6 address.

**Screenshots**
admin@RT-AX82U:/tmp/home/root# odhcp6c  -R -s /tmp/dhcp6c -N try -c 00030001c87f540ea510 -P 0:ea510 -r23 -r24 ppp0 -e -vvvv
```
odhcp6c[9068]: (re)starting transaction on ppp0
odhcp6c[9068]: Starting SOLICIT transaction (timeout 4294967295s, max rc 0)
```
[odhcp6c stuck here]

gnuton, 2024-09-08:
Hi!
did you try to run odhcp6c in debug mode (-d) to check what happens?

pymumu, 2024-09-09:
The -d option is `daemonize`, not DEBUG mode. 
The DEBUG option is not found in the parameter list.

```
Invocation options:
        -p <pidfile>    Set pidfile (/var/run/odhcp6c.pid)
        -d              Daemonize
        -e              Write logmessages to stderr
        -v              Increase logging verbosity
        -h              Show this help
```

gnuton, 2024-09-09:
How do you set the rule? What's the config you use?

pymumu, 2024-09-09:
The Rule:
```
Chain FORWARD (policy DROP)
ACCEPT     udp      anywhere             ::xxxx:67ff:fe3b:xxxx/::ffff:ffff:ffff:ffff  state NEW udp dpt:xxxx
```

Steps to reproduce the behavior:
1. Go to 'IPV6'
2. Select 'Connection type' 'Native', 'Interface' PPP
3. Set IPV6 Firmware Rules:
```
[Remote IP]: Empty
[Local IP]: ::xxxx:32ff:fe4a:xxxx/::ffff:ffff:ffff:ffff
[Port]: 1118
[Protocol]: udp
```

I just want to forward the IPV6 request for the corresponding port to an internal host.

MikeyMike80, 2024-11-17:
Just one thing to be mindful of ipv6, after setting it up, about 30/31 minutes will passs and it will appear to have dropped the IPv6 WN gateway, don't do what I did and get all flustered and stressed out.

Leave it for a few minutes and the router will sort itself out and WAP ip will appear again.

Im not sure if it's normal and the router just takes time for the settings to gel, or if it's a small bug, but it really doesn't matter because it all falls into place in the end anyway