gnuton/asuswrt-merlin.ng #685: Assassin Mode and UU Accelerator are not translated into English

alfky12, 2024-09-21:
**Router Model Affected**
Models: TUF-AX3000 (v1)

**Firmware Version Affected**
TUF-AX3000_3004_388.8_2-gnuton1_puresqubi.w
TUF-AX3000_3004_388.8_2-gnuton1_tuf_puresqubi.w

**Is this bug present in upstream Merlin releases too?**
Don't know

**Describe the bug**
After finally being able to enable Assassin Mode on TUF-AX3000 non-China region, it turns out that Assassin Mode and UU Accelerator labels are not translated into English, as seen in the screenshot. Please add appropriate translations.

**Expected behavior**
1. 刺客模式 = Assassin Mode
2. 网易UU加速器 = UU Accelerator
3. UU Accelerator page is also translated into English

**Screenshots**
Not translated to english:
[image]

UU Accelerator page:
[image]

BTW, 4T4R activated on non-china region TUF-AX3000:
[image]

Jacky10X, 2024-09-23:
@alfky12 Hello, I am sorry to bother you.
Could you teach us how to enable 4T4R?
Is it something related to cfe? or firmware?

Jacky10X, 2024-09-24:
It seems your VHT MCS is not fully unlocked?
All should be 0-11 instead of 0-9.

alfky12, 2024-09-25:
> It seems your VHT MCS is not fully unlocked? All should be 0-11 instead of 0-9.

That's because I previously set the Modulation Scheme up to MCS 9.
If it is set up to MCS 11, the result will be like this:

[image]

gnuton, 2024-09-25:
@alfky12 can you describe step by step how did you manage to get this working (posssibly without using CFE?)
I may think to enable this by default

Jacky10X, 2024-09-25:
My method is using [this](https://github.com/JasonJTucker/asus_nvram_editing) to enable 4T4R. (Change it to CN)
but with limited options.
Some firmware authors said it is the limit of the firmware. (The firmware only accepts CN from cfe to enable 4T4R)
I think the second method is to modify the firmware (with cfe?) to unlock the 4T4R for all regions/modes.

alfky12, 2024-10-01:
> @alfky12 can you describe step by step how did you manage to get this working (posssibly without using CFE?) I may think to enable this by default

@gnuton  Yes I edited the CFE, but, as far as I know, the CFE/bootloader is now separated from the factory nvram on new generation asus routers. So it can be said that what I edited was the factory nvram, not the CFE/bootloader that is on the mtd0 partition.
So this method is safer than editing the CFE/bootloader.

Modifications at the firmware level should be possible, as the SWRT dev did. I have compared, the factory nvram is not changed by flashing that firmware.

gnuton, 2025-06-24:
Hi!
Feel free to send me the translation as patch