gnuton/asuswrt-merlin.ng #687: Weird issues with XT8 being used as a mesh node

Igor-Yavych, 2024-09-26:
**Router Model Affected**
Models: ZenWifi AX | XT8

**Firmware Version Affected**
3004.388.8_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
Not sure

**Describe the bug**
I have the following mesh set up:
- XT12 (3004.388.8_2)
  - XT8 (3004.388.8_2-gnuton1)
  - RT-AX92U (3004.388.8_2-gnuton1)
  - RT-AX92U (3004.388.8_2-gnuton1)
 
Every time after power loss, when the XT8 powers back on, it has a blue led on it. It shows up in the network map, as if everything is fine, but nothing connects to it over WI-FI, no matter how much time passes. I have to remove the node from AI Mesh, reset it and then re-add it. After these steps, everything works fine. This issue only happens with XT8, RT-AX92U running the same firmware don't experience anything like this.

gnuton, 2024-09-26:
Hi,
Any log?

NeilS1966, 2024-09-27:
Not sure if it's related, I have an XT8 mesh system and just installed the same version as Igor. I have just one mesh node and it will not join the mesh network. I've placed the node 40 cm from the other node, recycled power on both a few times and I cannot get it to join. I just see a flashing blue LED on the node that won't join. 
It's the first time the devices have been installed with Merlin.

gnuton, 2024-09-28:
@NeilS1966 try to pair via ethernet cable. Then unplug it and it should work.

Igor-Yavych, 2024-10-02:
> Hi, Any log?

I know I said "every time" and that is how it was, but now when I try to reproduce it, I can't. Nevertheless, please keep this issue open. I'll upload a log as soon as I can.

punchoc, 2024-10-04:
it was a problem I had for a long time. I had to do a hard reset to fix the problem. I even thought for a moment that one of my meshes had a malfunction, which was not the case.

Igor-Yavych, 2024-10-24:
Hello.
Finally got a log. Somewhere between 16:00 and 17:00, power went out. 
It came back some time after 17:00. Around 19:30, I rebooted XT12 as well.
Logs include 16:17 and up till this moment. 
[syslog.txt](https://github.com/user-attachments/files/17510107/syslog.txt)