gnuton/asuswrt-merlin.ng #688: Randomly disconnected WAN

burakkara010, 2024-09-26:
**Router Model Affected**
Model: AX92U

**Firmware Version Affected**
3004.388.8_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
I don't know

**Describe the bug**
A random disconnection of the WAN. It already happened 2 times in 3 days. After the disconnection the router also gets unreachable over LAN. It looks like the router gets stuck or frozen. After a physical reboot everything works again. 

**Expected behavior**
If some kind of error happens somewhere, I would expect the router to selfheal. Or when a WAN hiccup happens the router should be reachable or reconnect automatically to the WAN and fix the connection.

georghuber, 2024-09-29:
i have this bug too. The WAN LED is red (however it seems that devices connected via LAN have Internet access, however the RT-AX92 does not have any WLAN devices connected to it).

gnuton, 2024-09-30:
logs?

georghuber, 2024-10-03:
> logs?

Hi, Sorry, I read the post and just posted that I seem to be affected too.

I do unfortunately not know, how I can generate logs for a specific AiMesh node. Can you advise? Thanks, Georg

PS: Thanks for making the MerlinWRT available to other models, BTW!

DrDagless, 2024-11-25:
I'm affected by this also. Have two AX92U units and randomly drop WAN. This has been going on with every firmware and I had to set up a script to reboot when the router detects no WAN connection.

As best as I can tell, it seems to often happen when I'm on a device (my phone), and I'm switching from the router to one of the nodes. I'll see if I can find a log file from one of the times it's happened, but they unfortunately don't seem to indicate much.

CKENT0369, 2024-12-04:
Have the AX52u and am having this issue. Same firmware.