gnuton/asuswrt-merlin.ng #691: ZenWiFi XT8 / RT-AX95Q HW V2 Support

francoism90, 2024-09-28:
**Is your feature request related to a problem? Please describe.**
I've bought a ZenWiFi XT8 / RT-AX95Q v2, which isn't compaitble with the current asuswrt-merlin.ng firmware (this only supports V1).

The issue #511 has been closed, since it's a time ago, I decided create a new ticket. :)

**Describe the solution you'd like**
Support V2, and I'm happy to help to make it work.

**Describe alternatives you've considered**
Stock

**Additional context**
Please let me know what you need, or things I could checkout.

Thanks!

aalmani-dev, 2024-10-03:
+1 for this :)

francoism90, 2024-10-05:
I'm not an expert, but if we could something (checkout source code, basic stuff), I'm happy to create PR. :)

ginodesilva, 2024-11-25:
+1. Let's hope support will come in 2025 since it's the most liked model [BY FAR](https://github.com/gnuton/asuswrt-merlin.ng/issues/523#issuecomment-1890066237) in "New Models to support for 2024".

madcar86, 2025-01-27:
how do you know if you have a v1 or v2? 

Also is there a way to go back to stock in cause things go wrong? And does the second node need to be flashed as well?

francoism90, 2025-01-27:
@madcar86 See README:

> Not sure which version you have? V1? V2? Enable SSH, then ssh into your router and run nvram get productid

It's also printed on the device-label/box.

You can always revert to stock, and no, the second node doesn't have to be Merlin. It does gives you benefits, you don't have to.

Unfortunately the V2 model isn't supported (yet).

Sheron-i, 2025-05-10:
+1 please

commit 7fd10285 (2025-01-06): Add missing 160 MHz support to GT-AC2900 target profile (fixes #691)
closed 2025-06-05 by gnuton