gnuton/asuswrt-merlin.ng #693: Vpn not working

ZliTebra, 2024-10-11:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ZliTebra, 2024-10-11:
Hello everyone, I'm a new member on the forum. I'm on the latest FW and vpn doesn't work on it.  I would appreciate your help

gnuton, 2024-10-14:
Hi,
no support requests here please.
Please use the forum or the chat.

closed 2024-10-14 by gnuton