gnuton/asuswrt-merlin.ng #694: WtFast

ZliTebra, 2024-10-11:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ZliTebra, 2024-10-11:
Is it possible to install the WTFast option for the Tuf ax5400?

BoostOver, 2024-10-15:
Asus no longer supports WtFast how app inside router.

closed 2024-10-17 by gnuton

ZliTebra, 2024-11-28:
I don't know where you got that information from, but all newer Asus routers have that option.

BoostOver, 2024-11-29:
> I don't know where you got that information from, but all newer Asus routers have that option.

Simple…..from Asus Website

ZliTebra, 2024-11-29:
It's a simple upgrade for someone who knows how @gnuton and I would appreciate it if the next firmware update would have that option for the TUF AX5400 as well.

BoostOver, 2024-11-29:
> You didn't understand something well. My friend has a Gt axe16000 and uses WTFast with no problem. I know people who have older model routers and use WTfast. Asus still has a contract with that company. Even the old RT AX88U model has the WTFast option.
> 
> https://youtu.be/EKGQ6tVpcic?si=x-WUUwfO5UZY866i

Againnnnnnnnnnnnnnnnnnnn:
https://www.asus.com/support/faq/1015726/
In the router the button is only a link to wtfast.
All router in the world can work with wtfast!

ZliTebra, 2024-11-29:
Okay.  But you can't say that all Asus routers have stopped working with WTFast.  You can say that only some are affected.  As far as I can see, others are working quite normally.

BoostOver, 2024-11-29:
> Okay. But you can't say that all Asus routers have stopped working with WTFast. You can say that only some are affected. As far as I can see, others are working quite normally.

https://www.asus.com/support/faq/1015726/
https://www.asus.com/support/faq/1015726/
https://www.asus.com/support/faq/1015726/
https://www.asus.com/support/faq/1015726/
https://www.asus.com/support/faq/1015726/
https://www.asus.com/support/faq/1015726/
https://www.asus.com/support/faq/1015726/
https://www.asus.com/support/faq/1015726/
Asus no longer supports WtFast how app inside router.
Asus no longer supports WtFast how app inside router.
Asus no longer supports WtFast how app inside router.
Asus no longer supports WtFast how app inside router.
Asus no longer supports WtFast how app inside router.
Asus no longer supports WtFast how app insiderouter.

ZliTebra, 2024-11-29:
Ok. What is next? FlexQoS for Tuf AX5400?

BoostOver, 2024-11-29:
> Ok. What is next? FlexQoS for Tuf AX5400?
Te lo scrivo in Italiano.
Non capisci niente.
Buon Natale

ZliTebra, 2024-11-30:
I love U 👯‍♂️