gnuton/asuswrt-merlin.ng #697: BE98 v2 - Firmware from beta 2 will not apply

unisoftenterprises, 2024-10-11:
**Router Model Affected**
Models: GT-BE98U 
Not the BE98 Pro model as in UK and BE98U is the model.

**Firmware Version Affected**
3006.102.1_1-gnuton0_beta2

**Is this bug present in upstream Merlin releases too?**
Not applicable as Merlin only does the PRO model.

**Describe the bug**
Brand new GT-BE98U hardware version 2. Applies latest Asus firmware fine.
When going to apply GNUTON build, a firmware unsuccessful message because of transmission error or model.
Checksum fine.

**To Reproduce**
Get v2 BE98
Apply GNUTON Beta 2 firmware, watch for error dialogue.

**Expected behaviour**
For it to apply.

gnuton, 2024-10-17:
I have never heard of a GT-BE98U V2.
If such revision exist than my firmware may not support it yet.
I do not have access to such router.
I will ask ask for clarifications.

lorenc-tomasz, 2024-10-17:
I have search information about v2, as I was wondering what they fixed in new revision, but I couldn't find any information. If you will get info from Asus, please post it here :)

unisoftenterprises, 2024-10-17:
Its the default  in UK for past couple of months, unless a retailer has old stock. It's likely to be similar in other countries now or if not, soon. Thanks GNUTON for reaching out to Asus/Merlin/Others for clarification. I only bought this because of Merlin/GNUTON firmware, the stock firmware is horrible, mainly because of lack of AMTM and add-ons as well as DNS Director and that awful ROG GUI :)

There is a thread on SNB Forums that says a barometer is going to be added to a 2.5/2.7 hardware revision for the BE98 **Pro** as well. I don't know if this will affect the BE98 later, as I THINK the requirement is a USA requirement and BE98 is not aimed at US market.

A customer references on Asus forums, the v2 BE98 here:
https://rog-forum.asus.com/t5/gaming-routers/next-firmware-update-for-gt-be98/td-p/1025377

And the SNB forum link for v2 of BE98 Pro:
https://www.snbforums.com/threads/standard-power-2-new-hardware-ver-for-be98-pro.91115/

It might be that the BE98 v2 simply has the AFC added as stated in the SNB forums.

gnuton, 2024-10-17:
I have just realeized we got this news from asus a few months back.
Correct the new HW version has indeed the barometer to become compliant to some regulations.
My current undersanding is that if yo try to flash the current firmware for V1 to a V2 HW, the process will crash because of a missing barometer.

So my current guess is that this will be fixed with the new GPL taht we will get from ASUS soon.
I will keep you posted for updates.

unisoftenterprises, 2024-10-17:
OK, thank you. I eagerly await a new version then, as the Asus stock firmware has driven me nuts. I reported so many issues back to Asus via feedback with bugs already. 

On Asus firmware, Network Map>Client list has display issues and issues around saying STATIC or MANUAL when its all DHCP, VPN Fusion with Sharksurf doesn't allow me to actually define which devices go through it due to mandatory ticks on most devices, Adaptive QOS just seems broken before I even get onto annoyances with WIFI setup and things on Guest network that you'd actually like to set on Main Network WiFi like bandwidth limiter, different DHCP IP address range to main LAN network etc.

mattheweli, 2024-10-19:
I've a v2 GT-BE98 and I installed Gnuton FW without any issue...

unisoftenterprises, 2024-10-19:
> I've a v2 GT-BE98 and I installed Gnuton FW without any issue...

I don't know how then. I have a BRAND NEW sealed one, with latest Asus Stock firmware, and before I even had it in service, it would not load GNUTON "3006.102.1_1-gnuton0_beta2" firmware using the inbuilt updater. Mine definitely says "H/W Ver 2.0" and "MFG Year 2024" on the box.

mattheweli, 2024-10-19:
If you have 3.0.0.6.102_36852, I'm quite sure you cannot downgrade it...:

*Please be advised that due to an architecture upgrade in AiMesh, we highly discourage downgrading to earlier firmware versions as it may result in connectivity problems. If you experience any difficulties, we suggest resetting the AiMesh router to its default settings and re-establishing the mesh connection.

This is the discalimer of the firmware download for that version

unisoftenterprises, 2024-10-19:
> If you have 3.0.0.6.102_36852, I'm quite sure you cannot downgrade it...:
>
> *Please be advised that due to an architecture upgrade in AiMesh, we highly discourage downgrading to earlier firmware [...]

Yep, that why I think this beta won't apply after I had completed the Asus setup process and Asus firmware update. I any event new customers will be landed with same issue going forward as v2 is current stock unless a retailer has older stock.

BoostOver, 2024-10-19:
Have you started flash from FW version?

unisoftenterprises, 2024-10-19:
> Have you started flash from FW version?

Went through Asus setup as brand new and arrived at 3.0.0.6.102_36852-gfe236b0_925-gd358a_BB0B.

Not sure if it will allow a flash back to April's Asus firmware from before this one - I know they recommend not doing because of AIMesh issues with previous firmware, but I don't have AIMESH. If flashing back means it doesn't cause a permanent issue then I could try as would want to be flashing to GNUTON after anyway. I've had Asus routers for years, using Merlin's firmware as been in the supported list and never have had Asus put stuff into firmware that prevents flashing.

BoostOver, 2024-10-19:
GPL Gnuton older
You can’t flash this firmware
Bye

josh-blake, 2024-10-30:
Just got the BE98 here in Australia; very keen for a Merlin update here. Have also tried putting the router in DFU mode and it still fails. The lack of AdGuard + unbound as stock on the router is killing me.

cbtham, 2024-11-05:
The BE-98 I've gotten in Singapore is V2 as well, no luck flashing firmware

unisoftenterprises, 2024-11-05:
Asus released 3.0.0.6.102_36855 today. Have not upgraded, for fear of new (as yet unreleased) GNUTON release not applying.

gnuton, 2024-11-07:
@unisoftenterprises I do not get that.. feel free to update the firmware.

AruChen, 2024-11-10:
GT-BE98 is sold as GT-BE25000 in the United States. Does the GT-BE25000 support this firmware?

gnuton, 2024-11-10:
you can flash the latest stock Asus gt-be98 on it. If that works then my version should work too. 
At moment I am not supporting V2 yet, but the support is coming.

unisoftenterprises, 2024-11-10:
Stock firmware for GT-BE98 (not Pro model) should apply to the GT-BE25000 as exactly the same version and change control notes.
However, Asus do have a download link for this model so you can flash with certainty:
https://rog.asus.com/networking/rog-rapture-gt-be25000/helpdesk_bios/

lorenc-tomasz, 2024-11-10:
> Can you flash the latest asus stock gt-be98 fw on it? If that works then my version should work too. At moment I am not supporting V2 yet, but the support is coming.

@gnuton Do you know the difference between v1.0 and v2.0? We know what 2.5/2.7 will bring, but I'm wondering if we know what 2.0 changed?

gnuton, 2024-11-10:
> Stock firmware for GT-BE98 (not Pro model) should apply to the GT-BE25000 as exactly the same version and change control notes.
> However, Asus do have a download link for this model so you can flash with certainty:
> https://rog.asus.com/networking/rog-rapture-gt-be25000/helpdesk_bios/
>
You can try to flash the firmware. If the firmware is not correct, then the web app will make the flashing operation fail. It s safe to test.

Meffistowy, 2024-11-25:
@gnuton Hi, do you know, roughly, when v2 will be supported? Thank you

gnuton, 2024-11-25:
No timeline yet. We are waiting for the goals from Asus. I can ping them again to get some insights.

Meffistowy, 2024-11-25:
@gnuton If you could that would be great, many thanks

Meffistowy, 2024-11-26:
@gnuton Thank you very much for the information

liamstears, 2024-11-29:
I am eagerly awaiting an update so we can get a newer @gnuton firmware but in the meantime I may have some good news for those having trouble flashing the current beta

So these routers simply do not allow a downgrade from the newest couple of firmwares and that's why you get the failed message, I bought a used GT-BE98 v1 and encountered the issue however I was able to get gnuton firmware working

The way to do it is to downgrade the current firmware to an earlier firmware so then you can upgrade to gnuton. This cannot be done from the standard router gui though. So these are the steps:

Head over to Asus support site: https://rog.asus.com/networking/rog-rapture-gt-be98-model/helpdesk_download/
Download the ASUS Firmware Restoration Utility
Download an older firmware, links to these have been removed from the Asus website though but I did find a working link to an earlier firmware here: https://dlcdnets.asus.com/pub/ASUS/wireless/GT-BE98/FW_GT_BE98_300610234372.zip?model=GT-BE98
Put the router into recovery mode. This is done by turning it off using the switch, push and hold a paperclip or something into the reset hole and keep it held while turning on the router again. Keep it held in, first you will get just a power light then eventually the aura rgb will be a pinkish colour. This is now recovery mode and reset can be let go of.
Set computer/laptop ethernet to ip 192.168.1.10, subnet 255.255.255.0, gateway etc 192.168.1.1
Using the firmware restoration utility flash the earlier firmware. Once its done its thing just wait and the router will boot up on its own using the older firmware.

At this point your free to flash gnuton using the GUI

I have only tested this on a v1, someone above mentioned gnuton works on a v2 so this process could very well work on a v2 also but you do this at your own risk, even if you have a v1 again you do this at your own risk, but the process worked for me and I'm happily now on gnuton so thanks @gnuton

Marchcat2008, 2024-12-19:
> I am eagerly awaiting an update so we can get a newer @gnuton firmware but in the meantime I may have some good news for those having trouble flashing the current beta
>
> So these routers simply do [...]

Can you upload the firmware to a file sharing service?
The link you provided is not working ((
Thank you.

dennissch, 2024-12-29:
Thank you for the description, @liamstears. I was able to successfully reset the firmware on my GT-BE98 (HW Version 2.0) using it and upload the latest gnuton version ([GT-BE98_3006_102.1_1-gnuton0_beta2](https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3006.102.1_1-gnuton0_beta2)). It was a bit tricky, but it worked flawlessly in the end. Perhaps my description will help others as well:

- See the instructions above for resetting the firmware. I was able to find the old firmware with some googling, and I’ve uploaded to Wetransfer [(GT-BE98_3.0.0.6_102_34372-gb0fcfc4_620-gaf023_BA08_nand_squashfs.zip)](https://we.tl/t-bIVisZxODp) since the original site no longer exists. Please use it at your own risk; it worked for me.
- After flashing, the router restarted. I couldn’t access the router via the network to upload the new firmware through the GUI. Therefore, after the restart, I put the router back into flash mode again and uploaded the new firmware using the tool.
- Afterward, I reset my network settings and was able to access the GUI via http://www.asusrouter.com/ or http://192.168.1.1. The new firmware was successfully uploaded.

Cheers!

Marchcat2008, 2025-01-04:
RMerlin wrote how to easily disable the firmware downgrade ban.
This method worked for me without any manipulation using a firmware recovery program.
--------------------------------------------------------------
If flashing on top of Asus stock firmware 3006.102_37000
or greater, then you first need to enable downgrade
capabilities before flashing Asuswrt-Merlin on top
of it.  Over SSH, run the following command:

      nvram set DOWNGRADE_CHECK_PASS=1

After that, you can upload Asuswrt-Merlin through
the webui like any regular firmware upgrade.

This is only required when flashing Asuswrt-Merlin
 for the first time.

gnuton, 2025-01-14:
Hi there,
The new beta should solve this issue.

unisoftenterprises, 2025-01-14:
Will try when back home. Thanks for all your work on this.
Out of interest, Merlin doesn't do the ROG interface version for GT-BE98 Pro, I don't care for ROG interface either, you could save yourself time by not doing ROG UI for future versions for GT-BE98?

Marchcat2008, 2025-01-14:
> Попробую, когда вернусь домой. Спасибо за всю твою работу над этим. Из интереса, Merlin не делает версию интерфейса ROG для GT-BE98 Pro, мне тоже не нравится интерфейс ROG, ты мог бы сэкономить себе время, не делая ROG UI для будущих версий для GT-BE98?

Gnuton offers a choice of interface options. I, on the contrary, like the ROG interface.

unisoftenterprises, 2025-01-15:
Does this beta firmware have the VLAN GUI under LAN menu option?

liamstears, 2025-01-15:
> Does this beta firmware have the VLAN GUI under LAN menu option?

Yes

unisoftenterprises, 2025-01-15:
> > Does this beta firmware have the VLAN GUI under LAN menu option?
> 
> Yes, running latest one myself and it's there: https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3006.102.3_1-gnuton0_beta1

thank you.

mattheweli, 2025-01-16:
> > Does this beta firmware have the VLAN GUI under LAN menu option?
> 
> Yes, running latest one myself and it's there: https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3006.102.3_1-gnuton0_beta1

So, still no 802.1q options on WAN page, right?

liamstears, 2025-01-17:
> > > Does this beta firmware have the VLAN GUI under LAN menu option?
> > 
> > 
> > Yes, running latest one myself and it's there: https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3006.102.3_1-gnuton0_beta1
> 
> So, still no 802.1q options on WAN page, right?

Not that I can see...

unisoftenterprises, 2025-01-18:
**### Some feedback on 3006.102.3_1-gnuton1 beta 1 using GT-BE98 (v2 hardware):**

1. Upon applying 3006.102.3_1-gnuton1 beta 1 over the top of Asus stock firmware 3.0.0.6.102_36878, upon firmware completion which was successful and rebooted, the router was no longer connecting to the WAN and red light was on WAN LED. I use the 10gbps WAN port. Upon going to the original IP address of my router which is different to default IP, I was presented with the first setup Out Of Box wizard. So a flag got reset, which made the router present this dialogue as it still had my WiFi SSIDs setup as before. I applied the update over a WAN connection to test applying remotely (it was a fast FTTP connection).

2. This option is missing under "T**OOLS > OTHER SETTINGS**": 
Wan: **Use local caching DNS server as system resolver (default: No)**

3. The description field for Profile 5 (WCG5) under WIREGUARD was missing yet labelled in Asus firmware. In ASUS firmware this is under default gateways - one was default without VPN and another one was to pass traffic for selected devices through Wireguard using Sharksurf VPN configuration details.

Other than this, so far its working well, and look forward to final version release.

unisoftenterprises, 2025-01-18:
Another observation:

On a mobile phone, when pressing the WIRELESS menu option, the page for wireless settings does not resize for mobile screen and is too big to fit. All the other menu options on classic Asus UI do resize

unisoftenterprises, 2025-01-20:
> > > > Does this beta firmware have the VLAN GUI under LAN menu option?
> > > 
> > > 
> > > Yes, running latest one myself and it's there: https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3006.102.3_1-gnuton0_beta1
> > 
> > 
> > So, still no 802.1q options on WAN page, right?
> 
> Not that I can see...

There wasn't in Asus stock firmware. VLAN options for WAN are under LAN > IPTV and then Manual Settings option.
VLAN options for ethernet and WiFi are under LAN > VLAN

mattheweli, 2025-01-20:
> There wasn't in Asus stock firmware. VLAN options for WAN are under LAN > IPTV and then Manual Settings option. VLAN options for ethernet and WiFi are under LAN > VLAN

Not for GT-BE98, but for RT-BE88U and RT-BE86U it's on WAN page and it could prevent the issue of not using the 2,5Gbit WAN port because of being specified on LAN section

gnuton, 2025-01-20:
@unisoftenterprises a factory reset from stock to merlin is necessary. That could also prevents small bugs and lags.