gnuton/asuswrt-merlin.ng #702: dose it work with the ZenWiFi XT9

DJkitty2, 2024-11-09:
I own the ZenWiFi XT9 and I’m curious to know if there are any compatible versions or if there are any other options that work. Please let me know if I’ve overlooked something.

James-s1126, 2024-11-09:
Same question here. +1

X1pheR, 2025-01-03:
Same. Hope it gets supported

X1pheR, 2025-01-03:
It actually looks like the XT8 and XT9 are identical except that the XT9 supports more channels. Somebody compared firmwares. See [Source](https://www.snbforums.com/threads/xt8-vs-xt9-decision.88820/). Perhaps this would make it easier for @gnuton to also support the XT9 AKA XT8-PRO. 🤞🏻😅

gnuton, 2025-01-03:
Hey,
I checked the configs and it looks like they build against the same SDK and same hardware (BCM53134) but they have different target profiles (947622GW vs 96756GW). I am not sure although much much compatible are the prebuilds.

This is from the target.mak
you can see there that they are different BCM53134 vs  
export HND-947622_BASE := HND_ROUTER=y HND_ROUTER_AX_675X=y PROFILE="947622GW" ...
export RT-AX95Q := $(HND-947622_BASE)
export RT-AX95Q += BUILD_NAME="RT-AX95Q" SWITCH2="BCM53134" 

export HND-96756_BASE := HND_ROUTER=y HND_ROUTER_AX_6756=y PROFILE="96756GW" ...
export XT8PRO := $(HND-96756_BASE)
export XT8PRO += BUILD_NAME="XT8PRO" SWITCH2="BCM53134" NVSIZE="128"

X1pheR, 2025-01-03:
> Hey, I checked the configs and it looks like they build against the same SDK and same hardware (BCM53134) but they have different target profiles (947622GW vs 96756GW). I am not sure although much [...]

Thanks for the quick reply! So I guess this means that they can't be easily supported? I was hoping it would be easily supported as they looked identical and thus we would get XT9 support on Merlin. 😁

X1pheR, 2025-01-03:
> I do not have access to XT9 GPL nor I do not have any hw, but I would guess it may be possible to hack something to get XT9 work with XT8 stuff we now have.

That would be great! Is there any way I can help? I'm no programmer but have 2 XT9's in mesh I could run things on to provide info or such maybe.

hbastos, 2025-01-03:
Hi all

GPL available on ASUS website (but old version I believe, unsure how it works): https://dlcdnets.asus.com/pub/ASUS/wireless/ZenWiFi_XT9/GPL_ZENWIFI_XT9_300438821963.zip?model=ZenWiFi%20XT9

I'm also interested in XT9 support.  willing to alpha/beta test.

Best,
HB

nschimme, 2025-01-31:
I was looking at how we integrated the XT8_V2 and noticed that there is extensive XT8PRO support already included as part of one of the previous merges: https://github.com/search?q=repo%3Agnuton%2Fasuswrt-merlin.ng%20XT8PRO&type=code

gnuton, 2025-02-03:
@nschimme I have started an experimental branch to support the XT9. I will share more updates when I have something concrete

zay2u, 2025-02-15:
This is a very interesting project. Looking forward to see how this progresses.

gnuton, 2025-02-17:
TLDR
Bad news. I cannot continue on this so I will close the ticket.

Long story:
I tried to mix stuff, but  the GPL is required. XT9 builds against a completely differnt SDK from 8.

So, there are 2 options:
1. Asus would provide GPLs for this rounter taylored for merlin - AFAIK ASUS has no interest in this.
2. someone forks the entire merlin and merge the public GPLs

zay2u, 2025-02-17:
Is option 2 how the XT8 v1 gained Merlin?

gnuton, 2025-02-19:
@zay2u my fork is actually in sync with merlin all the time and I get merlin GPLs straight from ASUS.
Option 2 means forking and diverging.

gnuton, 2025-02-20:
@eduardoRoth no  it won't. From the GPL we merge SDK, new code and the prebuilds.
Old prebuilds is the main issue since they will spark into linking errors during the building process

zay2u, 2025-02-20:
@gnuton how does a router become a candidate for merlin GPLs being provided  straight from ASUS? In addition to the XT9 GPL referenced by @eduardoRoth, even ones merlin himself supports are dated on the Asus official support page (for example the GT-AXE16000).

gnuton, 2025-02-20:
@zay2u I am not sure how the marketing strategy team really works. But in general they know which router wanna see more in the market or wanna have more sales. So they select some routers among those and I suggest also what I am interested in to support.
From what I see (my personal opinion) Asus is focused on WiFi 6 routers and the new major asuswrt release on them.

gnuton, 2025-02-21:
@zay2u  I see NO chance for the XT9 to be supported by ASUS.

closed 2025-02-21 by gnuton

zay2u, 2025-02-21:
@gnuton I trust you have insight to this. I would like to reiterate we appreciate your work; and I was simply suggesting to make the request to Asus to see if we might get lucky.  BTW, you mention that Asus has moved to WiFi6 routers. And Asus markets and states on their website that the XT9 is a WiFi6 router. What is Asus considering a WiFi6 router?

gnuton, 2025-02-21:
I asked this in the past. The current state is that Asus got very slow in providing GPLs. So adding more old models is not feasible anymore.