gnuton/asuswrt-merlin.ng #704: killed wifi on RT-AX92U after upgrade to original firmware

panariga, 2024-11-15:
**Router Model Affected**
Models: RT-AX92U

**Firmware Version Affected**
3004.388.8_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
After upgrading to 3004.388.8_2-gnuton1 from the previous gnuton version, I faced firewall issues. I decided to do a factory reset, but it didn't help. So, I upgraded to the latest official firmware, 2024/11/12 FW_RT_AX92U_300438823695.

After that, my Wi-Fi disappeared. Only the 2.4 GHz network shows channels for configuration, but it doesn't work as a real radio. The 5 GHz network configurations aren't available at all.

It seems like something got messed up in the NVRAM. Maybe the region configuration got lost during the upgrades.

Log

May  5 08:05:07 kernel: ubi1: detaching mtd10
May  5 08:05:07 kernel: ubi1: mtd10 is detached
May  5 08:05:07 kernel: Loading HND Module
May  5 08:05:07 kernel: igs_module_init:867:	 IGS 18 create network socket successful
May  5 08:05:07 kernel: cfg80211: Calling CRDA to update world regulatory domain
May  5 08:05:07 kernel: dpsta_init: msglevel set to 0x1
May  5 08:05:07 kernel: wl_module_init: passivemode set to 0x1
May  5 08:05:07 kernel: wl_module_init: txworkq set to 0x0
May  5 08:05:07 kernel: bcm43684_nic 0 devid 0x4360
May  5 08:05:07 kernel: wl 0000:01:00.0: enabling device (0000 -> 0002)
May  5 08:05:07 kernel: wl0: creating kthread wl0-kthrd
May  5 08:05:07 kernel: wl driver 17.10.121.41 (r796515) failed with code 1
May  5 08:05:07 kernel: bcm43684_nic 0 devid 0x4360
May  5 08:05:07 kernel: wl 0001:01:00.0: enabling device (0000 -> 0002)
May  5 08:05:07 kernel: wl0: creating kthread wl0-kthrd
May  5 08:05:07 kernel: wl driver 17.10.121.41 (r796515) failed with code 1
May  5 08:05:07 kernel: bcm43684_nic 0 devid 0xaaa4
May  5 08:05:07 kernel: skip probing ax chip in wl, devid 0xaaa4

**Screenshots**
[image]

Squall-Leonhart, 2024-11-19:
You were likely infected by the recent malwares affecting these AX devices, death of the Wifi on factory reset is one of the observed symptoms

panariga, 2024-11-19:
> You were likely infected by the recent malwares affecting these AX devices, death of the Wifi on factory reset is one of the observed symptoms
Seems like so. It is a good idea to have a generic NVRAM dump for the model so recovery settings can be generated. As I suspect, only the MAC addresses should be changed
Can anybody share a working NVRAM?

darcyrush, 2025-01-18:
> It is a good idea to have a generic NVRAM dump for the model so recovery settings can be generated

Anyone have a link to a method that supports recent merlin firmwares?