gnuton/asuswrt-merlin.ng #707: freezes/Crashes after VPN setup

Matt3ra, 2024-11-17:
**Router Model Affected**
Models: TUF-AX5400

**Firmware Version Affected**
3004.388.8_2-gnuton1

**Describe the bug**

The router crashes and is not able to reach when i set a new rule on the VPN director.
I want to direct all trafic in the router trough an VPN.
I use 192.168.1.0/24.

As soon as i hit "apply" the router freezes and it's not possible to enter 192.168.1.1 again.
You have to do a hard reset and install the firmware again to be able to reach the router.

I have followed this guide;
[https://mullvad.net/sv/help/asus-merlin-and-mullvad-vpn](url)

Any sugestions?

Matt3ra, 2024-11-28:
I have now taken some more time for this.

The router does not crash if i put specific units trough the VPN director. But if i choose all unit it crashes.

gnuton, 2024-12-02:
Can you share logs?

Matt3ra, 2024-12-02:
> Can you share logs?

I'm afraid not.
Since the router crashes and is not possible to reach without full hardware reset and then reinstalling the firmware.
So there is no possibility to reach the logs im afraid.

What im testing now is putting one unit trough the VPN director at the time. I dont know if it gets to much for the router with routing "192.168.1.0/24."

Aa123123-aa, 2024-12-20:
简单设置一键翻墙，速度瞬间提升5～10倍。我在大陆地区如何上网，从事涉外网络配置专业工程师10年经验分享。不走弯路，不被坑！2024年度最具性价比VPN机场真心推荐。下载地址：
https://expvpn.net
https://vpndapps.com
https://letsvpnt.com
https://letskuailian.com