gnuton/asuswrt-merlin.ng #713: FlexQoS

ZliTebra, 2024-11-30:
I would be very grateful if you could do FlexQoS for Tux AX5400 in the next firmware. Thanks for everything!

BoostOver, 2024-11-30:
Againnnnnnnnnnnnn:
Via AMTM install FlexQos.

ZliTebra, 2024-12-01:
I'm new to this forum and I don't even know what AMTM is.  Where do you look for it and where is it located? Is it a mini firmware that needs to be downloaded?  Thank you

ZliTebra, 2024-12-01:
[image]

ZliTebra, 2024-12-01:
Do we mean the same thing?  I found this image on Google

BoostOver, 2024-12-01:
> Do we mean the same thing? I found this image on Google

Install Gnuton FW:
https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3004.388.8_4-gnuton0_alpha1
Enable Telnet on router
On pc use a client Telnet to establish connection with your router
Login
Type: amtm
Configure initial startup of amtm
Install FlexQos from menu.

Buon Natale

ZliTebra, 2024-12-01:
I'm new to all this. I'm a slow learner. Thanks a lot for your help.  I will try to do what you wrote to me, but I can already see that it will be a bit complicated.  I use an Asus router exclusively for gaming.

ZliTebra, 2024-12-01:
Can you help me? I'll give you access to my PC to show me how to do it.  Thank you

BoostOver, 2024-12-01:
> Can you help me? I'll give you access to my PC to show me how to do it. Thank you

No.
Can you get help from:
https://www.snbforums.com/threads/flexqos-1-4-4-flexible-qos-enhancement-script-for-adaptive-qos.90645/

ZliTebra, 2024-12-01:
I don't have the ability to download.  It seems that the Tuf AX 5400 does not support this function.

alfky12, 2024-12-01:
Just use CAKE, because FlexQoS development is no longer continued by the dev.

[image]

It's useless if you ask a lot of questions here, github is not a place for beginners, come here to seek knowledge: https://www.snbforums.com/

ZliTebra, 2024-12-02:
How can I learn if I don't ask questions?

alfky12, 2024-12-07:
Man, if you've got 2-4 ms ping but still say "The delay is noticeably constant”, "I still feel like I'm lagging behind other players", it means that it's a skill issue, you're not competitive, but instead blame your internet connection. Sorry to say.

ZliTebra, 2024-12-07:
You know how, we can play 1vs1 and many things will be much clearer.  I left my YouTube channel where you can see everything about my skills. 

https://youtube.com/@zlitebra9487?si=jLcxztaTakMCbdhr

It's obvious that you don't understand some things.  When I said 2ms and 4ms I meant the stability of my network from the ISP where local servers are viewed.  If you mean Activision servers, that's another story and I have between 40 and 70ms.  Given that I'm from the Balkans, where they don't have servers for that game, and the closest one is 1600 km away.

andrewk123, 2024-12-10:
OK to make it simple (well the best I can anyway).  
- AMTM stands for Asuswrt-Merlin Terminal Menu.  If you have Asuswrt-Merlin 384.15 or greater, AMTM is built in.  If you have an earlier version than that, fix that first (update your router).
- After that is done you need to ssh into your router.  If you cannot figure that out on your own then just ignore the rest of this post. (Note: ASUS has instructions how to set that up on their website plus there a bunch of youtube videos out there that cover the same ground).
- Once you have logged into the router with SSH and have a terminal prompt simply type amtm and enter.  
- From the AMTM menu enter i to get a list of available scripts that can be installed.  As of now on my router (XT8) FlexQOS is item 3 on the list.
- type in the the number next to FlexQOS and enter and the application will load itself. 
- After that, [this link](https://github.com/dave14305/FlexQoS) gives you instructions how to use FlexQOS (Note: I suggest you use AMTM to install FlexQOS rather than the installation instructions in the link since AMTM is less prone to typing errors).
- After that, I cannot help any further since I'm not a gamer and don't use or need QOS but that should get you started.

ZliTebra, 2024-12-10:
> OK to make it simple (well the best I can anyway).  
> - AMTM stands for Asuswrt-Merlin Terminal Menu.  If you have Asuswrt-Merlin 384.15 or greater, AMTM is built in.  If you have an earlier version than that, fix that first (update your router).
> - After that is done you need to ssh into your router.  If you cannot figure that out on your own then just ignore the rest of this post. (Note: ASUS has instructions how to set that up on their website plus there a bunch of youtube videos out there that cover the same ground).  It took Google plus 60 secs to get youtube and ASUS links.
> - Once you have logged into the router with SSH and have a terminal prompt simply type amtm and enter.  
> - From the AMTM menu enter i to get a list of available scripts that can be installed.  As of now on my router (XT8) FlexQOS is item 3 on the list.
> - type in the the number next to FlexQOS and enter and the application will load itself. 
> - After that, [this link](https://github.com/dave14305/FlexQoS) gives you instructions how to use FlexQOS (Note: I suggest you use AMTM to install FlexQOS rather than the installation instructions in the link since AMTM is less prone to typing errors).
> - After that, I cannot help any further since I'm not a gamer and don't use or need QOS but that should get you started.
> 

I would be very grateful if you would like to install this application -AnyDesk Remote Desktop- on your computer and I would give you the password to access my computer and while you are doing it, I would record everything with my phone.  I'm not really a PC gamer and have always been on PlayStation.  Everything on the PC is foreign to me. I'm from the old school.  45 years old and 3 children.  I like to play just that one game when I get home from work and have some time.  Thank you! 

[Screenshot_2024-12-10-21-46-21-154_com.android.vending.jpg](https://github.com/user-attachments/assets/2bcffc0f-567c-4765-96fc-b29f2728b100)

andrewk123, 2024-12-10:
> I would be very grateful if you would like to install this application -AnyDesk Remote Desktop- on your computer and I would give you the password to access my computer and while you are doing it, I [...]

I'm sorry I would not be able to do that.  If you want this feature you will need to do the install yourself.

closed 2025-01-13 by ZliTebra