gnuton/asuswrt-merlin.ng #714: Gaming Improvements 🎮🕹️ TUF AX5400

ZliTebra, 2024-12-02:
Is there a possibility to make a Geo-Filter for the Asus router like the Netduma R3 has!?

https://youtube.com/shorts/nLuCUF2wwNI?si=lBno-gGMIA43bU1e

Can you do anything related to Gaming in any of the next Firmware? In every new Firmware you release, there is nothing improved and new for Gaming.  My router is called TUF GAMING, but it is not worthy of that name because its gaming functions are bad and everything is very simple.
Do anything new related to Gaming. Be creative and imaginative. Thanks a lot!

BoostOver, 2024-12-02:
Gnuton non ne fai una giusta🤣
Questo gli spieghi le cose e non le comprende

ZliTebra, 2024-12-02:
Man, you are becoming a real temptation for my brain. You are chasing me around the forums and leaving stupid comments.  Come on, take a break from me for 2 days, please.  hahahaha

gnuton, 2024-12-09:
@BoostOver che ho fatto?

gnuton, 2024-12-09:
@ZliTebra @BoostOver is respetable and experienced person and he is joking with you.

I do have really little time and no interest in gaming at all to promise you anything on this area

closed 2024-12-09 by gnuton

ZliTebra, 2024-12-09:
I went back to the official Asus Firmware, it's much more stable, and there was an update in November. I honestly don't even know what you could do for Gaming. It requires a little more work and skills.