gnuton/asuswrt-merlin.ng #716: Adding a scheduler feature for guest wifi networks

manvoud, 2024-12-12:
I am working from home and I use one of the guest networks to connect my work laptop only and I don't want it to be active let's say on weekend. 
And right now you can only set for how many days you want the guest network to be active.

What I propose as a feature is something like having the option to **activate the guest network for certain day(s)/time period**.  

Example scenarios:

- "everyday from 9 am to 5 pm"
- "on weekdays 9 am to 5pm"

xTudoS, 2024-12-20:
Nice! You can archive this using cru and a bash script as a workaround. but I think this is more related to https://github.com/RMerl/asuswrt-merlin.ng upstream, because [gnuton](https://github.com/gnuton) fetchs from there and is not a router related issue.

MNLierman, 2025-01-10:
Super easy to schedule. `wl -i wl1.1 down` something like that. Of course, you would need to check into it, but this could be accomplished in 10 min or less.